Koenkk/zigbee2mqtt #15859: zigbee-herdsman crash with Error: DatabaseEntry with ID '3' already exists

myuc, 2023-01-01:
### What happened?

Zigbee docker container crash when re-adding missing zigbee devices to Zigbee2MQTT 

### How to reproduce it (minimal and precise)

Crash happens intermittently when re-adding device. Unable to co-relate crash to a single type/model of device as the device that causes it to crash is different every time. 

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee Dongle Plus (ZBdongle-E) EZSP v8

### Debug log

```
2023-01-01T12:45:17.831Z zigbee-herdsman:adapter:ezsp:uart <-- [15feb1a9702aebb6d5e33600aa1e80499d4e27ab48b57e]
2023-01-01T12:45:17.831Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,5,0): 15feb1a9702aebb6d5e33600aa1e80499d4e27ab48b57e
2023-01-01T12:45:17.831Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2023-01-01T12:45:17.831Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2023-01-01T12:45:17.832Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: bc90012400fe048c777c25004b120001000000
2023-01-01T12:45:17.832Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":1278,"newNodeEui64":{"type":"Buffer","data":[0,18,75,0,37,124,119,140]},"status":1,"policyDecision":0,"parentOfNewNodeId":0}
2023-01-01T12:45:17.832Z zigbee-herdsman:adapter:ezsp:ezsp ==> findKeyTableEntry: {"address":{"type":"Buffer","data":[0,18,75,0,37,124,119,140]},"linkKey":true}
2023-01-01T12:45:17.832Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"findKeyTableEntry","_id_":117,"_isRequest_":true,"address":{"type":"Buffer","data":[0,18,75,0,37,124,119,140]},"linkKey":true}
2023-01-01T12:45:17.832Z zigbee-herdsman:adapter:ezsp:uart --> DATA (5,2,0): bd000175008c777c25004b120001
2023-01-01T12:45:17.832Z zigbee-herdsman:adapter:ezsp:uart --> [52ff21a9212a99c525b14a6eb85593cd877e]
2023-01-01T12:45:17.832Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (6)
2023-01-01T12:45:17.832Z zigbee-herdsman:adapter:ezsp:debg Device join request received: 1278 00124b00257c778c
2023-01-01T12:45:17.832Z zigbee-herdsman:controller:log Device '0x00124b00257c778c' joined
2023-01-01T12:45:17.832Z zigbee-herdsman:controller:log Device '0x00124b00257c778c' accepted by handler
2023-01-01T12:45:17.832Z zigbee-herdsman:controller:log Delete device '0x00124b00257c778c' joined, undeleting
Error: DatabaseEntry with ID '3' already exists
    at Database.insert (/app/node_modules/zigbee-herdsman/src/controller/database.ts:35:19)
    at Device.undelete (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:395:25)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:543:20)
```

closed 2023-02-08 by github-actions[bot]

jc-plhm, 2023-02-14:
I have the same issue,

```
  zigbee-herdsman:controller:log Device '0x84fd27fffe932847' accepted by handler +1ms
  zigbee-herdsman:controller:log Delete device '0x84fd27fffe932847' joined, undeleting +0ms
Error: DatabaseEntry with ID '4' already exists
    at Database.insert (/app/node_modules/zigbee-herdsman/src/controller/database.ts:40:19)
    at Device.undelete (/app/node_modules/zigbee-herdsman/src/controller/model/device.ts:396:25)
    at Controller.onDeviceJoined (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:543:20)
```

junosuarez, 2023-03-15:
I was unable to reproduce this. Anyone still having this issue, can you check your database.db file in the data directory (alongside your configuration.yaml) and see what device has the ID mentioned in the logs (it will be different for each installation).

My hunch from looking over the code in zigbee-herdsman is that this error is coming when trying to re-add a device that still exists in the database, and that the semantics for `Device.undelete` should be an upsert, rather than a strict insert or a strict update. I want to confirm that the IDs match the same device, though, and that it's not "the existing ID that has been in the meantime re-issued to another device".