Koenkk/zigbee2mqtt #15862: Issues with triggering updates via frontend

sjorge, 2023-01-01:
### What happened?

I think after the ota payload change, updates stopped being possible via the frontend.

I noticed this because all my trådfri remote suddenly showed as no update available, which made me panic a bit as I am keeping them on the last group binding supported version.

### What did you expect to happen?

Updates to still show as available and being triggerable via the frontend.

### How to reproduce it (minimal and precise)

Debug log when hitting the `Check for new update` button on a bulb that has an update available:
```
Zigbee2MQTT:debug 2023-01-01 16:19:50: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"light/livingroom/mood/bulb1","transaction":"qbt2k-1"}'
Zigbee2MQTT:info  2023-01-01 16:19:50: Checking if update available for 'light/livingroom/mood/bulb1'
Zigbee2MQTT:debug 2023-01-01 16:19:50: Check if update available for '0x001788010bb1e7d7' (LCA008)
Zigbee2MQTT:debug 2023-01-01 16:19:50: Using endpoint '11'
Zigbee2MQTT:debug 2023-01-01 16:19:50: Received Zigbee message from 'light/livingroom/mood/bulb1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784402,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2023-01-01 16:19:50: Got OTA request '{"fieldControl":0,"manufacturerCode":4107,"imageType":276,"fileVersion":16784402}'
Zigbee2MQTT:debug 2023-01-01 16:19:50: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2023-01-01 16:19:50: Is new image available for '0x001788010bb1e7d7', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":276,"fileVersion":16784402}', latest meta '{"fileVersion":16784906,"fileSize":478740,"url":"https://otau.meethue.com/storage/ZGB_100B_0114/090d8075-bcc7-4bf6-aead-bfc9d4926468/100B-0114-01001E0A-ConfLightBLE-Lamps-EFR32MG21.zigbee","sha512":"e097485e23a42329b6c18c0ab3561fd1491fcc2eda0058487987190566b6e3eecc788164cc89bee326083c799aa61a3bf077d375667f4aeb628060784c494599"}'
Zigbee2MQTT:debug 2023-01-01 16:19:50: Update available for '0x001788010bb1e7d7': NO
Zigbee2MQTT:info  2023-01-01 16:19:50: No update available for 'light/livingroom/mood/bulb1'
Zigbee2MQTT:info  2023-01-01 16:19:50: MQTT publish: topic 'zigbee2mqtt/light/livingroom/mood/bulb1', payload '{"brightness":118,"color":{"hue":30,"saturation":85,"x":0.4739,"y":0.4126},"color_mode":"xy","color_options":{"execute_if_off":true},"color_temp":396,"linkquality":87,"power_on_behavior":"off","state":"OFF","update":{"installed_version":16784402,"latest_version":16784906,"state":"idle"}}'
Zigbee2MQTT:info  2023-01-01 16:19:50: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"light/livingroom/mood/bulb1","updateAvailable":false},"status":"ok","transaction":"qbt2k-1"}'
```

Frontend of course believes there is no update based on the response. I can manual trigger the update as there is one available. (Installed: 1.93.11, Available:  1.101.7)

Manually triggeren an update (non foced):
```
mosquitto_pub -t zigbee2mqtt/bridge/request/device/ota_update/update -m '{"id": "light/livingroom/mood/bulb1"}'
```

```
Zigbee2MQTT:debug 2023-01-01 16:22:04: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id": "light/livingroom/mood/bulb1"}'
Zigbee2MQTT:info  2023-01-01 16:22:04: Updating 'light/livingroom/mood/bulb1' to latest firmware
Zigbee2MQTT:debug 2023-01-01 16:22:04: Received Zigbee message from 'light/livingroom/mood/bulb1', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20220502","swBuildId":"1.93.11"}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2023-01-01 16:22:04: Updating to latest '0x001788010bb1e7d7' (LCA008)
Zigbee2MQTT:debug 2023-01-01 16:22:04: Using endpoint '11'
Zigbee2MQTT:debug 2023-01-01 16:22:04: Received Zigbee message from 'light/livingroom/mood/bulb1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784402,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID 0
Zigbee2MQTT:debug 2023-01-01 16:22:04: Got OTA request '{"fieldControl":0,"manufacturerCode":4107,"imageType":276,"fileVersion":16784402}'
Zigbee2MQTT:debug 2023-01-01 16:22:04: ZigbeeOTA: downloaded main index
Zigbee2MQTT:debug 2023-01-01 16:22:04: getNewImage for '0x001788010bb1e7d7', meta {"fileVersion":16784906,"fileSize":478740,"url":"https://otau.meethue.com/storage/ZGB_100B_0114/090d8075-bcc7-4bf6-aead-bfc9d4926468/100B-0114-01001E0A-ConfLightBLE-Lamps-EFR32MG21.zigbee","sha512":"e097485e23a42329b6c18c0ab3561fd1491fcc2eda0058487987190566b6e3eecc788164cc89bee326083c799aa61a3bf077d375667f4aeb628060784c494599"}
Zigbee2MQTT:debug 2023-01-01 16:22:04: ZigbeeOTA: downloading firmware image from https://otau.meethue.com/storage/ZGB_100B_0114/090d8075-bcc7-4bf6-aead-bfc9d4926468/100B-0114-01001E0A-ConfLightBLE-Lamps-EFR32MG21.zigbee
Zigbee2MQTT:debug 2023-01-01 16:22:04: OTA update checksum validation succeeded for '0x001788010bb1e7d7'
Zigbee2MQTT:debug 2023-01-01 16:22:04: getNewImage for '0x001788010bb1e7d7', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4107,"imageType":276,"fileVersion":16784906,"zigbeeStackVersion":2,"otaHeaderString":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":478740}
Zigbee2MQTT:debug 2023-01-01 16:22:04: Got new image for '0x001788010bb1e7d7'
Zigbee2MQTT:debug 2023-01-01 16:22:04: Starting upgrade
Zigbee2MQTT:debug 2023-01-01 16:22:04: Received Zigbee message from 'light/livingroom/mood/bulb1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784402,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID 0
```

### Zigbee2MQTT version

1.29.0-dev commit: a49079f5

### Adapter firmware version

20221226

### Adapter

zzhp-lite

WhimsySpoon, 2023-01-01:
Have also noticed this with the update to `1.29.0`. Installed via the Home Assistant Add-on.

The json `state` for each device shows an update is available in `update` object, however the front end seems to show the update button based on the value of `update_available` which is being set to `null`

Example below:

```
{
    "brightness": 254,
    "color": {
        "h": 212,
        "hue": 212,
        "s": 10,
        "saturation": 10,
        "x": 0.3131,
        "y": 0.3232
    },
    "color_mode": "color_temp",
    "color_temp": 153,
    "last_seen": "2023-01-01T17:44:29+00:00",
    "linkquality": 116,
    "state": "ON",
    "update": {
        "installed_version": 16784402,
        "latest_version": 16784906,
        "state": "idle"
    },
    "power_on_behavior": null,
    "update_available": null
}
```

Home Assistant is reporting the devices as having firmware updates available (I guess it's using the `update` object to confirm this, and installing the firmware via HA works fine.

rccoleman, 2023-01-02:
I think it's [this](https://github.com/Koenkk/zigbee-herdsman-converters/issues/5246) .

Koenkk, 2023-01-02:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

sjorge, 2023-01-02:
This indeed fixed it for me. <3

closed 2023-01-02 by sjorge