Koenkk/zigbee2mqtt #15864: Warning Messages after Upgrade to 1.29.0-1

UPEngineer, 2023-01-01:
### What happened?

After updating the zigbee2mqtt addon to version 1.29.0-1, I am seeing a lot of warning messages in the HA logs. Granted they are just warnings, but should not be there nevertheless.

The warning message states:
Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'

I can see multiple messages in the zigbee2mqtt logs for my devices with this message.

### What did you expect to happen?

No error or warning messages.

### Zigbee2MQTT version

1.29.0-1

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

arnishow, 2023-01-01:
Same here after upgrade to 1.29.0-1

cinghialino, 2023-01-01:
+1 here,  same problem~!

AbeltjeNL, 2023-01-01:
Same here. Nothing more to add because I'm getting exactly the same error messages.

evangelo18, 2023-01-01:
I am also tons of messages so much that it appears the network is being flooded

Koenkk, 2023-01-02:
Same as https://github.com/Koenkk/zigbee2mqtt/issues/15747

Please check if this has been fixed in the dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jjansen85, 2023-01-02:
Running latest dev, still an issue
_1.29.0-dev commit: [826f032](https://github.com/Koenkk/zigbee2mqtt/commit/826f032)_

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:589
First occurred: 16:27:44 (406 occurrences)
Last logged: 20:25:37

Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'
```

Koenkk, 2023-01-03:
Can you provide the debug log when starting z2m?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

jjansen85, 2023-01-03:
Sure
[_zigbee2mqtt_logs (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/10336442/_zigbee2mqtt_logs.1.txt)

Koenkk, 2023-01-04:
This should be fixed now in the latest dev.

Anyway, let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/15747

closed 2023-01-04 by Koenkk