Koenkk/zigbee2mqtt #15867: Random flash and reset

johanneswilkens, 2023-01-01:
### What happened?

I recently switched to z2m, everything seems to work okay, but I experienced some weird behavior.
Some bulbs randomly flash full brightness. They do this when they are on (go full brightness for a sec and go back to normal) or when they are off (flash for a period almost like someone is taking a picture with flash). They seem to flash in the same color temperature as they are set to, so not their default color. The bulbs are [wifilampkoning own brand CCT filament bulbs](https://www.wifilampkoning.nl/slimme-led-lampen/led-lampen-e27-fitting/zigbee-led-filament-lamp-dual-white-7w-e27-fitting-hue-alternatief-led-lamp/) and use cheap no-brand touchlink controllers. They identify themselves as single color [ecodim bulbs](https://www.zigbee2mqtt.io/devices/ED-10041.html). If this is happening the logs stay empty, so I don't know where this is coming from (the bulbs didn't flash on deconz, only dropped connection after a while)

### Zigbee2MQTT version

1.29.0 (and previous)

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus (the one with Texas Instruments chip)

ASNNetworks, 2023-01-06:
I have some EcoDim filament bulbs myseld and have noticed this for a few months now. After a while I figured out what caused it. For some reason, if you enter Z2M pairing mode (where you don't select a specific zigbee router) and then disable pairing mode again, all EcoDim bulbs randomly flash ever few minutes. This will never stop, untill you restart z2m add-on. So my guess is you have been pairing some devices lately as well. Restart z2m add-on and you will notice it wil stop the flashing. Enable pairing mode again and they will all flash randomly (even when off).

When using pairing mode where you use a specific zigbee router to pair through (NOT selecting an EcoDim bulb) will not cause this. So I just use a specific device to pair through now.

I never reported it because I have no idea how to even troubleshoot stuff like this, since there isn't logged anything.

johanneswilkens, 2023-01-06:
Interesting, I'll check if it indeed solves the issue for this type of bulb as well. I do automatically restart HA every week, I didn't know this would only affect the HA container itself and doesn't restart addons.
One thing I did notice that might be related to this issue is that those bulbs are requesting an OTA every minute. I see these messages only for this type of bulb. 

```
Debug 2023-01-06 11:40:52Received Zigbee message from 'Veranda 1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-06 11:40:52Device 'Veranda 1' requested OTA
Debug 2023-01-06 11:41:31Received Zigbee message from 'Veranda 3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-06 11:41:31Device 'Veranda 3' requested OTA
Debug 2023-01-06 11:41:52Received Zigbee message from 'Veranda 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537135121,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-06 11:41:52Device 'Veranda 2' requested OTA
Debug 2023-01-06 11:41:53Received Zigbee message from 'Veranda 1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-06 11:41:53Device 'Veranda 1' requested OTA
Debug 2023-01-06 11:42:32Received Zigbee message from 'Veranda 3', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-06 11:42:32Device 'Veranda 3' requested OTA
Debug 2023-01-06 11:43:29Received Zigbee message from 'Veranda 2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537135121,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Debug 2023-01-06 11:43:29Device 'Veranda 2' requested OTA
```

ASNNetworks, 2023-01-06:
> I didn't know this would only affect the HA container itself and doesn't restart addons.

Yes correct, this is a huge benefit because you will be able to control devices even when restarting or updating HA Core. I use Node-RED for automations and with the mqtt and zigbee2mqtt nodes, you can have Node-RED directly communicate with z2m through mqtt. It also allows for faster response times with stuf like brightness stepping.

Interesting find about the OTA every minute. That seems unnecessary excessive and not ideal for the zigbee network.

johanneswilkens, 2023-01-06:
I tried restarting the Z2M addon but the light is still flashing, probably you got a different type of bulb. I think all cheap bulbs, tuya or other non tuya home brands are all identifying themselves as ecodim in Z2M. It might be that the generic ecodim converter doesn't work for all of those bulbs?

ASNNetworks, 2023-01-07:
> I tried restarting the Z2M addon but the light is still flashing, probably you got a different type of bulb. I think all cheap bulbs, tuya or other non tuya home brands are all identifying [...]

Sorry, my mistake! I had EcoDim lights before, but replaced them with [Müller Lights](https://www.zigbee2mqtt.io/devices/404065.html). The EcoDim lights were from their own brand (so not like in your case). They worked fine, but they became defective so I got a refund from Amazon, after they were not sold anymore. 

The Müller lights have the issue I described, but after restarting Z2M it works fine again. Or using a specific device to pair also prevents the issue. So I think my issue is unrelated.

closed 2023-02-15 by github-actions[bot]