Koenkk/zigbee2mqtt #15868: No device found after coordinator upgrade

maxime1992, 2023-01-01:
### What happened?

Wanted to upgrade my [SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator](https://www.amazon.fr/gp/product/B09KXTCMSC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) to latest firmware.

- Had my z2m running as usual
- Stopped it (`docker compose stop z2m`)
- Ran `git clone git@github.com:JelmerT/cc2538-bsl.git and cd cc2538-bsl`
- Grabbed `CC1352P2_CC2652P_launchpad_coordinator_20220219.zip` from  https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
- `wget` + `unzip` the file above
- Flashed the hex file using `cc2538-bsl.py`
- It was all successful
- Restarted the entire server just in case
- Ran a `docker compose up -d` to start everything

Except that now I do not have any device shown in z2m. Checked the logs and it says: `Zigbee2MQTT:info  2023-01-01 20:44:39: Currently 0 devices are joined`, nothing else really interesting.

I've got 80 devices, including some in the attic, some behind my walls (to control lights and heaters). It'd be *SO* painful to have to pair them all again. I desperately hope there'll be a quick fix :pray: 

### What did you expect to happen?

I thought Z2M had a feature to automagically restore everything transparently after a coordinator upgrade.

- https://community.home-assistant.io/t/zigbee2mqtt-coordinator-upgrade-how-to-backup-restore-devices/275986/13?u=maxime1992
- https://github.com/Koenkk/zigbee-herdsman/pull/395#issuecomment-891672790
- https://www.zigbee2mqtt.io/guide/faq/#doesn-t-require-repairing

Did I somehow misunderstood this? Is there anything else required?

### How to reproduce it (minimal and precise)

Explained in first section

### Zigbee2MQTT version

1.28.4-dev commit: 3481423

### Adapter firmware version

20220219

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P Coordinator

### Debug log

```
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-01-01 20:44:35: Logging to console and directory: '/app/data/log/2023-01-01.20-44-35' filename: log.txt
Zigbee2MQTT:info  2023-01-01 20:44:35: Starting Zigbee2MQTT version 1.28.4-dev (commit #3481423)
Zigbee2MQTT:info  2023-01-01 20:44:35: Starting zigbee-herdsman (0.14.80)
Zigbee2MQTT:info  2023-01-01 20:44:39: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-01-01 20:44:39: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2023-01-01 20:44:39: Currently 0 devices are joined:
Zigbee2MQTT:info  2023-01-01 20:44:39: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2023-01-01 20:44:40: Connecting to MQTT server at mqtt://192.168.0.100:1883
Zigbee2MQTT:info  2023-01-01 20:44:40: Connected to MQTT server
Zigbee2MQTT:info  2023-01-01 20:44:40: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-01-01 20:44:40: MQTT publish: topic 'zigbee2mqtt/z2m_all_lights/availability', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-01-01 20:44:40: Started frontend on port 0.0.0.0:8264
Zigbee2MQTT:info  2023-01-01 20:44:40: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2023-01-01 20:44:40: Zigbee2MQTT started!
Zigbee2MQTT:info  2023-01-01 20:51:42: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
Zigbee2MQTT:info  2023-01-01 20:51:42: Disconnecting from MQTT server
Zigbee2MQTT:info  2023-01-01 20:51:42: Stopping zigbee-herdsman...
```

maxime1992, 2023-01-01:
Ok eventually... [I got it working](https://discord.com/channels/556563650429583360/1052136886912688199/1059223981623877712). But it was quite a close call due to unfortunate event with my backups... 

Anyway, I'm pretty sure something went wrong the first time as I didn't erase the file manually with z2m, but somehow my database.db file got overridden. I do not have anything to prove this, so feel free to close the issue as I may honestly just be loosing my mind at this point. But I could swear I didn't touch the database file myself. So might be worth checking if somehow z2m might have written over it somehow after I flashed the coordinator :shrug: Unsure sorry!

maxime1992, 2023-01-01:
I'm apparently not the only one. See this: https://github.com/Koenkk/zigbee-herdsman/pull/395#issuecomment-1368530835

I know everyone should make regular backups but it's still a bit concerning IMO, hence why I'm keeping the issue open :)

Koenkk, 2023-01-02:
Z2M should not clear the database.db as long as the `coordinator-backup.json` file is present. Was this file present when you updated the dongle?

maxime1992, 2023-01-02:
> Z2M should not clear the database.db as long as the `coordinator-backup.json` file is present. Was this file present when you updated the dongle?

Yes for sure, I've got all the config files saved in my GIT, this file included and it didn't appeared as removed. Even when the database was empty I could still see `coordinator_backup.json` and `state.json` with all my 80 devices

Koenkk, 2023-01-02:
Can you reproduce the issue? If yes, can you provide me the herdsman debug logging of this? 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

maxime1992, 2023-01-02:
I'm afraid I won't be able to. Got back to work today so limited time for me and I only have 1 coordinator that I'd rather prefer not to mess up too much with, sorry.

If that ever happens again though I'll do the herdsman debug. As I'm not helping meanwhile, feel free to close that issue if it's not possible to debug without a repro

closed 2023-02-09 by github-actions[bot]