Koenkk/zigbee2mqtt #15870: Template variable warning: 'dict object' has no attribute 'latest_version'

haforme, 2023-01-01:
### What happened?

Upgrade to zigbee2mqtt version 1.29.0-1 from version 1.28.4-1 and started receiving error messages in the log. 

Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'

### What did you expect to happen?

No error messages.

### How to reproduce it (minimal and precise)

Started up zigbe2mqtt addon.

### Zigbee2MQTT version

1.29.0-1

### Adapter firmware version

Could not locate info requested for adapter using above link.

### Adapter

Conbee II

DarwinData, 2023-01-01:
Same here.  I have about 60 devices and a ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus and I can't tell which device is causing this same error.

peterbogar, 2023-01-02:
The same for me, with Sonoff

Source: helpers/template.py:589

SVH-Powel, 2023-01-02:
I have around 15000 messages each hour, so it is kind of filling up the log........

Koenkk, 2023-01-02:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/15747

closed 2023-01-02 by Koenkk