Koenkk/zigbee2mqtt #15873: Xiaomi ZNJLBL01LM no longer reports Open or Closed just Opening and Closing

10Thirty, 2023-01-02:
### What happened?

After updating to 1.29 this morning my Aqara E1 motor no longer reports to home assistant that it has opened or closed instead staying on opening or closing long after the motor has stopped running.

I have tried removing and re-adding the motor and also updated the coordinator to the latest firmware but it has made no difference.

In zigbee2mqtt in the state tab shows "state": "OFF", no mater the position of the blind unless you hit the stop button from which point on it will show "state": "STOP", even when the motor is running

[image]

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20221226

### Adapter

Slaesh CC2652RB

buhito81, 2023-01-02:
Same here, from the MQTT log in HA

```
2023-01-02 09:56:20.544 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T08:56:20.526Z","linkquality":255,"motor_state":"pause","position":6,"power_outage_count":4,"running":false,"state":"STOP","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'cover.kitchen_cover', 'name': 'Kitchen Cover', 'this': <template TemplateStateFromEntityId(cover.kitchen_cover)>} and Template("{{ value_json.motor_state if value_json.motor_state and value_json.motor_state != 'pause' else value_json.state }}")
2023-01-02 09:56:20.544 WARNING (MainThread) [homeassistant.components.mqtt.cover] Payload is not supported (e.g. open, closed, opening, closing, stopped): STOP
```

```
2023-01-02 09:59:21.765 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":61,"charging_status":false,"device_temperature":18,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T08:59:21.746Z","linkquality":255,"motor_state":"pause","position":76,"power_outage_count":3,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'cover.dressingroom_cover', 'name': 'Dressingroom Cover', 'this': <template TemplateStateFromEntityId(cover.dressingroom_cover)>} and Template("{{ value_json.motor_state if value_json.motor_state and value_json.motor_state != 'pause' else value_json.state }}")
2023-01-02 09:59:21.766 WARNING (MainThread) [homeassistant.components.mqtt.cover] Payload is not supported (e.g. open, closed, opening, closing, stopped): OFF
```
Same Zigbee2MQTT version, but ConbeeII adapter.

Koenkk, 2023-01-02:
Can you provide the debug log when closing and opening the track?

buhito81, 2023-01-02:
Sure. This **opening**:

```
2023-01-02 10:12:34.118 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on zigbee2mqtt/Kitchen Cover: b'{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T09:12:34.110Z","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}'
2023-01-02 10:12:34.119 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T09:12:34.110Z","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'sensor.kitchen_cover_battery', 'name': 'Kitchen Cover battery', 'this': <template TemplateStateFromEntityId(sensor.kitchen_cover_battery)>} with default value 'default' and Template("{{ value_json.battery }}")
2023-01-02 10:12:34.120 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T09:12:34.110Z","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'sensor.kitchen_cover_motor_state', 'name': 'Kitchen Cover motor state', 'this': <template TemplateStateFromEntityId(sensor.kitchen_cover_motor_state)>} with default value 'default' and Template("{{ value_json.motor_state }}")
2023-01-02 10:12:34.120 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T09:12:34.110Z","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'sensor.kitchen_cover_device_temperature', 'name': 'Kitchen Cover device temperature', 'this': <template TemplateStateFromEntityId(sensor.kitchen_cover_device_temperature)>} with default value 'default' and Template("{{ value_json.device_temperature }}")
2023-01-02 10:12:34.120 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T09:12:34.110Z","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'binary_sensor.kitchen_cover_charging_status', 'name': 'Kitchen Cover charging status', 'this': <template TemplateStateFromEntityId(binary_sensor.kitchen_cover_charging_status)>} and Template("{{ value_json.charging_status }}")
2023-01-02 10:12:34.121 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T09:12:34.110Z","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'binary_sensor.kitchen_cover_running', 'name': 'Kitchen Cover running', 'this': <template TemplateStateFromEntityId(binary_sensor.kitchen_cover_running)>} and Template("{{ value_json.running }}")
2023-01-02 10:12:34.121 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T09:12:34.110Z","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'binary_sensor.kitchen_cover_update_available', 'name': 'Kitchen Cover update available', 'this': <template TemplateStateFromEntityId(binary_sensor.kitchen_cover_update_available)>} and Template("{{ value_json['update']['state'] == "available" }}")
2023-01-02 10:12:34.121 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"battery":100,"charging_status":false,"device_temperature":19,"illuminance":0,"illuminance_lux":0,"last_seen":"2023-01-02T09:12:34.110Z","linkquality":255,"motor_state":"pause","position":100,"power_outage_count":4,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'position_open': 100, 'position_closed': 0, 'tilt_min': 0, 'tilt_max': 100, 'entity_id': 'cover.kitchen_cover', 'name': 'Kitchen Cover', 'this': <template TemplateStateFromEntityId(cover.kitchen_cover)>} and Template("{{ value_json.position }}")
[... cut ...] ty":255,"motor_state":"declining","position":100,"power_outage_count":4,"running":true,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"},"update_available":false}' with variables {'entity_id': 'update.kitchen_cover', 'name': 'Kitchen Cover', 'this': <template TemplateStateFromEntityId(update.kitchen_cover)>} and Template("{{ value_json['update']['latest_version'] }}")
```

I hope I've filtered out all lines correctly; but I have about 150 MQTT devices.... :-)

CMDR-Sloma, 2023-01-02:
I have the same problem :

```
Logger: homeassistant.components.mqtt.cover
Source: components/mqtt/cover.py:390
Integration: MQTT (documentation, issues)
First occurred: 1 January 2023 at 11:10:08 (32940 occurrences)
Last logged: 09:45:58

Payload is not supported (e.g. open, closed, opening, closing, stopped): OFF
Payload is not supported (e.g. open, closed, opening, closing, stopped): None
```

The following warning seems to be related to the above, as it started to appear only after the latest update:

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:589
First occurred: 1 January 2023 at 11:10:07 (30359 occurrences)
Last logged: 09:43:28

Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'
```
Also, I don't know why the same devices behave differently, both on the same firmware both work, but only one of them fully reports additional information. Any suggestions to fix other controller?

Master:
```
{
    "battery": 100,
    "charging_status": false,
    "device_temperature": 18,
    "illuminance": 0,
    "illuminance_lux": 0,
    "linkquality": 156,
    "motor_state": "pause",
    "position": 70,
    "power_outage_count": 2,
    "running": false,
    "state": "OFF",
    "update": {
        "installed_version": 3870,
        "latest_version": 3870,
        "state": "idle"
    },
    "update_available": false
}
```
Livingroom blinds:
```
{
    "battery": 98,
    "charging_status": false,
    "linkquality": 51,
    "position": 72,
    "update": {
        "installed_version": 3870,
        "latest_version": 3870,
        "state": "idle"
    },
    "update_available": false,
    "device_temperature": null,
    "motor_state": null,
    "running": null,
    "state": null
}
```

Relevant debug lines from Zigbee2MQTT:
```
2023-01-02 10:22:17 Received MQTT message on 'zigbee2mqtt/Livingroom blinds/set' with data '{ "position": 71 }'
2023-01-02 10:22:17 Publishing 'set' 'position' to 'Livingroom blinds'
2023-01-02 10:22:30 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,18,5,33,3,0,9,33,0,1,10,33,138,231,11,32,0,12,32,1,13,35,30,15,0,0,17,35,1,0,0,0,100,32,30,101,32,100,102,32,128,103,32,6,105,32,2,106,33,37,32,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
2023-01-02 10:22:30 lumi.curtain.acn002: Processed buffer into data {"3":18,"5":3,"9":256,"10":59274,"11":0,"12":1,"13":3870,"17":1,"100":30,"101":100,"102":128,"103":6,"105":2,"106":8229,"107":0}
2023-01-02 10:22:30 lumi.curtain.acn002: unknown key 13 with value 3870
2023-01-02 10:22:30 lumi.curtain.acn002: unknown key 17 with value 1
2023-01-02 10:22:30 lumi.curtain.acn002: Processed data into payload {"device_temperature":18,"power_outage_count":2,"state":"OFF","battery":100}
2023-01-02 10:22:30 lumi.curtain.acn002: Processed data into payload {"device_temperature":18,"power_outage_count":2,"state":"OFF","battery":100}
2023-01-02 10:23:33 Received MQTT message on 'zigbee2mqtt/MasterBedroomBlinds/set' with data '{ "position": 70 }'
2023-01-02 10:23:33 Publishing 'set' 'position' to 'MasterBedroomBlinds'

2023-01-02 10:27:59 Received MQTT message on 'zigbee2mqtt/MasterBedroomBlinds/set' with data '{ "position": 0 }'
2023-01-02 10:27:59 Publishing 'set' 'position' to 'MasterBedroomBlinds'
2023-01-02 10:28:00 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":117440543,"presentValue":1}' from endpoint 1 with groupID 0
2023-01-02 10:28:00 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'aqaraOpple', data '{"1033":1}' from endpoint 1 with groupID 0
2023-01-02 10:28:00 lumi.curtain.acn002: Processed data into payload {"charging_status":true}
2023-01-02 10:28:04 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":1514496,"presentValue":2}' from endpoint 1 with groupID 0
2023-01-02 10:28:05 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":100}' from endpoint 1 with groupID 0
2023-01-02 10:28:05 Received Zigbee message from 'MasterBedroomBlinds', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":100}' from endpoint 1 with groupID 0
2023-01-02 10:28:08 Received MQTT message on 'zigbee2mqtt/MasterBedroomBlinds/set' with data '{ "position": 70 }'
2023-01-02 10:28:08 Publishing 'set' 'position' to 'MasterBedroomBlinds'
2023-01-02 10:28:09 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":117440545,"presentValue":0}' from endpoint 1 with groupID 0
2023-01-02 10:28:14 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'genMultistateOutput', data '{"61440":1514496,"presentValue":2}' from endpoint 1 with groupID 0
2023-01-02 10:28:14 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":30}' from endpoint 1 with groupID 0
2023-01-02 10:28:15 Received Zigbee message from 'MasterBedroomBlinds', type 'readResponse', cluster 'genAnalogOutput', data '{"presentValue":30}' from endpoint 1 with groupID 0
2023-01-02 10:28:15 Received Zigbee message from 'MasterBedroomBlinds', type 'attributeReport', cluster 'aqaraOpple', data '{"1033":2}' from endpoint 1 with groupID 0
2023-01-02 10:28:15 lumi.curtain.acn002: Processed data into payload {"charging_status":false}
```

CMDR-Sloma, 2023-01-02:
Quick update, I repaired the other controller, and it reports all parameters as the first one. The main issue still stands.

golles, 2023-01-02:
I could not open/close my 2 blinds through HA (automations). My logs don't show anything that hasn't been already shared here before.
Rolling back to `1.28.4` fixes the problem for now.

Btw, the release notes don't mention any changes to ZNJLBL01LM

Koenkk, 2023-01-03:
This might be caused by https://github.com/Koenkk/zigbee2mqtt/pull/15555 , reverted this now. Can you check if the latest dev works OK now?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

golles, 2023-01-03:
The current `latest-dev` image with sha256:baee78c37c0e1083bcedc1d45fe6e7063bee986fbcc8a6a672b23d5fdd4771bd works fine for me.

```log
info  2023-01-03 15:33:33: Starting Zigbee2MQTT version 1.29.0-dev (commit #15ccf94)
info  2023-01-03 15:33:33: Starting zigbee-herdsman (0.14.83)
```

ASNNetworks, 2023-01-03:
I have the ZNCLDJ11LM model and the state is also broken. In HA it always shows as open, in Z2M frontend it has no state. Attributes work fine and open and close command work as well, just the state that is broken. I updated to the latest dev build and that also seemed to fix my model. Reporting here incase other come across the same issue that have my model.

10Thirty, 2023-01-04:
Just tried 1.29.0-dev commit: 15ccf94 and can confirm that it is working as expected again.

Xitro01, 2023-01-04:
Same issue for ZNCLDJ12LM. Hopefully this will be pushed to main asap.

Koenkk, 2023-01-04:
Great, I'll make a new hotfix release version soon (1.29.1)

cmiguelcabral, 2023-01-07:
Is someone willing to help me fix this?
I don't have any Xiaomi blind controller, and it looks like I cannot buy it anymore as they are reported as discontinued.
I just need someone that could test my changes to be sure it is ok before opening a new PR.

Koenkk, 2023-01-08:
@10Thirty @Xitro01 @ASNNetworks @golles @CMDR-Sloma @buhito81  can you check if your Xiaomi cover still works correctly in the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

buhito81, 2023-01-08:
> @10Thirty @Xitro01 @ASNNetworks @golles @CMDR-Sloma @buhito81  can you check if your Xiaomi cover still works correctly in the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Yes, they do.

Koenkk, 2023-01-08:
@buhito81 sorry, the changes were not available yet, can you check again in 1 hour from now?

ASNNetworks, 2023-01-08:
> @buhito81 sorry, the changes were not available yet, can you check again in 1 hour from now?

I just updated the edge version and log says: 1.29.1-dev (commit #30df0e2). Can you tell me what the commit will be, so I'll know for sure if I have the latest build when I re-install and test again?

cmiguelcabral, 2023-01-08:
I don't understand how can this happen. The changes I made now are not supposed to break anything as they only apply when there's the 'moving' property.
Those Xiaomi blinds are a bit weird as the report state, motor_state, and running at the same time (at least from the logs you presented before).

ASNNetworks, 2023-01-08:
> I don't understand how can this happen. The changes I made now are not supposed to break anything as they only apply when there's the 'moving' property. Those Xiaomi blinds are a bit weird as the report state, motor_state, and running at the same time (at least from the logs you presented before).

With my model the running and motor_state has never worked. Even though it exposes running and motor_state, those variables are always N/A with my model in the frontend.

buhito81, 2023-01-08:
This is very similar to the issue I've reported previously in https://github.com/Koenkk/zigbee2mqtt/issues/15116

cmiguelcabral, 2023-01-08:
Can someone please post the payload of the MQTT message published from z2m when the cover is stopped and while moving, and also the discovery message published from z2m to home assistant?

delinend, 2023-01-09:
Hmm.. I updated to 1.29.1 last evening, and now my Zigbee2mqtt periodic crach.
It has something to do, with my Aqara roller shade companion E1 (Xiaomi) and OTA.

"Zigbee2MQTT:debug 2023-01-09 05:22:06: lumi.curtain.acn002: unknown key 13 with value 3870"
"Zigbee2MQTT:debug 2023-01-09 05:23:41: lumi.curtain.acn002: unknown key 17 with value 1"

Value 3870 is Firmware version.

But all Binds works (start, stop, up and down), untill crach.

`Zigbee2MQTT:debug 2023-01-09 05:22:06: Received Zigbee message from 'Roller - Bryggers1', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,20,5,33,1,0,9,33,0,1,10,33,201,24,11,32,0,12,32,1,13,35,30,15,0,0,17,35,1,0,0,0,100,32,0,101,32,50,102,32,192,103,32,3,105,32,2,106,33,134,28,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-01-09 05:22:06: lumi.curtain.acn002: Processed buffer into data {"3":20,"5":1,"9":256,"10":6345,"11":0,"12":1,"13":3870,"17":1,"100":0,"101":50,"102":192,"103":3,"105":2,"106":7302,"107":0}
Zigbee2MQTT:debug 2023-01-09 05:22:06: lumi.curtain.acn002: unknown key 13 with value 3870
Zigbee2MQTT:debug 2023-01-09 05:22:06: lumi.curtain.acn002: unknown key 17 with value 1
Zigbee2MQTT:debug 2023-01-09 05:22:06: lumi.curtain.acn002: Processed data into payload {"device_temperature":20,"power_outage_count":0,"state":"OFF","battery":50}
Zigbee2MQTT:debug 2023-01-09 05:22:06: lumi.curtain.acn002: Processed data into payload {"device_temperature":20,"power_outage_count":0,"state":"OFF","battery":50}
Zigbee2MQTT:info  2023-01-09 05:22:06: MQTT publish: topic 'zigbee2mqtt/Roller - Bryggers1', payload '{"battery":50,"charging_status":false,"device_temperature":20,"linkquality":112,"motor_state":"pause","position":0,"power_outage_count":0,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"}}'
Zigbee2MQTT:debug 2023-01-09 05:23:41: Received Zigbee message from 'Roller - Kontor1', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,23,5,33,1,0,9,33,0,1,10,33,31,44,11,32,0,12,32,1,13,35,30,15,0,0,17,35,1,0,0,0,100,32,0,101,32,85,102,32,192,103,32,3,105,32,2,106,33,193,30,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-01-09 05:23:41: lumi.curtain.acn002: Processed buffer into data {"3":23,"5":1,"9":256,"10":11295,"11":0,"12":1,"13":3870,"17":1,"100":0,"101":85,"102":192,"103":3,"105":2,"106":7873,"107":0}
Zigbee2MQTT:debug 2023-01-09 05:23:41: lumi.curtain.acn002: unknown key 13 with value 3870
Zigbee2MQTT:debug 2023-01-09 05:23:41: lumi.curtain.acn002: unknown key 17 with value 1
Zigbee2MQTT:debug 2023-01-09 05:23:41: lumi.curtain.acn002: Processed data into payload {"device_temperature":23,"power_outage_count":0,"state":"OFF","battery":85}
Zigbee2MQTT:debug 2023-01-09 05:23:41: lumi.curtain.acn002: Processed data into payload {"device_temperature":23,"power_outage_count":0,"state":"OFF","battery":85}
Zigbee2MQTT:info  2023-01-09 05:23:41: MQTT publish: topic 'zigbee2mqtt/Roller - Kontor1', payload '{"battery":85,"charging_status":false,"device_temperature":23,"linkquality":109,"motor_state":"pause","position":0,"power_outage_count":0,"running":false,"state":"OFF","update":{"installed_version":3870,"latest_version":3870,"state":"idle"}}'`

CMDR-Sloma, 2023-01-09:
> @10Thirty @Xitro01 @ASNNetworks @golles @CMDR-Sloma @buhito81 can you check if your Xiaomi cover still works correctly in the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Yes, it is working fine now.

@delinend - I think I had a similar issue a while ago with one of my controllers, after removing and re-pairing it went away.

cmiguelcabral, 2023-01-09:
@delinend and @CMDR-Sloma, is it working on the latest dev branch?
@Koenkk, are my changes still on that branch?

cmiguelcabral, 2023-01-09:
In the first PR, the one which broke Xiaomi ZNJLBL01LM and similars, I changed the behavior so it used the state property when not moving.
It looks like those devices report 'off' when no moving, which is not a valid state for a cover. That's why it was not working.
The changes I made on this last PR, are supposed not to affect anything previously done, but to make use of the property 'moving' present in some devices. So I believe everything should be working as expected now.

ASNNetworks, 2023-01-09:
> In the first PR, the one which broke Xiaomi ZNJLBL01LM and similars, I changed the behavior so it used the state property when not moving. It looks like those devices report 'off' when no moving, [...]

Latest dev build 1.29.1-dev (commit #2288c0f) is still broken with ZNCLDJ11LM, but different behaviour than the previous dev build. It's actually now like the first 1.29 stable release. Exactly like this ticket I closed: https://github.com/Koenkk/zigbee2mqtt/issues/15926

This hotfix (https://github.com/Koenkk/zigbee2mqtt/issues/15873#issuecomment-1369579513) fixed it for me. Then later dev build broke it again, but in reverse (always showing closed). And now the latest dev build basically reverts it back to the bug like it was with the first 1.29 release.

Here are the logs, starting in opened position, then I closed it, waited for a few seconds and opened it again.

```
Debug 2023-01-09 20:08:33Received MQTT message on 'zigbee2mqtt/Kantoor Gordijnen/set' with data 'CLOSE'
Debug 2023-01-09 20:08:33Publishing 'set' 'state' to 'Kantoor Gordijnen'
Debug 2023-01-09 20:08:34Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440598,"presentValue":0}' from endpoint 1 with groupID 0
Info 2023-01-09 20:08:34MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:33+01:00","linkquality":112,"motor_state":null,"position":0,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
Info 2023-01-09 20:08:34MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:33+01:00","linkquality":116,"motor_state":null,"position":0,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
Debug 2023-01-09 20:08:35Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440598,"presentValue":99}' from endpoint 1 with groupID 0
Info 2023-01-09 20:08:35MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:35+01:00","linkquality":116,"motor_state":null,"position":99,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
Debug 2023-01-09 20:08:40Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440598,"presentValue":90}' from endpoint 1 with groupID 0
Info 2023-01-09 20:08:40MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:40+01:00","linkquality":116,"motor_state":null,"position":90,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
Debug 2023-01-09 20:08:43Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":0}' from endpoint 1 with groupID 0
Info 2023-01-09 20:08:43MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:42+01:00","linkquality":116,"motor_state":null,"position":0,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
Debug 2023-01-09 20:08:46Received MQTT message on 'zigbee2mqtt/Kantoor Gordijnen/set' with data 'OPEN'
Debug 2023-01-09 20:08:46Publishing 'set' 'state' to 'Kantoor Gordijnen'
Debug 2023-01-09 20:08:46Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440599,"presentValue":100}' from endpoint 1 with groupID 0
Info 2023-01-09 20:08:46MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:46+01:00","linkquality":112,"motor_state":null,"position":100,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
Info 2023-01-09 20:08:46MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:46+01:00","linkquality":116,"motor_state":null,"position":100,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
Debug 2023-01-09 20:08:50Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440599,"presentValue":0}' from endpoint 1 with groupID 0
Info 2023-01-09 20:08:50MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:50+01:00","linkquality":112,"motor_state":null,"position":0,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
Debug 2023-01-09 20:08:55Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":100}' from endpoint 1 with groupID 0
Info 2023-01-09 20:08:55MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-09T20:08:55+01:00","linkquality":109,"motor_state":null,"position":100,"power_outage_count":43,"running":null,"state":"OPEN","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
```

golles, 2023-01-09:
> @10Thirty @Xitro01 @ASNNetworks @golles @CMDR-Sloma @buhito81  can you check if your Xiaomi cover still works correctly in the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

For me looks good for this version:
```
info  2023-01-09 21:22:29: Starting Zigbee2MQTT version 1.29.1-dev (commit #d90dae5)
info  2023-01-09 21:22:29: Starting zigbee-herdsman (0.14.86)
```

Also `motor_state` and `running` have always worked for me:

```json
{
    "battery": 69,
    "charging_status": false,
    "device_temperature": 15,
    "illuminance": 0,
    "illuminance_lux": 0,
    "last_seen": "2023-01-09T20:27:36.350Z",
    "linkquality": 91,
    "motor_state": "pause",
    "position": 0,
    "power_outage_count": 1,
    "running": false,
    "state": "OFF",
    "temperature": 18,
    "update": {
        "installed_version": 3870,
        "latest_version": 3870,
        "state": "idle"
    },
    "update_available": false
}
```

ASNNetworks, 2023-01-09:
> > @buhito81 sorry, the changes were not available yet, can you check again in 1 hour from now?
> 
> ~I just updated the edge version and log says: 1.29.1-dev (commit #30df0e2). Can you tell me what the commit will be, so I'll know for sure if I have the latest build when I re-install and test again?~
> 
> edit:
> 
> re-installed and got 1.29.1-dev (commit #9c4abb6) now. My model (ZNCLDJ11LM) is now again broken unfortunately with this latest dev build. It's different broken than before the hotfix btw. It now is always set as closed in HA, while position attribute is working as intended. I attached logs as well. I logged when the curtain was in closed state. I opened the curtain, waited and then closed again.
> 
> ```
> Debug 2023-01-08 13:15:22Received MQTT message on 'zigbee2mqtt/Kantoor Gordijnen/set' with data 'OPEN'
> Debug 2023-01-08 13:15:22Publishing 'set' 'state' to 'Kantoor Gordijnen'
> Debug 2023-01-08 13:15:22Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440588,"presentValue":100}' from endpoint 1 with groupID 0
> Info 2023-01-08 13:15:22MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-08T13:15:22+01:00","linkquality":102,"motor_state":null,"position":100,"power_outage_count":43,"running":null,"state":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> Info 2023-01-08 13:15:22MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-08T13:15:22+01:00","linkquality":102,"motor_state":null,"position":100,"power_outage_count":43,"running":null,"state":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> Debug 2023-01-08 13:15:24Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440588,"presentValue":0}' from endpoint 1 with groupID 0
> Info 2023-01-08 13:15:24MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-08T13:15:24+01:00","linkquality":94,"motor_state":null,"position":0,"power_outage_count":43,"running":null,"state":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> Debug 2023-01-08 13:15:29Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440588,"presentValue":24}' from endpoint 1 with groupID 0
> Info 2023-01-08 13:15:29MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-08T13:15:29+01:00","linkquality":102,"motor_state":null,"position":24,"power_outage_count":43,"running":null,"state":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> Debug 2023-01-08 13:15:31Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":100}' from endpoint 1 with groupID 0
> Info 2023-01-08 13:15:31MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-08T13:15:31+01:00","linkquality":94,"motor_state":null,"position":100,"power_outage_count":43,"running":null,"state":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> Debug 2023-01-08 13:15:34Received MQTT message on 'zigbee2mqtt/Kantoor Gordijnen/set' with data 'CLOSE'
> Debug 2023-01-08 13:15:34Publishing 'set' 'state' to 'Kantoor Gordijnen'
> Debug 2023-01-08 13:15:34Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440592,"presentValue":0}' from endpoint 1 with groupID 0
> Info 2023-01-08 13:15:34MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-08T13:15:34+01:00","linkquality":102,"motor_state":null,"position":0,"power_outage_count":43,"running":null,"state":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> Info 2023-01-08 13:15:34MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-08T13:15:34+01:00","linkquality":102,"motor_state":null,"position":0,"power_outage_count":43,"running":null,"state":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> Debug 2023-01-08 13:15:34Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440592,"presentValue":99}' from endpoint 1 with groupID 0
> Info 2023-01-08 13:15:34MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":34,"last_seen":"2023-01-08T13:15:34+01:00","linkquality":102,"motor_state":null,"position":99,"power_outage_count":43,"running":null,"state":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> Debug 2023-01-08 13:15:43Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":0}' from endpoint 1 with groupID 0
[... cut ...] te":"STOP","update":{"installed_version":9,"latest_version":36,"state":"available"},"update_available":true}'
> ```
> 
> To clarify: latest stable build with the hotfix (1.29.1-1) did fix my model.

Zigbee2MQTT version 1.29.1-dev (commit #d90dae5) behaves exactly like above commit in my case. So always showing as closed now. Until now, the only working 1.29 build is the 1.29-1 hotfix release for me. All dev builds after that either always show as closed or as open, like the initial 1.29 release.

CMDR-Sloma, 2023-01-09:
> @delinend and @CMDR-Sloma, is it working on the latest dev branch? @Koenkk, are my changes still on that branch?

My blinds controllers are working now without any problems. I had this problem before with the one in the Living room:

```
{
    "battery": 98,
    "charging_status": false,
    "linkquality": 51,
    "position": 72,
    "update": {
        "installed_version": 3870,
        "latest_version": 3870,
        "state": "idle"
    },
    "update_available": false,
    "device_temperature": null,
    "motor_state": null,
    "running": null,
    "state": null
}
```

Also in the logs quite often I was seeing this:

> "Zigbee2MQTT:debug 2023-01-09 05:22:06: lumi.curtain.acn002: unknown key 13 with value 3870"
> "Zigbee2MQTT:debug 2023-01-09 05:23:41: lumi.curtain.acn002: unknown key 17 with value 1"

Both problems were solved by removal and re-pairing, but this controller started reporting temp, motor_state, running and state right away after this. (see my first and second post in this topic). The main issue was fixed by Zigbee2MQTT update, I'm on: 1.29.1-1

ASNNetworks, 2023-01-12:
I was wondering what the current status regarding the different behaviours with these models. The latest stable hotfix  1.29.1-1 fixed it for me (after the initial 1.29 broke it). But the latest 1.29 dev builds broke it again, so I'm trying to prevent that the next stable release will break my curtains if the dev changes make it to the next stable release.

Koenkk, 2023-01-22:
@ASNNetworks 

- what coordinator/firmware are you using?
- have you already tried re-pairing the device? 

As @cmiguelcabral already indicated, `motor_state` and `running` are `null` which should not happen.

ASNNetworks, 2023-01-22:
> @ASNNetworks
> 
> * what coordinator/firmware are you using?
> * have you already tried re-pairing the device?
> 
> As @cmiguelcabral already indicated, `motor_state` and `running` are `null` which should not happen.

Hi Koen, so I just updated everything just to make sure we're testing the same thing.

- Re-installed the Z2M dev branche and have now 1.29.2-dev (commit #f4f2c49).
- Also updated to the latest coördinator firmware (from the one before that) to 20221226.
- Removed my curtain motor and re-paired it with Z2M.

It still behaves the same with the latest dev build. State is shown always as open. while the position does change. And the `motor_state` and `running` are still null, which was always with this motor (have them for like 3 years now).

```
Debug 2023-01-22 12:43:49Received MQTT message on 'zigbee2mqtt/Kantoor Gordijnen/set' with data 'OPEN'
Debug 2023-01-22 12:43:49Publishing 'set' 'state' to 'Kantoor Gordijnen'
Debug 2023-01-22 12:43:49Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440645,"presentValue":100}' from endpoint 1 with groupID 0
Info 2023-01-22 12:43:49MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:43:51+01:00","linkquality":80,"motor_state":null,"position":100,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-01-22 12:43:49MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:43:51+01:00","linkquality":76,"motor_state":null,"position":100,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Debug 2023-01-22 12:43:54Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440645,"presentValue":0}' from endpoint 1 with groupID 0
Info 2023-01-22 12:43:54MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:43:56+01:00","linkquality":72,"motor_state":null,"position":0,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Debug 2023-01-22 12:43:59Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":100}' from endpoint 1 with groupID 0
Info 2023-01-22 12:43:59MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:44:00+01:00","linkquality":65,"motor_state":null,"position":100,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Debug 2023-01-22 12:44:01Received MQTT message on 'zigbee2mqtt/Kantoor Gordijnen/set' with data 'CLOSE'
Debug 2023-01-22 12:44:01Publishing 'set' 'state' to 'Kantoor Gordijnen'
Debug 2023-01-22 12:44:01Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440654,"presentValue":0}' from endpoint 1 with groupID 0
Info 2023-01-22 12:44:01MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:44:03+01:00","linkquality":69,"motor_state":null,"position":100,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-01-22 12:44:01MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:44:03+01:00","linkquality":69,"motor_state":null,"position":0,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Debug 2023-01-22 12:44:04Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440654,"presentValue":99}' from endpoint 1 with groupID 0
Info 2023-01-22 12:44:04MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:44:06+01:00","linkquality":69,"motor_state":null,"position":99,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Debug 2023-01-22 12:44:09Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440654,"presentValue":67}' from endpoint 1 with groupID 0
Info 2023-01-22 12:44:09MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:44:11+01:00","linkquality":83,"motor_state":null,"position":67,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Debug 2023-01-22 12:44:10Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":0}' from endpoint 1 with groupID 0
Info 2023-01-22 12:44:10MQTT publish: topic 'zigbee2mqtt/Kantoor Gordijnen', payload '{"device_temperature":33,"last_seen":"2023-01-22T12:44:12+01:00","linkquality":76,"motor_state":null,"position":0,"power_outage_count":44,"running":null,"state":"STOP","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
```

Koenkk, 2023-01-23:
It seems to work for the ZNCLDJ11LM from @Drafteed (based on https://github.com/Koenkk/zigbee-herdsman-converters/pull/4027), could you both provide the data/database.db entry of your device?

Drafteed, 2023-01-23:
> could you both provide the data/database.db entry of your device?

```json
{
   "id":23,
   "type":"Router",
   "ieeeAddr":"0x00158d0006d55493",
   "nwkAddr":40776,
   "manufId":4447,
   "manufName":"LUMI",
   "powerSource":"Mains (single phase)",
   "modelId":"lumi.curtain",
   "epList":[
      1
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":514,
         "inClusterList":[
            0,
            4,
            3,
            5,
            10,
            258,
            13,
            19,
            6,
            1,
            1030
         ],
         "outClusterList":[
            25,
            10,
            13,
            258,
            19,
            6,
            1,
            1030
         ],
         "clusters":{
            "genAnalogOutput":{
               "attributes":{
                  "61440":0,
                  "presentValue":100
               }
            },
            "genBasic":{
               "attributes":{
                  "1028":0,
                  "65281":{
                     "3":33,
                     "5":19,
                     "7":"0x0000000000000000",
                     "8":4388,
                     "9":1793,
                     "100":100
                  },
                  "modelId":"lumi.curtain",
                  "manufacturerName":"LUMI",
                  "powerSource":1,
                  "zclVersion":1,
                  "appVersion":1,
                  "stackVersion":2,
                  "hwVersion":17,
                  "dateCode":"01-25-2022"
               }
            },
            "closuresWindowCovering":{
               "attributes":{
                  "currentPositionLiftPercentage":100
               }
            }
         },
         "binds":[
            {
               "cluster":13,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b002291d5c2",
               "endpointID":1
            },
            {
               "cluster":0,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b002291d5c2",
               "endpointID":1
            }
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":34,
   "stackVersion":2,
   "hwVersion":17,
   "dateCode":"01-25-2022",
   "zclVersion":1,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1674499557170,
   "defaultSendRequestWhen":"immediate"
}
```

ASNNetworks, 2023-01-23:
> It seems to work for the ZNCLDJ11LM from @Drafteed (based on [Koenkk/zigbee-herdsman-converters#4027](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4027)), could you both provide the data/database.db entry of your device?

```
{"id":17,"type":"Router","ieeeAddr":"0x00158d00028350f7","nwkAddr":26091,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.curtain","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,4,3,5,10,258,13,19,6,1,1030],"outClusterList":[25,10,13,258,19,6,1,1030],"clusters":{"genBasic":{"attributes":{"65281":{"3":32,"5":47,"7":"0x0000000000000000","8":4361,"9":3587,"100":99},"modelId":"lumi.curtain","appVersion":1,"manufacturerName":"LUMI","powerSource":1,"zclVersion":1,"stackVersion":2,"hwVersion":17,"dateCode":"04-13-2017"}},"genAnalogOutput":{"attributes":{"61440":0,"presentValue":100}},"genOnOff":{"attributes":{"onOff":0}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":36}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002281a16d","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b002281a16d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":17,"dateCode":"04-13-2017","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1674499180595,"defaultSendRequestWhen":"immediate"}

```

@Drafteed yes this you are right, great find, this is the exact same issue. It was broken with 1.29.1, then the latest stable 
1.29.2 hotfix fixed it, but the current dev build broke it again and works exactly like the bug you linked to. So I am reporting here to prevent the next stable release to break it again.

Drafteed, 2023-01-23:
> "dateCode":"04-13-2017"

My firmware from 01-25-2022.
Are you still unable to update the firmware for the device via OTA?

ASNNetworks, 2023-01-23:
> > "dateCode":"04-13-2017"
> 
> My firmware from 01-25-2022. Are you still unable to update the firmware for the device via OTA?

It just so happened I tried last week for the last time. Once it reaches 100% it gives an error and stops sadly.

Drafteed, 2023-01-23:
Sorry for asking questions, I'm just trying to figure out what the problem is. 
I correctly understood this appeared after #16017?

Koenkk, 2023-01-23:
@cmiguelcabral  can you make an update to the changes of https://github.com/Koenkk/zigbee2mqtt/pull/16017 to make it robust against `null` values (as it was before your change)?

ASNNetworks, 2023-01-23:
> 

Yes I believe this was the dev build that broke it, after the stable hotfix that fixed the bug in 1.29.0-1 (if I still make sense)

Drafteed, 2023-01-23:
> Yes I believe this was the dev build that broke it, after the stable hotfix that fixed the bug in 1.29.0-1 (if I still make sense)

Once again, in stable 1.29.2 everything works ok?

cmiguelcabral, 2023-01-23:
> @ASNNetworks @cmiguelcabral Can you check this change #16324?

It is just the same thing in a different order.
But if it works, I would be happy to have it that way :-)

> @cmiguelcabral can you make an update to the changes of #16017 to make it robust against `null` values (as it was before your change)?

I don't think I understand what you mean, I did not change anything related to null values or other stuff, only added the same logic to use when there is a moving property...
But @Drafteed changes also seem ok, if it works that way, let's go with it!

ASNNetworks, 2023-01-23:
> @ASNNetworks @cmiguelcabral Can you check this change #16324?

Hi, I don't think the new PR is online yet. I just re-installed dev build, but it's different commit and still bugged.

cmiguelcabral, 2023-01-23:
> > @ASNNetworks @cmiguelcabral Can you check this change #16324?
> 
> Hi, I don't think the new PR is online yet. I just re-installed dev build, but it's different commit and still bugged.

You need to checkout the PR and build the image yourself.

Drafteed, 2023-01-23:
> Hi, I don't think the new PR is online yet. I just re-installed dev build, but it's different commit and still bugged.

You can checkout from my fork. Make sure you have Node v.14 or above installed.

```sh
git clone git@github.com:Drafteed/zigbee2mqtt.git z2m-test
cd z2m-test
git checkout ha-moving-fix
npm i
```
Copy your configuration to z2m-test/data/ and start z2m:

```sh
npm run start
```

cmiguelcabral, 2023-01-23:
@Drafteed, your changes work perfectly on my side!!
Much better than my solution. @Koenkk, you shoud go fot it!

ASNNetworks, 2023-01-23:
> > Hi, I don't think the new PR is online yet. I just re-installed dev build, but it's different commit and still bugged.
> 
> You can checkout from my fork. Make sure you have Node v.14 or above installed.
> 
> ```shell
> git clone git@github.com:Drafteed/zigbee2mqtt.git z2m-test
> cd z2m-test
> git checkout ha-moving-fix
> npm i
> ```
> 
> Copy your configuration to z2m-test/data/ and start z2m:
> 
> ```shell
> npm run start
> ```

I'm running Z2M add-on on Home Assistant OS. I don't think I can install this, since I'd need an add-on right?

Drafteed, 2023-01-23:
> I'm running Z2M add-on on Home Assistant OS. I don't think I can install this, since I'd need an add-on right?

Then let's just wait for the changes to the dev branch.

cmiguelcabral, 2023-01-23:
I find it really hard to understand how can this break those Xiaomi blinds.
The only thing that really changed was this:
````yaml
`{% if not value_json.${motorState.property} %}` +
                        ` ${stoppedState} {% else %} {{ value_json.${motorState.property} }} {% endif %}`;
````
For this:
````yaml
`{{ ${stoppedState} if not value_json.${movingState.property} ` +
                        `else value_json.${movingState.property} }}`;
````

But they should behave the same as the second one it's just an inline version of the former one.

Drafteed, 2023-01-23:
> I find it really hard to understand how can this break those Xiaomi blinds.
> The only thing that really changed was this:

Let`s try:

```jinja
{% set value_json = {"moving": None} %}

{% if not value_json.moving %} stopped {% else %} {{ value_json.moving }} {% endif %}
```

Result:

```
stopped 
```

Now, yours:

```jinja
{% set value_json = {"moving": None} %}

{{ stopped if not value_json.moving else value_json.moving }}
```

Result:
```
UndefinedError: 'stopped' is undefined
```

cmiguelcabral, 2023-01-23:
Ah... I see.
Missing quotes... dahhh

cmiguelcabral, 2023-01-23:
If you do:

````yaml
{% set value_json = {"moving": None} %}

{{ "stopped" if not value_json.moving else value_json.moving }}
````

Result:
````
stopped
````

Drafteed, 2023-01-23:
I know, but I preferred the previous option, since you don’t have to fence second-level quotes:

```js
discoveryEntry.discovery_payload.value_template =
                        `{{ "${stoppedState}" if not value_json.${movingState.property} ` +
                        `else value_json.${movingState.property} }}`;
```

```js
discoveryEntry.discovery_payload.value_template = `{% if not value_json.${motorState.property} %}` +
                        ` ${stoppedState} {% else %} {{ value_json.${motorState.property} }} {% endif %}`;
```

cmiguelcabral, 2023-01-23:
In my blinds the 'moving' property never gets 'null', so it never failed here.

Anyway, we don't need any of that if using your solution.
So, yours should be the one to make it to master.

Drafteed, 2023-01-23:
> Anyway, we don't need any of that if using your solution.
> So, yours should be the one to make it to master.

I think we need to add a test to catch this.

ASNNetworks, 2023-01-24:
> > I'm running Z2M add-on on Home Assistant OS. I don't think I can install this, since I'd need an add-on right?
> 
> Then let's just wait for the changes to the dev branch.

Is there a way to get a notification when the next dev branch has been released? I just re-installed again but is still on commit #7639ce4.

Drafteed, 2023-01-24:
@ASNNetworks Changes available in [latest-dev](https://hub.docker.com/layers/koenkk/zigbee2mqtt/latest-dev/images/sha256-771744d220235009191912ed8c9fef269b2a92a52b4f489fc05efdcf41870f3d?context=explore) build. Please test this.

ASNNetworks, 2023-01-25:
> @ASNNetworks Changes available in [latest-dev](https://hub.docker.com/layers/koenkk/zigbee2mqtt/latest-dev/images/sha256-771744d220235009191912ed8c9fef269b2a92a52b4f489fc05efdcf41870f3d?context=explore) build. Please test this.

Just installed commit #2f0a2ce, works great with my motor! 🙌 🫶

Aepx, 2023-06-15:
I just started using Z2M, on the latest version (1.31.2) and using the latest firmware available for these (10-09-2022).
My "state" never changes. All the controls work, open, stop and close, but the status of the state never changes.
Sounds like it might be the same as this thread since it's reporting "Off" and there is no "off" state.
Using a Sonoff zigbee 3.0 usb dongle plus zbdongle-p with the latest firmware available.
Sounds like this should have been fixed in 1.29.2.
Is anyone else seeing this with the Sonoff -p dongle?

Aepx, 2023-06-21:
No one else is still seeing this error?

[8 more replies]

PR #16324 (2023-01-24): Fix `motor_state` and `moving` in HA covers
This PR reverts #16017 to fix #15873 and add handle a `moving` state in `motorState`.