Koenkk/zigbee2mqtt #15882: [New device support]: YSR-MINI-01-rgb

eloo, 2023-01-02:
### Link

https://aliexpress.com/item/1005002295847611.html?spm=a2g0o.order_list.order_list_main.11.4d845c5foHEqo6&gatewayAdapt=glo2deu

### Database entry

{"id":35,"type":"Router","ieeeAddr":"0xa4c13841b98289bf","nwkAddr":45743,"manufId":4107,"manufName":"YSRSAI","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZB-CL01","manufacturerName":"eWeLight","powerSource":1,"zclVersion":3,"appVersion":6,"stackVersion":2,"hwVersion":0,"dateCode":"20221027","swBuildId":"2.0.1"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":6,"stackVersion":2,"hwVersion":0,"dateCode":"20221027","swBuildId":"2.0.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672485804900,"defaultSendRequestWhen":"immediate"}

### Comments

Similar to to the rgbcct there is also a only rgb controller.

Right now i'm also affected with the bug below.
https://github.com/Koenkk/zigbee2mqtt/issues/9665

So i have changed the manufName to YSRSAI but then nthe device is recognized as rgbcct.

Note: tge db entry was changed manually

eloo, 2023-02-02:
not stale

eloo, 2023-03-05:
Not stale

closed 2023-04-14 by github-actions[bot]

stalniy, 2024-01-29:
@eloo does it restore state after reconnection to mains power? I have the same and it doesn't restore state (color, brightness)

eloo, 2024-01-29:
@stalniy 
I dont know anymore.. i have the device not running anymore. Sorry