Koenkk/zigbee2mqtt #15883: Error message when adding Philips HUE Festavia (929003535301) into my network

FredericGuardia, 2023-01-02:
### What happened?

During initial interview of the device,  some error messages appear in the log file during the configuration step.
This error message appears several time inside log:
Failed to configure '0x001788010d629539', attempt 1 (TypeError: extendDontUse.light_onoff_brightness_colortemp_color(...).configure is not a function

### What did you expect to happen?

I expect to have device integrated and configured without any troubles during the interview phase.

### How to reproduce it (minimal and precise)

Integrate this device into your network and check zigbee2mqtt logs.

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20221102

### Adapter

SMLIGHT CC2652P Zigbee USB Adapter SLZB-02

### Debug log

info  2023-01-02 09:41:07: Device '0x001788010d629539' joined
info  2023-01-02 09:41:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d629539","ieee_address":"0x001788010d629539"},"type":"device_joined"}'
info  2023-01-02 09:41:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788010d629539"},"type":"device_connected"}'
info  2023-01-02 09:41:07: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539/availability', payload 'online'
info  2023-01-02 09:41:07: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"friendlyName":"0x001788010d629539","ieeeAddr":"0x001788010d629539","model":"unknown","networkAddress":2368,"type":"Unknown"},"last_seen":"2023-01-02T09:41:07+01:00"}'
info  2023-01-02 09:41:07: Starting interview of '0x001788010d629539'
info  2023-01-02 09:41:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d629539","ieee_address":"0x001788010d629539","status":"started"},"type":"device_interview"}'
info  2023-01-02 09:41:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788010d629539"},"type":"pairing"}'
info  2023-01-02 09:41:08: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"friendlyName":"0x001788010d629539","ieeeAddr":"0x001788010d629539","model":"unknown","networkAddress":2368,"type":"Unknown"},"last_seen":"2023-01-02T09:41:08+01:00"}'
info  2023-01-02 09:41:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d629539","ieee_address":"0x001788010d629539"},"type":"device_announce"}'
info  2023-01-02 09:41:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x001788010d629539"},"type":"device_announced"}'
info  2023-01-02 09:41:09: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"friendlyName":"0x001788010d629539","ieeeAddr":"0x001788010d629539","manufacturerID":4107,"model":"unknown","networkAddress":2368,"type":"Router"},"last_seen":"2023-01-02T09:41:09+01:00","linkquality":61}'
info  2023-01-02 09:41:09: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"friendlyName":"0x001788010d629539","ieeeAddr":"0x001788010d629539","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"unknown","networkAddress":2368,"type":"Router"},"last_seen":"2023-01-02T09:41:09+01:00","linkquality":105}'
info  2023-01-02 09:41:09: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"friendlyName":"0x001788010d629539","ieeeAddr":"0x001788010d629539","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"unknown","networkAddress":2368,"powerSource":"Mains (single phase)","type":"Router"},"last_seen":"2023-01-02T09:41:09+01:00","linkquality":105}'
info  2023-01-02 09:41:10: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"friendlyName":"0x001788010d629539","ieeeAddr":"0x001788010d629539","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"unknown","networkAddress":2368,"powerSource":"Mains (single phase)","type":"Router","zclVersion":8},"last_seen":"2023-01-02T09:41:10+01:00","linkquality":98}'
info  2023-01-02 09:41:10: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"applicationVersion":2,"friendlyName":"0x001788010d629539","ieeeAddr":"0x001788010d629539","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"unknown","networkAddress":2368,"powerSource":"Mains (single phase)","type":"Router","zclVersion":8},"last_seen":"2023-01-02T09:41:10+01:00","linkquality":102}'
info  2023-01-02 09:41:10: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"applicationVersion":2,"friendlyName":"0x001788010d629539","ieeeAddr":"0x001788010d629539","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"unknown","networkAddress":2368,"powerSource":"Mains (single phase)","stackVersion":1,"type":"Router","zclVersion":8},"last_seen":"2023-01-02T09:41:10+01:00","linkquality":98}'
info  2023-01-02 09:41:10: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"applicationVersion":2,"friendlyName":"0x001788010d629539","hardwareVersion":1,"ieeeAddr":"0x001788010d629539","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"unknown","networkAddress":2368,"powerSource":"Mains (single phase)","stackVersion":1,"type":"Router","zclVersion":8},"last_seen":"2023-01-02T09:41:10+01:00","linkquality":69}'
info  2023-01-02 09:41:10: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"applicationVersion":2,"dateCode":"20220928","friendlyName":"0x001788010d629539","hardwareVersion":1,"ieeeAddr":"0x001788010d629539","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"unknown","networkAddress":2368,"powerSource":"Mains (single phase)","stackVersion":1,"type":"Router","zclVersion":8},"last_seen":"2023-01-02T09:41:10+01:00","linkquality":43}'
info  2023-01-02 09:41:10: MQTT publish: topic 'zigbee2mqtt/0x001788010d629539', payload '{"device":{"applicationVersion":2,"dateCode":"20220928","friendlyName":"0x001788010d629539","hardwareVersion":1,"ieeeAddr":"0x001788010d629539","manufacturerID":4107,"manufacturerName":"Signify Netherlands B.V.","model":"unknown","networkAddress":2368,"powerSource":"Mains (single phase)","softwareBuildID":"1.101.1","stackVersion":1,"type":"Router","zclVersion":8},"last_seen":"2023-01-02T09:41:10+01:00","linkquality":58}'
info  2023-01-02 09:41:10: Successfully interviewed '0x001788010d629539', device has successfully been paired
info  2023-01-02 09:41:10: Device '0x001788010d629539' is supported, identified as: Philips Hue Festavia gradient light string 250 (929003535301)
info  2023-01-02 09:41:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Hue Festavia gradient light string 250","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":153},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":153},{"access":7,"description":"Color temperature after cold power on of this light","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":153},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max" [... cut ...]
    at Configure.configure (/app/lib/extension/configure.ts:117:37)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:32)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:140:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:118:27)
    at WebSocket.emit (node:events:513:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1178:20)
    at Receiver.emit (node:events:513:28))

zegl, 2023-01-02:
@FredericGuardia Could you please try upgrading to the latest version (v1.29.0) and try again?

FredericGuardia, 2023-01-02:
I am already on v1.29.0 version.

zegl, 2023-01-02:
Sorry, I see that now. I think I know what the issue is, I'm working on a fix!

closed 2023-01-03 by Koenkk