Koenkk/zigbee2mqtt #15885: [New device support]: Philips Hue White and Color Ambiance GU10 Xamento 3 spotlight pack

fireRoads, 2023-01-02:
### Link

https://www.philips-hue.com/sv-se/p/hue-white-och-color-ambiance-xamento-infalld-spot-3-pack/8719514355392

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0x001788010b55c9fe","nwkAddr":61279,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"929003074801_01","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64514,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003074801_01","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20200327","swBuildId":"1.65.11_hB798F2B"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":31248,"currentY":27106,"colorTemperature":400}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":134}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20200327","swBuildId":"1.65.11_hB798F2B","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672653452128,"defaultSendRequestWhen":"immediate"}

### Comments

This is a GU10 3 spotlight pack with extra hardware. The GU10 bulbs seems to work exactly like the regular GU10 color spotlights (929001953101). 

Model should probably be assigned to 929003074801 but each bulb in the pack has an ending with _01, _02 and _03. 

zigbeeModel: ['929003074801_01', '929003074801_02', '929003074801_03']

Below external converter works with 1.28.4-1.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extend.light_onoff_brightness(options),
        toZigbee: extend.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
        toZigbee: extend.light_onoff_brightness_colortemp(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extend.light_onoff_brightness_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_color({supportsHS: true, ...options}).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};

const definition = {
    zigbeeModel: ['929003074801_01', '929003074801_02', '929003074801_03'],
    model: '929001953101',
    vendor: 'Philips',
    description: 'Hue White and Color Ambiance GU10 (Xamento)',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    // extend: extend.light_onoff_brightness_colortemp_color(),
    extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    meta: {turnsOffAtBrightness1: true},
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

### Supported color modes

color temperature, color xy & color hs (rgb)

### Color temperature range

{colorTempRange: [153, 500]}

closed 2023-01-03 by Koenkk

Koenkk, 2023-01-29:
@Grandma-Betty 

> Or do I have to wait until the next Z2M release?

Will be supported in the 1 February release.

Koenkk, 2023-02-15:
@Grandma-Betty added them!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Koenkk, 2023-02-18:
This doesn't matter, it's just a cosmetic thing.

Koenkk, 2023-03-02:
I've fixed the issue now but it will be part of the next release. Until then you can use the following external converter:

```js
const philips = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['LCG002', '929003047701', '929003047701', '929003526202_01', '929003526202_02', '929003526202_03'],
    model: '929001953101',
    vendor: 'Philips',
    description: 'Hue White and Color Ambiance GU10',
    extend: philips.extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
};

module.exports = definition;

```
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m,

Koenkk, 2023-03-11:
added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Ruvetuve, 2025-01-30:
Hi, new serial numbers arrived with v2 resulting in unknown device when added to the network

929003812901_01 etc

Is this an easy one to add?