Koenkk/zigbee2mqtt #15887: Moes BRT-100-TRV - system mode "off" seem not work

Ivano62, 2023-01-02:
### What happened?

When I click on the "off" value nothing happens, as far as I can see it seems that the mqtt message with the correct value is not being sent. However, the state of the valve does not change and the HA climate remains unchanged.

### Zigbee2MQTT version

1.29.0-dev (commit: 826f032)

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

Ivano62, 2023-01-04:
The problem persists even with the latest dev version

Ivano62, 2023-01-06:
The problem persists even with the latest version 1.29.1

Ivano62, 2023-01-08:
The problem persists even with the latest version 1.29.1-dev commit: [e773511]

Ivano62, 2023-01-09:
The problem persists with the latest version 1.29.1-dev commit [d90dae5]

Ivano62, 2023-01-12:
problem persists with the latest version 1.29.1-dev commit [1d8b19d]

Ivano62, 2023-01-13:
Nothing, problem persists with the latest version 1.29.2

airestonia, 2023-01-19:
I can confirm that unfortunately this indeed doesn't work, even on the latest dev branch of the Zigbee2MQTT Edge.

Ivano62, 2023-01-20:
Problem persists with the latest dev version

rtsymbal, 2023-01-21:
I have same issue

svisso, 2023-01-22:
I have used couple of these TRVs up and until V 1.28.0. After upgrading to 1.28.1 and (today) 1.29.1, the Thermostats don't get read correctly, mostly returning null values to z2m. Unusable. I have to stick with 1.28.0 until rescue will be engineered.
Any help highly appreciated.

Ivano62, 2023-01-23:
I don't think it's a hardware problem, however with the latest dev versions the "off" value has been removed. Thanks

closed 2023-03-03 by github-actions[bot]