Koenkk/zigbee2mqtt #15888: zlinky lixee: missing functionalities

foormea, 2023-01-02:
### What happened?

At least the "Linky acquisition time" feature is missing: https://github.com/fairecasoimeme/Zlinky_TIC#linky-acquisition-time-from-v7

I'm at a loss as to how to implement this in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js: all attribute addresses as just comments, therefore I don't know how to add a new attribute in there. I've tried to compare with other js def files but so far I haven't figured things out.

Could I please have your guidance here?

Thank you.

### What did you expect to happen?

I want to be able to adjust the "linky acquisition time" in my device's specific settings, or to view/adjust it using the dev console.

### How to reproduce it (minimal and precise)

"Linky acquisition time" is not listed in my device's specific settings and is not listed in the dev console.

### Zigbee2MQTT version

1.28.0 (docker image koenkk/zigbee2mqtt)

### Adapter firmware version

20220219

### Adapter

Zigbee 3.0 USB Dongle Plus

mwi001, 2023-01-15:
Hello,

I have te same issue here and I would be interested in having the ability to set the linky acquisition time value. By default it is 10 seconds, which is a bit too long to implement a software load shedding function using Home Assistant (for instant put warming in away mode when power reaches a certain threshold). 

Thanks 

Matthieu

foormea, 2023-01-15:
The 0xFF66 / 0x0100 attribute is defined [here](https://github.com/fairecasoimeme/Zlinky_TIC/blob/master/ZLinky/Source/LixeeCluster.h#L35) and is named `E_CLD_LINKY_PERIODIC_SEND`.

Trying to reference that in the [lixee.js file](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js).

Still trying to figure out how this works.

foormea, 2023-02-15:
Would be great if this could be addressed at some point :) Thank you.

foormea, 2023-03-19:
.

milouz1985, 2023-04-22:
Same problem.
I use ZHA to set the value, but would be greate to be able to do it in Z2M.

milouz1985, 2023-05-24:
Comment to prevent from closing.

milouz1985, 2023-11-01:
could you consider reopening this issue?

lpaso, 2023-11-16:
@Koenkk : thanks !

milouz1985, 2024-12-11:
@Koenkk like OP I tried to understand how attributes names and hexadecimal addresses are linked to try to implement this myself, but it's quite obscure to me.
Maybe you could help us to help you?

closed 2025-02-17 by github-actions[bot]