Koenkk/zigbee2mqtt #15889: Gledopto GL-C-008-2ID Wrong Color_Temp

meditation1985, 2023-01-02:
### What happened?

 When i set the maximum color_temp in Zigbee2MQTT to 500 , The light is change to coolest.
 When i set the minimum color_temp in Zigbee2MQTT to 150 , The light is change to warmest.

### What did you expect to happen?

 When i set the maximum color_temp in Zigbee2MQTT to 500 , The light is change to warmest.
 When i set the minimum color_temp in Zigbee2MQTT to 150 , The light is change to coolest.

### Zigbee2MQTT version

 1.28.4

### Adapter firmware version

20210813

### Adapter

zStack3x0

Koenkk, 2023-01-03:
Try swapping the white and warm wires, for some of these devices the marking on the enclosure is wrong.

meditation1985, 2023-01-03:
you are correct. Enclosure is wrong.

closed 2023-01-03 by meditation1985