Koenkk/zigbee2mqtt #15891: [Feature request]: Expose current Philips Hue Effect

zegl, 2023-01-02:
### Is your feature request related to a problem? Please describe

#15699 introduced the ability to set effects on Philips Hue lights, and https://github.com/Koenkk/zigbee-herdsman-converters/pull/5227 added support to decode the current effect used on the light.

I propose to add an `effect` fromZigbee converter for the supported bulbs, that exposes the current effect used (if any).

### Describe the solution you'd like

I propose to adopt the current fromZigbee.gradient parser to work will all types of Philips Hue bulbs that have the `0xFC03` / `0x0002` attribute (seems to be all bulbs, when upgraded to the latest firmware).

Something like this (might need some tweaks to only expose the gradient and/or effect if the bulb supports setting it):

```diff
diff --git a/devices/philips.js b/devices/philips.js
index f46fa263..0776a72d 100644
--- a/devices/philips.js
+++ b/devices/philips.js
@@ -236,16 +236,26 @@ const fzLocal = {
             return payload;
         },
     },
-    gradient: (opts = {reverse: false}) => {
+    philipsState: (opts = {reverse: false}) => {
         return {
             cluster: 'manuSpecificPhilips2',
             type: ['attributeReport', 'readResponse'],
             convert: (model, msg, publish, options, meta) => {
                 if (msg.data && msg.data.hasOwnProperty('state')) {
                     const input = msg.data['state'].toString('hex');
-                    const decoded = philips.decodeGradientColors(input, opts);
+                    const decoded = philips.decodeState(input, opts);
+
+                    const ret = {
+                        gradient: [],
+                        effect: null,
+                    };
+
                     if (decoded.color_mode === 'gradient') {
-                        return {gradient: decoded.colors};
+                        ret.gradient = decoded.colors;
+                    }
+
+                    if (decoded.color_mode === 'xy' && decoded.name !== undefined) {
+                        ret.effect = decoded.name;
                     }
                 }
                 return {};
```

### Describe alternatives you've considered

😎

### Additional context

😎

Koenkk, 2023-01-03:
Sounds good! The `effect` expose `access` should become `access.STATE_SET` then.

closed 2023-02-10 by github-actions[bot]

jonrupell, 2025-06-11:
Was this ever implemented? All my hue bulbs in z2m still show null no matter if an effect is being played or not