Koenkk/zigbee2mqtt #15892: Zigbee Frontend not loading

coralseas, 2023-01-02:
### What happened?

Error 404 / not loading with Chrome browser on Windows 10 and Windows 7, 64b and 32b. Working with Windows 11 and with Firefox on all Windows versions. 
Older Z2M versions had no issues.

### How to reproduce it (minimal and precise)

Just trying to open with http://IPAdr:8080 on Chrome

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

2022

### Adapter

Electrolama

### Debug log

console window at Chrome shows "error 404"

lux73, 2023-01-02:
https://github.com/Koenkk/zigbee2mqtt/issues/15856

DavidDeSloovere, 2023-01-23:
I tried Firefox (108.0.2) on Win11 (22H2) with Z2M 1.29.2 and it works for me.
Did you set and `auth_token` in the config?

[image]

closed 2023-03-03 by github-actions[bot]