Koenkk/zigbee2mqtt #15896: [New device support]:

jacekowski, 2023-01-02:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-filament-1-pack-g125-b22-filament-globe/8719514279155

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x0017880109049648","nwkAddr":15903,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LWO004","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWO004","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190425","swBuildId":"1.55.8_r28815"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190425","swBuildId":"1.55.8_r28815","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672614144135,"defaultSendRequestWhen":"immediate"}

### Comments

Looks like a standard hue bulb. Worked with very simple external converter until #15858 broke external converters.

Koenkk, 2023-01-03:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Koenkk, 2023-01-03:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/5256 thanks to @zegl 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-03 by Koenkk