Koenkk/zigbee2mqtt #15897: Cannot bind switches to Hue lamp groups

brantmacga, 2023-01-02:
### What happened?

Continuation of a discussion I started here while adding support for a new device: https://github.com/Koenkk/zigbee2mqtt/issues/15846

I now have my Legrand/Netatmo switch working in Z2M. I can bind it to single/multiple hue lamps, but I cannot bind it to a group. I have tried adding the switch to the Group, but I get a timeout error. 

When I go to the Dimmer device page and bind the on/off and level control, I get a confirmation that both clusters were bound successfully to the group. But it doesn't actually work. 

The issue with multiple lamp binding vs. group binding is the timing delay between lamps when toggle off/on and dimming functions. 

I noticed the same issue with the Legrand 067694 Toggle Switch. 

### What did you expect to happen?

I expected to be able to bind this switch to a group of Hue lamps. 

### How to reproduce it (minimal and precise)

Attempt to bind it, see success message, then find that it doesn't actually work. 

### Zigbee2MQTT version

1.29.0-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

brantmacga, 2023-01-02:
I noticed this in the log:

> 2022-12-31 17:37:36.760 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.pool_bathroom_light_4_2
2022-12-31 17:37:36.911 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.pool_bathroom_light_4_2
2022-12-31 17:37:36.927 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.pool_bathroom_light_2_2
2022-12-31 17:37:37.029 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.pool_bathroom_light_2_2
2022-12-31 17:37:37.112 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received for entity light.pool_bathroom_light_1_2
2022-12-31 17:37:37.374 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json]

These are Hue ambient white lamps in the group I'm trying to bind. 

I saw this line at the end. 

> (JSON) payload detected after processing payload '1.93.11' on topic zigbee2mqtt/Pool Bathroom Light 1
2023-01-02 14:05:24.176 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"brightness":249,"color_mode":"color_temp","color_temp":454,"linkquality":148,"power_on_behavior":"previous","state":"ON","update":{"installed_version":"1.93.11","state":"available"},"update_available":null}' with variables {'entity_id': 'update.pool_bathroom_light_1', 'name': 'Pool Bathroom Light 1', 'this': <template TemplateStateFromEntityId(update.pool_bathroom_light_1)>} and Template("{{ value_json['update']['latest_version'] }}")
2023-01-02 14:05:24.176 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'

closed 2023-03-13 by github-actions[bot]

celestianx, 2023-03-20:
Hi I observe the same issue.
I managed to add once a power outlet but all others are not