Koenkk/zigbee2mqtt #15898: Timeout

Bastiencc, 2023-01-02:
### What happened?

Hi

I got some timeout with several devices, how can I fixe that ?

That not always the same error message : 
`Timeout - 46847 - 1 - 193 - 8 - 11 after 10000ms`
`SRSP - AF - dataRequest after 6000ms`

### What did you expect to happen?

Do not have timeout

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

20220219

### Adapter

zzh

### Debug log

info  2023-01-02 18:20:35: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":62.5,"brightness":255,"counter":1,"duration":0,"linkquality":131,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-02 18:20:35: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"","action_duration":null,"battery":62.5,"brightness":255,"counter":1,"linkquality":131,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-02 18:20:35: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur/action', payload 'on-press'
info  2023-01-02 18:20:35: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":87,"power_on_behavior":null,"power_outage_memory":true,"state":"ON","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:20:35: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":83,"power_on_behavior":null,"power_outage_memory":true,"state":"ON","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:20:36: MQTT publish: topic 'zigbee2mqtt/bureau_move', payload '{"battery":null,"battery_low":false,"keep_time":30,"linkquality":40,"occupancy":true,"sensitivity":"low","tamper":false,"voltage":null}'
info  2023-01-02 18:20:43: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":62.5,"brightness":255,"counter":1,"duration":0,"linkquality":131,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-02 18:20:43: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"","action_duration":null,"battery":62.5,"brightness":255,"counter":1,"linkquality":131,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-02 18:20:43: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur/action', payload 'on-press'
info  2023-01-02 18:20:50: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":62.5,"brightness":255,"counter":1,"duration":0,"linkquality":131,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-02 18:20:50: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"","action_duration":null,"battery":62.5,"brightness":255,"counter":1,"linkquality":131,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-02 18:20:50: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur/action', payload 'on-press'
info  2023-01-02 18:20:55: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":62.5,"brightness":255,"counter":1,"duration":0,"linkquality":131,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-02 18:20:55: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"","action_duration":null,"battery":62.5,"brightness":255,"counter":1,"linkquality":131,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-02 18:20:55: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur/action', payload 'on-press'
error 2023-01-02 18:20:58: Publish 'set' 'brightness' to 'bureau_plafond' failed: 'Error: Command 0x00158d0002d3c697/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 46847 - 1 - 193 - 8 - 11 after 10000ms)'
info  2023-01-02 18:20:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'bureau_plafond' failed: 'Error: Command 0x00158d0002d3c697/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 46847 - 1 - 193 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"bureau_plafond"},"type":"zigbee_publish_error"}'
info  2023-01-02 18:20:58: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":87,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:20:58: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":91,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:20:58: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":87,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:20:58: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":87,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:20:58: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":83,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:20:58: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":83,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:20:58: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":87,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'
info  2023-01-02 18:21:03: MQTT publish: topic 'zigbee2mqtt/salon_capteur', payload '{"battery":100,"humidity":58.47,"linkquality":138,"power_outage_count":34289,"pressure":1004.4,"temperature":20.51,"voltage":3005}'
error 2023-01-02 18:21:04: Publish 'set' 'color_temp' to 'bureau_plafond' failed: 'Error: Command 0x00158d0002d3c697/1 lightingColorCtrl.moveToColorTemp({"colortemp":338,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
info  2023-01-02 18:21:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color_temp' to 'bureau_plafond' failed: 'Error: Command 0x00158d0002d3c697/1 lightingColorCtrl.moveToColorTemp({\"colortemp\":338,\"transtime\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disabl [... cut ...]
info  2023-01-02 18:21:15: MQTT publish: topic 'zigbee2mqtt/bureau_move', payload '{"battery":null,"battery_low":false,"keep_time":30,"linkquality":54,"occupancy":false,"sensitivity":"low","tamper":false,"voltage":null}'
info  2023-01-02 18:21:37: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":235,"color_mode":"color_temp","color_temp":338,"linkquality":80,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":"1.32\u0000","state":"idle"},"update_available":false}'

Koenkk, 2023-01-03:
`Timeout - 46847 - 1 - 193 - 8 - 11 after 10000ms SRSP - AF - dataRequest after 6000ms` -> means the adapter disconnected. First thing to do is to try with this fw: https://github.com/Koenkk/Z-Stack-firmware/tree/6.10.01.01/coordinator/Z-Stack_3.x.0/bin

Bastiencc, 2023-01-03:
Ok thanks, I'll try. Have a nice day !

Bastiencc, 2023-01-03:
@Koenkk I still get the same issue with this fw

Koenkk, 2023-01-04:
What is your setup? Are you running HA? If yes, make sure ZHA is not enabled.

Bastiencc, 2023-01-05:
I use a rpi 3b+ with HA 2022.12.9 (HassOs).
ZHA is not enabled

Bastiencc, 2023-01-05:
@Koenkk I use a rpi 3b+ with HA 2022.12.9 (HassOs).
ZHA is not enabled

Koenkk, 2023-01-05:
Can you provide the herdsman debug logging of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Bastiencc, 2023-01-06:
@Koenkk here it is. I hope this log is ok for you. Thank you very much for your help.

`info  2023-01-06 08:41:59: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"on-press","action_duration":null,"battery":68.5,"brightness":255,"counter":1,"duration":0,"linkquality":149,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-06 08:41:59: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur', payload '{"action":"","action_duration":null,"battery":68.5,"brightness":255,"counter":1,"linkquality":149,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":false}'
info  2023-01-06 08:41:59: MQTT publish: topic 'zigbee2mqtt/bureau_switch_mur/action', payload 'on-press'
error 2023-01-06 08:42:19: Publish 'set' 'brightness' to 'bureau_plafond' failed: 'Error: Command 0x00158d0002d3c697/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 46847 - 1 - 58 - 8 - 11 after 10000ms)'
info  2023-01-06 08:42:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'bureau_plafond' failed: 'Error: Command 0x00158d0002d3c697/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":0}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 46847 - 1 - 58 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"bureau_plafond"},"type":"zigbee_publish_error"}'
info  2023-01-06 08:42:19: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":76,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":32,"latest_version":34,"state":"idle"},"update_available":false}'
info  2023-01-06 08:42:19: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":76,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":32,"latest_version":34,"state":"idle"},"update_available":false}'
info  2023-01-06 08:42:19: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":76,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":32,"latest_version":34,"state":"idle"},"update_available":false}'
info  2023-01-06 08:42:19: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":76,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":32,"latest_version":34,"state":"idle"},"update_available":false}'
info  2023-01-06 08:42:19: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":76,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":32,"latest_version":34,"state":"idle"},"update_available":false}'
`

`2023-01-06T07:42:19.307Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2023-01-06T07:42:19.307Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2023-01-06T07:42:19.308Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2023-01-06 08:42:19: MQTT publish: topic 'zigbee2mqtt/bureau_plafond', payload '{"brightness":254,"color_mode":"color_temp","color_temp":338,"linkquality":76,"power_on_behavior":null,"power_outage_memory":true,"state":"OFF","update":{"installed_version":32,"latest_version":34,"state":"idle"},"update_available":false}'
2023-01-06T07:42:19.324Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,107,173]
2023-01-06T07:42:19.324Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,107,173]
2023-01-06T07:42:19.324Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,107] - 173
2023-01-06T07:42:19.325Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":107}
2023-01-06T07:42:19.325Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-01-06T07:42:19.357Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,35,68,129,0,0,0,3,255,182,1,1,0,76,0,188,185,0,0,0,15,24,247,10,7,0,33,82,1,0,240,35]
2023-01-06T07:42:19.358Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,35,68,129,0,0,0,3,255,182,1,1,0,76,0,188,185,0,0,0,15,24,247,10,7,0,33,82,1,0,240,35]
2023-01-06T07:42:19.361Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [60,0,0,7,255,182,29,198,254,25,68,129,0,0,0,3,255,182,1,1,0,76,0,180,188,0,0,0,5,24,60,11]
2023-01-06T07:42:19.362Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,35,68,129,0,0,0,3,255,182,1,1,0,76,0,188,185,0,0,0,15,24,247,10,7,0,33,82,1,0,240,35,60,0,0,7,255,182,29,198,254,25,68,129,0,0,0,3,255,182,1,1,0,76,0,180,188,0 [... cut ...]
2023-01-06T07:42:26.700Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,109,171]
2023-01-06T07:42:26.701Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,109,171]
2023-01-06T07:42:26.701Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,109] - 171
2023-01-06T07:42:26.701Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":109}
2023-01-06T07:42:26.702Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []`

Koenkk, 2023-01-06:
I'm especially interested in:

```info 2023-01-02 18:21:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color_temp' to 'bureau_plafond' failed: 'Error: Command 0x00158d0002d3c697/1 lightingColorCtrl.moveToColorTemp({"colortemp":338,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":```

How are you running z2m? Docker/Ha addon?

Bastiencc, 2023-01-06:
HassOs 
Raspberry pi3b+
z2m addon 
coordinator is zzh

Home Assistant 2022.12.9
Supervisor 2022.12.1
Operating System 9.4
Interface utilisateur : 20221213.1 - latest

Koenkk, 2023-01-06:
Then your problem is likely the same as https://github.com/Koenkk/zigbee2mqtt/issues/15856, I'll continue investigating there.

Bastiencc, 2023-01-06:
Thanks @Koenkk I will follow this thread.

closed 2023-01-18 by Bastiencc