Koenkk/zigbee2mqtt #15899: Moes BHT-002-GCLZB: Setting decimal values for local_temperature_calibration fails

bjwo, 2023-01-02:
### What happened?

I have the device [Moes BHT-002-GCLZB](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html).

I tried to set the _local_temperature_calibration_ using the Z2M config UI to -0.2. However, the UI only accepts integer values  like "2", "-1" but not any decimal values.

### What did you expect to happen?

I expect that I can set decimal values.

### How to reproduce it (minimal and precise)

Please find the attached screen capture.
[image]

### Zigbee2MQTT version

 1.29.0

### Adapter firmware version

20201127

### Adapter

TI CC2531

bjwo, 2023-02-07:
This issue is not yet resolved. There is a workaround for the "Better Thermostat" custom integration for home assistant: https://github.com/KartoffelToby/better_thermostat/pull/897

closed 2023-03-19 by github-actions[bot]

SSG1111, 2023-07-21:
I'm using local_temperature_calibration for continues temperature adjustment (from separate sensor), because this thermostat heats itself up to +5..7deg when the heating is on, which has impact on the heating control. Would be really nice to have possibility to set offset at least with +/- 0.5 resolution.

almazikv, 2023-12-07:
also interested in calibration precision with 0.5 accuracy