Koenkk/zigbee2mqtt #15900: Innr SP 220 is pairing correctly but not switching on or off

stijnder, 2023-01-02:
### What happened?

I'm running the docker image of 1.28.4 combined with a docker image of eclipse-mosquitto and a Conbee 2 USB adapter. My config is 
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mosquitto
  user: mqtt
  password: ***
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
serial:
  port: /dev/ttyACM0
  adapter: deconz
advanced:
  network_key:
    - ***
  log_level: debug
frontend: true
```

I added the availability config recently. I've got three Innr SP 220 switches in my network. Pairing works without a flaw, but all of them show as offline regularly.  The log shows: `Failed to ping 'Smartplug hallway' (attempt 1/1, Read 0xc4988600000f341b/0 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received))`.

The state is updated correctly when I press the on/off button on the switch. But when I try to change the state in zigbee2mqtt (or HA), the state doesn't change in the application, or on the device itself. The log shows:
```
Error 2023-01-02 22:35:53Publish 'set' 'state' to 'Smartplug hallway' failed: 'Error: Command 0xc4988600000f341b/0 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Info 2023-01-02 22:35:53MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Smartplug hallway' failed: 'Error: Command 0xc4988600000f341b/0 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"Smartplug hallway"},"type":"zigbee_publish_error"}'
Error 2023-01-02 22:35:54Publish 'set' 'state' to 'Smartplug hallway' failed: 'Error: Command 0xc4988600000f341b/0 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Info 2023-01-02 22:35:54MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Smartplug hallway' failed: 'Error: Command 0xc4988600000f341b/0 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"Smartplug hallway"},"type":"zigbee_publish_error"}'
```

### What did you expect to happen?

I expected to be able to change state of the smart plug from zigbee2mqtt or HA.

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

0x26720700

### Adapter

ConBee2

shrynx, 2023-01-05:
I am having the same issue aswell

```js
'Error: Command 0x18fc2600000ae50c/0 genOnOff.off({}, {"send When":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
```

closed 2023-02-12 by github-actions[bot]

omixximo, 2023-02-19:
I have same problem, thay are working but from time to time if they lose electricity or restart my host can make them stop working.

Wibias, 2023-08-04:
Yup having the same problem.

mos3sman, 2023-11-13:
I'm having the same problem.

mdp08, 2024-03-02:
same problem here. with all innr 220 

`Publish 'set' 'state' to 'EG - Esszimmer Vitrine Klein ' failed: 'Error: Command 0x18fc260000014106/0 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (200))'`

wolfhard84, 2024-03-11:
hello, I wanted to join in briefly to report that I have the exact same problem :-> Pairing was fine, function after paring also. Problem started after Rebooting zigbee2mqtt.