Koenkk/zigbee2mqtt #15902: TS110E

gras55, 2023-01-03:
### What happened?

Turning on and off the light does not update the front end

### What did you expect to happen?

I expect the physical switch to update the front end

### How to reproduce it (minimal and precise)

Turn the light on or off

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20221102

### Adapter

Sonoff c2652p

gras55, 2023-01-03:
Am assuming this has to do with some change on this https://github.com/Koenkk/zigbee2mqtt/issues/15120 
I noticed that when turning on and off the switch it triggers the following in “state_1”

"brightness": 173,
    "brightness_1": 174,
    "linkquality": 105,
    "state": "ON",
    "state_1": "OFF",
    "power_on_behavior": null

before this update it triggered it in “state”

Koenkk, 2023-01-03:
can you provide the exact model number as shown in the z2m frontend?

gras55, 2023-01-03:
Zigbee Model
TS110E
Zigbee Manufacturer
_TZ3210_zxbtub8r

Koenkk, 2023-01-04:
Found the issue and fixed it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

gras55, 2023-01-06:
I’m afraid still not fixed in latest 1.29.1 release! Reverted to 1.28.4.

emimor123, 2023-01-06:
Confirmed, still not fixed, even on/off control stop working, only work dimmer and external switch

zollek, 2023-01-07:
After update I had the same issue but I can confirm, that it has been fixed in the dev branch.

Model: TS110E
Manufacturer:_TZ3210_zxbtub8r
Zigbee2MQTT Current version: edge

When we can expect the fix to be implemented in the main branch ?

Koenkk, 2023-01-08:
@gras55 @emimor123 can you check if this works in the latest -dev?

@zollek next release is on 1 feb

emimor123, 2023-01-08:
> @gras55 @emimor123 can you check if this works in the latest -dev?
> 
> @zollek next release is on 1 feb

I'm sorry, I wish I could try it, but I'm very new in this world and I don't know how to do it.

emimor123, 2023-01-08:
> @gras55 @emimor123 can you check if this works in the latest -dev?
> 
> @zollek next release is on 1 feb

Sorry, it doesn't work well, I think the problem we have is the following, knowing that the structure is:

     "brightness": 0,
     "brightness_1": 254,
     "color_mode": "xy",
     "linkquality": 142,
     "max_brightness": 255,
     "min_brightness": 4,
     "power_on_behavior": "off",
     "power_on_behavior_1": "off",
     "state": "OFF",
     "state_1": "OFF",
     "switch_type": "toggle",
     "switch_type_1": "toggle"

     "state": corresponds to the state of the device, ie. on or off
     "state_1" corresponds to the state of the light, ie. light on or light off.

that is, if "state" is off the device is off and there is no effect when pressing the wall switch.

I think the solution would be to leave "state" hidden in the ON state and have both the switch in HA and the wall switch act on "state1".

zollek, 2023-01-08:
@emimor123 What model of the device do You have?

In mine ([_Model:TS110E_](https://www.zigbee2mqtt.io/devices/TS110E_1gang_1.html#tuya-ts110e_1gang_1)) there is no _color_mode_ to be set:
```   "brightness": 213,
    "brightness_1": 118,
    "linkquality": 57,
    "max_brightness": 255,
    "min_brightness": 4,
    "power_on_behavior": "off",
    "power_on_behavior_1": "off",
    "state": "OFF",
    "state_1": "ON",
    "switch_type": "momentary",
    "switch_type_1": "momentary"
```

For me, when I used Zigbee2MQTT 1.29.1-1 version, physical switch coresponded to a _state_1_.

After moving to the newest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), both HA and physical switch uses the same: _"state": "ON/OFF"_
_state_1_ is always on and don't represent anything ( I can assume that all _ xxxxxx_1 _ would dispear after removing and adding the device once again).

emimor123, 2023-01-08:
> 

Hi, This is my model

Modelo Zigbee TS110E
Fabricante Zigbee _TZ3210_k1msuvg6
Descripción 1 channel dimmer

I remove and add again device and now this is the list of states:

    "brightness": 252,
    "brightness_1": 253,
    "linkquality": 116,
    "max_brightness": 255,
    "min_brightness": 4,
    "power_on_behavior": "off",
    "power_on_behavior_1": "off",
    "state": "OFF",
    "state_1": "OFF",
    "switch_type": "toggle",
    "switch_type_1": "toggle"

color mode disapear, but for me, "state_1" is always OFF, now works better, only one bug. If I switch on in HA the light increasing brightness (dont touching HA switch) then have to press 2 times phisical switch to switch off light, everything else works fine.

zollek, 2023-01-08:
Physical switch should work on a dimmer independetly from HA, the problem here was no frontend state updates.

> color mode disapear, but for me, "state_1" is always OFF, now works better, only one bug. If I switch on in HA the light increasing brightness (dont touching HA switch) then have to press 2 times phisical switch to switch off light, everything else works fine.

I admit it's the same for me. 
It's like a dimmer itself has different ON circuits/states (one for zigbee and one for physical switch) and while HA state change changes them both to ON, changing brightness only turns ON zigbee one.

gras55, 2023-01-08:
So this is still not fixed entirely?

Koenkk, 2023-01-09:
@gras55 can you provide the debug log when physically turning of the switch?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

gras55, 2023-01-10:
**This is the log when turning the switch on and off (works as it should both physically and in the frontend)**

Debug 2023-01-10 10:50:41Received Zigbee message from 'Middle Parents Bathroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-01-10 10:50:41MQTT publish: topic 'zigbee2mqtt/Middle Parents Bathroom', payload '{"brightness":167,"brightness_1":223,"linkquality":54,"power_on_behavior":null,"state":"ON","state_1":"OFF"}'
Debug 2023-01-10 10:50:43Received Zigbee message from 'Middle Parents Bathroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-01-10 10:50:43MQTT publish: topic 'zigbee2mqtt/Middle Parents Bathroom', payload '{"brightness":167,"brightness_1":223,"linkquality":54,"power_on_behavior":null,"state":"OFF","state_1":"OFF"}'

**This is the log when turning on dimmer in the frontend by using the brightness slider it then requires twice pressing physical switch to turn off**

Debug 2023-01-10 10:53:20Received MQTT message on 'zigbee2mqtt/Middle Parents Bathroom/set' with data '{"state":"ON","brightness":175}'
Debug 2023-01-10 10:53:20Publishing 'set' 'brightness' to 'Middle Parents Bathroom'
Info 2023-01-10 10:53:20MQTT publish: topic 'zigbee2mqtt/Middle Parents Bathroom', payload '{"brightness":175,"brightness_1":223,"linkquality":51,"power_on_behavior":null,"state":"ON","state_1":"OFF"}'
Debug 2023-01-10 10:53:21Received Zigbee message from 'Middle Parents Bathroom', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":175}' from endpoint 1 with groupID 0
Info 2023-01-10 10:53:21MQTT publish: topic 'zigbee2mqtt/Middle Parents Bathroom', payload '{"brightness":175,"brightness_1":223,"linkquality":51,"power_on_behavior":null,"state":"ON","state_1":"OFF"}'
Debug 2023-01-10 10:53:23Received Zigbee message from 'Middle Parents Bathroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-01-10 10:53:23MQTT publish: topic 'zigbee2mqtt/Middle Parents Bathroom', payload '{"brightness":175,"brightness_1":223,"linkquality":51,"power_on_behavior":null,"state":"ON","state_1":"OFF"}'
Debug 2023-01-10 10:53:24Received Zigbee message from 'Middle Parents Bathroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-01-10 10:53:24MQTT publish: topic 'zigbee2mqtt/Middle Parents Bathroom', payload '{"brightness":175,"brightness_1":223,"linkquality":51,"power_on_behavior":null,"state":"OFF","state_1":"OFF"}'

Koenkk, 2023-01-11:
Sorry, I needed the herdsman debug logging, can you provide it with the same sequence as above?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

gras55, 2023-01-12:
```
Zigbee2MQTT:debug 2023-01-12 09:36:03: Received Zigbee message from 'Ground Cloakroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-01-12 09:36:03: MQTT publish: topic 'zigbee2mqtt/Ground Cloakroom', payload '{"brightness":255,"linkquality":87,"power_on_behavior":null,"state":"OFF"}'
```

```
Zigbee2MQTT:debug 2023-01-12 09:37:32: Received Zigbee message from 'Ground Cloakroom', type 'attributeReport', cluster 'genBasic', data '{"65506":37,"65508":0,"appVersion":65}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-01-12 09:38:39: Received MQTT message on 'zigbee2mqtt/Ground Cloakroom/set' with data '{"state":"ON","brightness":200}'
Zigbee2MQTT:debug 2023-01-12 09:38:39: Publishing 'set' 'brightness' to 'Ground Cloakroom'
Zigbee2MQTT:info  2023-01-12 09:38:39: MQTT publish: topic 'zigbee2mqtt/Ground Cloakroom', payload '{"brightness":200,"linkquality":131,"power_on_behavior":null,"state":"ON"}'
Zigbee2MQTT:debug 2023-01-12 09:38:40: Received Zigbee message from 'Ground Cloakroom', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":200}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-01-12 09:38:40: MQTT publish: topic 'zigbee2mqtt/Ground Cloakroom', payload '{"brightness":200,"linkquality":131,"power_on_behavior":null,"state":"ON"}'
Zigbee2MQTT:debug 2023-01-12 09:38:42: Received Zigbee message from 'Ground Cloakroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-01-12 09:38:42: MQTT publish: topic 'zigbee2mqtt/Ground Cloakroom', payload '{"brightness":200,"linkquality":102,"power_on_behavior":null,"state":"ON"}'
Zigbee2MQTT:debug 2023-01-12 09:38:42: Received Zigbee message from 'Ground Cloakroom', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2023-01-12 09:38:42: MQTT publish: topic 'zigbee2mqtt/Ground Cloakroom', payload '{"brightness":200,"linkquality":145,"power_on_behavior":null,"state":"OFF"}'
Zigbee2MQTT:debug 2023-01-12 09:39:02: Saving state to file /config/zigbee2mqtt/state.json
```

gras55, 2023-01-12:
@zollek @emimor123 any chance of providing this? I’m unable to access stdout many thanks

zollek, 2023-01-13:
Here are my log files:
[NOK.txt](https://github.com/Koenkk/zigbee2mqtt/files/10413294/NOK.txt) - turning on dimmer in the frontend by using the brightness slider, it then requires pressing physical switch twice to turn off the light
[Ok.txt](https://github.com/Koenkk/zigbee2mqtt/files/10413296/Ok.txt) - turning on dimmer in the frontend by using the On/Off switch, it then requires pressing physical switch once to turn off - this is how it should works

Koenkk, 2023-01-14:
Looks like the firmware is bug (not supporting the `moveToLevelWithOnOff` correctly), let's see if we can work around it. Does the physical switch turn if off after one press in the following scenarios:

- Scenario 1
  - Turn off switch
  - Publish to `zigbee2mqtt/Light dimmer (1 gang)/set` payload `{"state":"ON"}`
  - Publish to `zigbee2mqtt/Light dimmer (1 gang)/set` payload `{"brightness":190}`
  - Switch should be ON now
  - Now turn off the switch, does it turn off after one press?
- Scenario 2
  - Turn off switch
  - Publish to `zigbee2mqtt/Light dimmer (1 gang)/set` payload `{"brightness":190}`
  - Switch should be ON now
  - Now turn off the switch, does it turn off after one press?

zollek, 2023-01-14:
Scenario 1: Yes

Scenario 2: Light physically is still OFF - it's dark ( ZIGBEE2MQTT state says that it's ON). It happens when brithness is published to the same value as it was before. 
Different brightness value turns physically light ON but then to turn it off, two presses are required.

Koenkk, 2023-01-15:
Could you check if the issue is fixed with the following external converter:  https://gist.github.com/Koenkk/6146510be30d373b8d87b8fe614a46c6

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed, check both scenarios from https://github.com/Koenkk/zigbee2mqtt/issues/15902#issuecomment-1382703850

zollek, 2023-01-15:
I'm geting an error:
```[21:10:11] INFO: Preparing to start...
[21:10:12] INFO: Socat not enabled
[21:10:13] INFO: Starting Zigbee2MQTT...
/app/data/extension/externally-loaded.js:32
        .fromZigbee.concat([tuya.fz.power_on_behavior, fzLocal.TS110E_switch_type, fzLocal.TS110E]),
                                                       ^
ReferenceError: fzLocal is not defined
    at /app/data/extension/externally-loaded.js:32:56
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:150:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:157:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:167:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

After changing:
`tuya.fz.power_on_behavior, fzLocal.TS110E_switch_type, fzLocal.TS110E `
to
`tuya.fz.power_on_behavior, fz.TS110E_switch_type, fz.TS110E`
I'm getting:
```[21:11:42] INFO: Preparing to start...
[21:11:42] INFO: Socat not enabled
[21:11:44] INFO: Starting Zigbee2MQTT...
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:106:18)
```

Koenkk, 2023-01-16:
@zollek updated https://gist.github.com/Koenkk/6146510be30d373b8d87b8fe614a46c6

zollek, 2023-01-16:
I've just tested it and:
Scenario 1: Yes
Scenario 2: Yes

Now it works like it should :)
Thanku You for your work and time.

Koenkk, 2023-01-16:
Pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-16 by Koenkk

emimor123, 2023-01-16:
Working perfectly for me too. 
Thanks for all.

micihol, 2023-01-18:
The physical switch status reporting issue has been fixed however I still have the issue of not being able to turn the light on via physical switch after it was disabled via front-end or voice command. Front-end or voice-command switch-off changes state to OFF and sets the brightness level to 0 and the brightness stays that way when the physical switch is pressed.

zollek, 2023-01-18:
For me light turn off command:
```
service: light.turn_off
data: {}
target:
  entity_id: light.light_dimmer_1_gang
```
turns off the light and doesn't change the brightness.

If you turn off this dimmer by lowering your brightenss to 0 and then try to turn it on by phisicall switch, it turns on with the brigthess 0 (HA state says ON).
The simplest workaround of it , is to turn off device using its state.

micihol, 2023-01-18:
Well, yours acts differently than mine then - this what the device state is, after I call the light.turn_off service:

{
    **"brightness": 0,**
    "brightness_1": 114,
    "linkquality": 90,
    "max_brightness": 115,
    "min_brightness": 61,
    "power_on_behavior": "previous",
    "power_on_behavior_1": "previous",
    **"state": "OFF",**
    "state_1": "OFF",
    "switch_type": "momentary",
    "switch_type_1": "momentary"
}

Now I press the physical switch:

{
    **"brightness": 0,**
    "brightness_1": 114,
    "linkquality": 36,
    "max_brightness": 115,
    "min_brightness": 61,
    "power_on_behavior": "previous",
    "power_on_behavior_1": "previous",
    **"state": "ON",**
    "state_1": "OFF",
    "switch_type": "momentary",
    "switch_type_1": "momentary"
}

Now I set the brightness via physical switch:

{
    "brightness": 60,
    "brightness_1": 114,
    "linkquality": 60,
    "max_brightness": 115,
    "min_brightness": 61,
    "power_on_behavior": "previous",
    "power_on_behavior_1": "previous",
    "state": "ON",
    "state_1": "OFF",
    "switch_type": "momentary",
    "switch_type_1": "momentary"
}

Now I switch off by a service call:

{
    **"brightness": 0,**
    "brightness_1": 114,
    "linkquality": 87,
    "max_brightness": 115,
    "min_brightness": 61,
    "power_on_behavior": "previous",
    "power_on_behavior_1": "previous",
    **"state": "OFF",**
    "state_1": "OFF",
    "switch_type": "momentary",
    "switch_type_1": "momentary"
}

As you see, the brightness parameter changes itself, without being called.

r4f4w, 2023-02-04:
Hey! Where did you buy your TS110E? I was looking and found this module https://www.aliexpress.com/item/1005003693644741.html but I'm not sure if this is the same one

emimor123, 2023-02-04:
Yes, it is

r4f4w, 2023-02-04:
> Yes, it is

Thanks! But how do I know if this is the TS110E_2gang_2 or TuYa TS110E_1gang_2 (both two gang, but only one of them can change power_on_behavior)

bzyx, 2023-11-05:
Sorry, for taking this issue from the grave after so long time. But the issue just showed to me. I observed the same behavior as @micihol described in his last post. I did not use dimmers from Alixepress but Polish white-label/importer OXT. I have this behavior happening on 1 or 2-channel dimmers.
I use [1.33.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2) commit: [9996c93](https://github.com/Koenkk/zigbee2mqtt/commit/9996c93) with the latest HA 2023.11.1. 

When I use the service to turn off the lights, the brightness value goes to 0 and the state is set to "off", on the other hand when I use the physical switch, the brightness stays only the state changes to "off".

tbuaron, 2023-11-08:
Hi, I have same problem as @micihol and @bzyx 
I have tested it on latest "Zigbee2MQTT Edge" and "Zigbee2MQTT 1.33.2-1"
help needed.
Thanks.

bzyx, 2023-11-08:
I'm staring to think that this is not the case for Z2M but I can gather and provide logs if needed. I've tested multiple dimmers from the same venedor with 1 and 2 channel that I've installed in my house. After my tests I've found out that 2 of them behaves wrongly, the rest of them preserves the brightness value between setting it of by Z2M and turning on by hardware switch.

It seems that some of them might have some other firmware version and behaves differently. I've checked and the device type in Z2M is the same.

micihol, 2023-11-09:
I am not sure it's a firmware issue. Surely there is a firmware difference however ZHA works properly with the hardware and this unresolved problem forced me to resign from Z2M and migrate my network to ZHA. The issue made the dimmers unusable to those using both ZigBee and physical controls.

patapovich, 2024-01-12:
Having exactly same problems

micihol, 2024-01-14:
Hi @Koenkk, can this problem be reopened or new one should be created based on the recent reports. It's definitely not fixed.

tbuaron, 2024-01-15:
Hi,
I have same problem.
Hoping for solution.
Thanks

patapovich, 2024-01-15:
This is way out of my expertise, but [this](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/commit/a16baf5300cde2efef55140c3380c0fda3c41f43) is apparently how _Zigbee for Domoticz_ deals with the issue.

mkarbowy, 2024-07-02:
Hi @Koenkk, I'm experiencing the same issue on 1.39.0-1. Any chance this can be revisited?

RoGu777, 2024-09-14:
I've now created a new issue  #23967 with debug information. I hope it will help to resolve it.