Koenkk/zigbee2mqtt #15904: Change color using remote controller will not appear in homeassistant.

Hadatko, 2023-01-03:
### What happened?

Hi, i have remote controller  `MiBoxer | FUT089Z`.
I am controlling `TuYa | TS0504B` led controller. 
When i turn on/off light led using remote controller everything is just right. 
If i change color using remote controller, the color of device is changed but in dashboard i see old color. 
[image]

### What did you expect to happen?

When updating color, update also dasboard color

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20221213.1

### Adapter

Sonof plus

Hadatko, 2023-01-03:
I took a look into zigbee settings directly and there is no color update too:
[image]

Hadatko, 2023-01-03:
Same for other parameters like brightness,...

danieledwardgeorgehitchcock, 2023-01-03:
Your devices may not support reporting of state changes [see here](https://www.zigbee2mqtt.io/guide/usage/binding.html#state-changes) for an explanation

Hadatko, 2023-01-03:
I am not very wise from this
```
If your devices do not support reporting put the device in a group and bind the remote to the group instead of directly to the device. This will make Zigbee2MQTT poll the device for updates when the bound remote controls the device. To minimize traffic this has not been enabled for all devices. 
```

Do i need change this configuration, or should i report bug?
[image]

Hadatko, 2023-01-05:
I accidentally added remoteControll to this group and i cannot remove it. It is failing. Also force removing group and creating it again doesn't help (all devices present)

Hadatko, 2023-01-05:
```log
Error 2023-01-05 02:26:09Request 'zigbee2mqtt/bridge/request/group/remove' failed with error: 'Failed to remove group 'lightLivingRoomGroup' (force: false) (Error: Command 0x94deb8fffebe2b76/1 genGroups.remove({"groupid":101}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 63129 - 1 - 14 - 4 - 3 after 10000ms))'
Error 2023-01-05 02:26:32Failed to remove from group (Command 0x94deb8fffebe2b76/1 genGroups.remove({"groupid":101}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 63129 - 1 - 16 - 4 - 3 after 10000ms))
Error 2023-01-05 02:26:52Failed to bind cluster 'genOnOff' from 'remoteController' to 'lightLivingRoomGroup' (Error: Bind 0x94deb8fffebe2b76/1 genOnOff from '101' failed (AREQ - ZDO - bindRsp after 10000ms))
Error 2023-01-05 02:27:02Failed to bind cluster 'genLevelCtrl' from 'remoteController' to 'lightLivingRoomGroup' (Error: Bind 0x94deb8fffebe2b76/1 genLevelCtrl from '101' failed (AREQ - ZDO - bindRsp after 10000ms))
Error 2023-01-05 02:27:02Failed to bind
```

Hadatko, 2023-02-05:
?

Hadatko, 2023-03-08:
?

closed 2023-04-16 by github-actions[bot]

T-vK, 2023-10-12:
The `MiBoxer FUT089Z` is a very though device to properly integrate into Home Assistant etc. 
The only way to do this is by translating the group messages sent from the remote into properly crafted MQTT messages published to the appropriate topics. I am currently working on that here: https://github.com/Koenkk/zigbee2mqtt-user-extensions/pull/3
If you paste the code of this user extension in the extensions tab in Zigbe2MQTT, the all the controls will show up in Home Assistant.

SimeJah, 2024-05-15:
Hey! I have the same remote and I am running Z2MQTT 1.37. I only see Battery, Voltage, Action and Linkquality exposed. How did you get color and brightness in there at all?

T-vK, 2024-05-16:
> Hey! I have the same remote and I am running Z2MQTT 1.37. I only see Battery, Voltage, Action and Linkquality exposed. How did you get color and brightness in there at all?

Did you follow my instructions?

SimeJah, 2024-05-16:
Hey, Sorry it was my lack of understanding. The screenshot was of the parameters of the light. not the remote. I installed the extension and the remote works. Just like you said only the color in the HA UI does not update. But this is a minor inconvenience I can live with IMO.