Koenkk/zigbee2mqtt #15905: Schneider Electric WDE002961 / Elko SmartDim vri MultiWire does not work properly

henrikeri, 2023-01-03:
### What happened?

It is not possible to control the dimmer from zigbee2mqtt or home assistant. I can only see what changes are physically made on the dimmer. It pairs ok without errors. I have 21 of these and all added correctly, but none are controllable.

### What did you expect to happen?

I expected to be able to turn the dimmer on/off and to change brightness from zigbee2mqtt

### How to reproduce it (minimal and precise)

Pair dimmer and attempt to control from zigbee2mqtt

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

0x26720700

### Adapter

Conbee II

### Debug log

```

Debug 2023-01-03 11:06:00Received MQTT message on 'zigbee2mqtt/Taklys Spisestue/set' with data '{"brightness":47}'
Debug 2023-01-03 11:06:00Publishing 'set' 'brightness' to 'Taklys Spisestue'
Info 2023-01-03 11:06:00MQTT publish: topic 'zigbee2mqtt/Taklys Spisestue', payload '{"ballast_maximum_level":null,"ballast_minimum_level":null,"brightness":47,"dimmer_mode":null,"linkquality":255,"state":"ON"}'
Debug 2023-01-03 11:06:06Received MQTT message on 'zigbee2mqtt/Taklys Spisestue/get' with data '{"brightness":""}'
Debug 2023-01-03 11:06:06Publishing get 'get' 'brightness' to 'Taklys Spisestue'
Debug 2023-01-03 11:06:06Received Zigbee message from 'Taklys Spisestue', type 'readResponse', cluster 'genLevelCtrl', data '{}' from endpoint 21 with groupID null
Info 2023-01-03 11:06:06MQTT publish: topic 'zigbee2mqtt/Taklys Spisestue', payload '{"ballast_maximum_level":null,"ballast_minimum_level":null,"dimmer_mode":null,"linkquality":255,"state":"ON"}'
Error 2023-01-03 11:06:06Publish 'get' 'brightness' to 'Taklys Spisestue' failed: 'Error: Read 0x94deb8fffe552491/21 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2023-01-03 11:06:06Error: Read 0x94deb8fffe552491/21 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22) at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:986:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:273:21)
Info 2023-01-03 11:06:06MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'brightness' to 'Taklys Spisestue' failed: 'Error: Read 0x94deb8fffe552491/21 genLevelCtrl([\"currentLevel\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"Taklys Spisestue"},"type":"zigbee_publish_error"}'
```

verydrunk, 2023-01-05:
U probably need to change the endpoint for onOff and levelCtrl. I noticed that Schneider doesn't like to be consistent in there products.

juha077, 2023-02-03:
Did you ever figure out the issue? I also have bunch of these dimmers with exactly the same issue? I tried to change the endpoints but did not get anywhere with that.

henrikeri, 2023-02-03:
Sort of. Not actually solving the underlying issue, but for me they started working if I did a hard reset of z2m

The0d-lin, 2023-02-09:
I have this exact problem. How did you hard reset zigbee2mqtt? Did you have to reconnect every zigbee device after that?

peec, 2023-03-17:
Having same problem with my elko paired as WDE002961

closed 2023-04-25 by github-actions[bot]