Koenkk/zigbee2mqtt #15907: Cant update to latest code "error: Your local changes to the following files would be overwritten by merge:"

TzachiGuetta, 2023-01-03:
### What happened?

I'm tring to update my zigbee2mqtt to the latest code using:
`# Stop Zigbee2MQTT and go to directory
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt

# Backup configuration
cp -R data data-backup

# Update
git pull
npm ci

# Restore configuration
cp -R data-backup/* data
rm -rf data-backup

# Start Zigbee2MQTT
sudo systemctl start zigbee2mqtt`

and I'm getting:

`root@debianLinux:/opt/zigbee2mqtt# git pull
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
remote: Enumerating objects: 759, done.
remote: Counting objects: 100% (477/477), done.
remote: Compressing objects: 100% (139/139), done.
remote: Total 290 (delta 224), reused 212 (delta 151), pack-reused 0
Receiving objects: 100% (290/290), 100.19 KiB | 932.00 KiB/s, done.
Resolving deltas: 100% (224/224), completed with 44 local objects.
From https://github.com/Koenkk/zigbee2mqtt
   52e545f9..dddce571  master     -> origin/master
 * [new tag]           1.29.0     -> 1.29.0
Updating 52e545f9..dddce571
error: Your local changes to the following files would be overwritten by merge:
        .github/workflows/ci.yml
        .github/workflows/stale.yml
        .github/workflows/update_deps.yml
        .github/workflows/update_frontend.yml
        .github/workflows/update_zh.yml
        .github/workflows/update_zhc.yml
        docker/Dockerfile
        lib/controller.ts
        lib/extension/bridge.ts
        lib/extension/configure.ts
        lib/extension/externalExtension.ts
        lib/extension/frontend.ts
        lib/extension/groups.ts
        lib/extension/homeassistant.ts
        lib/extension/networkMap.ts
        lib/extension/otaUpdate.ts
        lib/state.ts
        lib/types/types.d.ts
        lib/util/settings.schema.json
        lib/util/settings.ts
        lib/util/utils.ts
        package-lock.json
        package.json
        scripts/generateNewSupportedDevices.js
        test/availability.test.js
        test/bridge.test.js
        test/configure.test.js
        test/controller.test.js
        test/frontend.test.js
        test/group.test.js
        test/homeassistant.test.js
        test/networkMap.test.js
        test/otaUpdate.test.js
Please commit your changes or stash them before you merge.
Aborting`

### What did you expect to happen?

to be able to run git pull

### How to reproduce it (minimal and precise)

git pull

### Zigbee2MQTT version

1.28.4

### Adapter firmware version

 20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (CC2652P)

TzachiGuetta, 2023-01-03:
@Koenkk previously I had issues with:

> Zigbee2MQTT:error 2022-12-28 15:00:42: Error: The module '/opt/zigbee2mqtt/node_modules/@serialport/bindings/build/Release/bindings.node'
> was compiled against a different Node.js version using

So I ran those comamds: 

> root@:/var/lib/homebridge# cd /opt/zigbee2mqtt/
> root@:/opt/zigbee2mqtt# rm -rf node-modules/serialport
> root@:/opt/zigbee2mqtt# rm -rf node-modules/@serialport
> root@:/opt/zigbee2mqtt# rm -rf packages-lock.json
> root@:/var/lib/homebridge# sudo npm install --save-dev electron-rebuild
> root@:/opt/zigbee2mqtt# sudo npm i
> root@:/opt/zigbee2mqtt# sudo chmod 666 /dev/ttyUSB0
> root@:/opt/zigbee2mqtt# sudo systemctl start zigbee2mqtt

danieledwardgeorgehitchcock, 2023-01-03:
Looks like you need to rebase or merge the files with conflicts. Is there a reason why you are building from source as opposed to using one of the releases?

TzachiGuetta, 2023-01-03:
@danieledwardgeorgehitchcock I'n not sureI follow. can you explain ?

danieledwardgeorgehitchcock, 2023-01-03:
You're trying to update Z2M using git pull and node build which makes me think you've installed Z2M through a non-supported [installation method](https://www.zigbee2mqtt.io/guide/installation/) 

You shouldn't need to build Z2M unless you are running a highly customised code base or are developing improvements to be merged with the Z2M project.

It would be a lot easier to maintain your installation if you just followed one of the recommended [installation methods](https://www.zigbee2mqtt.io/guide/installation/)

TzachiGuetta, 2023-01-03:
Now I follows you, 
the changes in the zigbeee2mqtt folder were made due to issues I had previously which forced me to run:

> root@:/var/lib/homebridge# cd /opt/zigbee2mqtt/
> root@:/opt/zigbee2mqtt# rm -rf node-modules/serialport
> root@:/opt/zigbee2mqtt# rm -rf node-modules/https://github.com/serialport
> root@:/opt/zigbee2mqtt# rm -rf packages-lock.json
> root@:/var/lib/homebridge# sudo npm install --save-dev electron-rebuild
> root@:/opt/zigbee2mqtt# sudo npm i
> root@:/opt/zigbee2mqtt# sudo chmod 666 /dev/ttyUSB0
> root@:/opt/zigbee2mqtt# sudo systemctl start zigbee2mqtt

I didn't performed any attentional changes on the folder - just tried to solved issues around: https://github.com/Koenkk/zigbee2mqtt/issues/2425

TzachiGuetta, 2023-01-03:
@Koenkk what should be the right approach here?

kremik, 2023-01-08:
I'm havig the same problems when updating from 1.29.0 to 1.29.1. I haven't done any changes to my configuration, purely using the core from git.

$ git pull
remote: Enumerating objects: 49, done.
remote: Counting objects: 100% (49/49), done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 49 (delta 35), reused 44 (delta 31), pack-reused 0
Unpacking objects: 100% (49/49), done.
From https://github.com/Koenkk/zigbee2mqtt
   dddce571..7d67ffc4  master     -> origin/master
   826f0322..e92d3f4e  dev        -> origin/dev
 * [new tag]           1.29.1     -> 1.29.1
Updating dddce571..7d67ffc4
error: Your local changes to the following files would be overwritten by merge:
	package-lock.json
Please commit your changes or stash them before you merge.
Aborting

Koenkk, 2023-01-08:
@TzachiGuetta not sure what happened, probably you changed permission on the files. I suggest to backup the z2m `data` directory, remove the whole z2m dir, clone again and copy `data` folder back

@kremik for you `git checkout package-lock.json` should be enough.

closed 2023-01-10 by TzachiGuetta

pnjoki, 2023-10-09:
i'm having this error
git pull origin master
From https://github.com/JOHNMARK-cmd/phase5project
 * branch            master     -> FETCH_HEAD
error: Your local changes to the following files would be overwritten by merge:
        GROUP-SIX.ipynb
Please commit your changes or stash them before you merge.
Aborting
Updating 132c3a5..f12bd3e