Koenkk/zigbee2mqtt #15912: Problem with +20dBm Sonoff Zigbee stick router

Smandurlo, 2023-01-03:
### What happened?

Hello, yesterday I added another sonoff zigbee router. I will write down my results:

- I used the latest router firmware dev first. I wasn't able to increase the dBm, I got timeout writing the setting. The first sonoff zigbee router worked already @+20dBm, but I wan't able to change the setting. Into the exposes page, the update button doesn't work;
- I updated to the latest dev to the first router too. Nothing changed;
- I updated both the first and the new router to the 6.10.01.01 version, no change. I can't change the +20dBm on the first router, the new router timeout writing the setting;
- I read there is a new coordinator version to solve the problem with the timeout. Updated the coordinator firmware (6.10.01.01), rebooted zigbee2mqtt and now I am able to change the setting to the new router. With +20dBm I saw a huge difference and the update button works (I can see the lqi changing);
- the first router doesn't change the dBi, it is stuck to +20dBm and I don't see any change to the lqi pushing the update button.

What is the problem? The coordinator firmware? The router firmware? The z2m version (I have the lastest, not dev)?
Any idea?

Thanks

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee stick

### Debug log

changing the dBm to the new router
info 2023-01-03 12:56:46MQTT publish: topic 'zigbee2mqtt/0x00124b0024c0af49', payload '{"last_seen":"2023-01-03T12:56:47+01:00","linkquality":43,"transmit_power":20}'
info 2023-01-03 12:56:46MQTT publish: topic 'zigbee2mqtt/0x00124b0024c0af49', payload '{"last_seen":"2023-01-03T12:56:47+01:00","linkquality":43,"transmit_power":10}'
info 2023-01-03 12:57:06MQTT publish: topic 'zigbee2mqtt/0x00124b0024c0af49', payload '{"last_seen":"2023-01-03T12:57:07+01:00","linkquality":72,"transmit_power":10}'
info 2023-01-03 12:57:07MQTT publish: topic 'zigbee2mqtt/0x00124b0024c0af49', payload '{"last_seen":"2023-01-03T12:57:07+01:00","linkquality":72,"transmit_power":20}'

Changing the dBm to the first router:

nothing on the log, it seems I don't issue any command

radek-kafka, 2023-01-03:
I have Sonoff Dongle-P with router firmware :  CC1352P2_CC2652P_launchpad_router_20221102.hex

and while setting the power I'm receiving:
 Zigbee2MQTT:error 2023-01-03 15:18:02: Publish 'set' 'transmit_power' to 'SonoffRouter' failed: 'Error: Write 0x00124b00257b2eb5/242 genBasic({"4919":{"value":20,"type":40}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34526 - 242 - 72 - 0 - 4 after 10000ms)'
Jan 03 15:18:02 homeassistant addon_45df7312_zigbee2mqtt[638]: Zigbee2MQTT:debug 2023-01-03 15:18:02: Error: Write 0x00124b00257b2eb5/242 genBasic({"4919":{"value":20,"type":40}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34526 - 242 - 72 - 0 - 4 after 10000ms)
Jan 03 15:18:02 homeassistant addon_45df7312_zigbee2mqtt[638]:     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
Jan 03 15:18:02 homeassistant addon_45df7312_zigbee2mqtt[638]:     at listOnTimeout (node:internal/timers:559:17)
Jan 03 15:18:02 homeassistant addon_45df7312_zigbee2mqtt[638]:     at processTimers (node:internal/timers:502:7)
Jan 03 15:18:02 homeassistant addon_45df7312_zigbee2mqtt[638]: Zigbee2MQTT:info  2023-01-03 15:18:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'transmit_power' to 'SonoffRouter' failed: 'Error: Write 0x00124b00257b2eb5/242 genBasic({\"4919\":{\"value\":20,\"type\":40}}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 34526 - 242 - 72 - 0 - 4 after 10000ms)'","meta":{"friendly_name":"SonoffRouter"},"type":"zigbee_publish_error"}'

Smandurlo, 2023-01-03:
Try the 6.10.01.01 firmware version, it should solve the timeout problem

radek-kafka, 2023-01-03:
Hmm... I have tried and nothing has changed. I downloaded the firmware from : https://github.com/Koenkk/Z-Stack-firmware/raw/6.10.01.01/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20221102.zip

xekil, 2023-01-06:
I have the same problem, I tried version 6.10.01.01 on the Router (Sonoff) and also on the Coordinator (ZZH), without success, I get this error :(

`Error 2023-01-06 23:00:05Publish 'get' 'transmit_power' to '0x00124b00257b3eb7' failed: 'Error: Read 0x00124b00257b3eb7/242 genBasic([4919], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 27116 - 242 - 24 - 0 - 1 after 10000ms)'`

manumora, 2023-01-16:
I have the same error:

> `Error 2023-01-16 13:34:21Publish 'set' 'transmit_power' to 'device.sonoff.router' failed: 'Error: Write 0x00124b0029dbfc9d/8 genBasic({"4919":{"value":6,"type":40}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

Firmware version: 20220125

Smandurlo, 2023-01-16:
> 

that firmware doesn't support the +20dBm feature

Smandurlo, 2023-02-07:
I think I've narrowed down the problem: it seems related to the latest router you add.

My second router didn't want to update the transmit power: timeout error. Yesterday I added a 3rd router: now the second router updates the transmit power and the 3rd doesn't with the transmit power timeout error. I have no other router available, but I am pretty sure that if I add a 4th router, the 3rd will start to work properly and the 4th won't.

Lastest master firmware for the coordinator and dev for the routers.

underqualifieditguy, 2023-02-08:
Am I experiencing the same issue? I am on the latest firmware and I am getting this error which led me to this page..

```
Zigbee2MQTT:error 2023-02-07 19:40:28: Publish 'get' 'transmit_power' to '0×00124b0026b806f8' failed: 'Err or: Read 0x00124b0026b806f8/8 genBasic([4919], {"sendWhen":"immediate","timeout": 10000, "disableResponse":f alse, "disableRecovery": false,"disableDefaultResponse":true, "direction":, "srcEndpoint":null, "reservedBits" :,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORT ED ATTRIBUTE')'
Zigbee2MQTT: info 2023-02-07 19:40:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":" Publish 'get' 'transmit_power' to '0×001240026b806f8' failed: 'Error: Read 0×00124b0026b806f8/8 genBasicl
(4919], {\"sendWhen\": \"immediate\", \"timeout\": 10000, \"disableResponse\": false, l"disableRecovery\": false,
"disableDefaultResponseve-truebertrection.tesreEndpoints"null,lereservedBiteNsurroranufacturercode
'": null, "transactionSequenceNumber)":null, ("writeUndivi": false)) failed (Status 'UNSUPPORTED _ATTRIBUTE') ,"meta": {"friendly_name": "Ox00124b0026b806f8"}, "type":"zigbee_publish_error"}'
Zigbee2MQTT: info
2023-02-07 19:40:34: MQTT publish: topic 'zigbee2mqtt/Kitchen Window 6',
```

s0ftice, 2023-02-08:
Same here.
What's worse, in my setup, the affected "lastly added" router also drops connected devices. Not sure if that's the case for you guys as well.

This started happening early Dec 2022:

- around the Z2M 1.28.4 upgrade (I am now on 1.30.1-1, the issue persists).
- when the four routers ran firmware 20220125, but the issue persists after upgrading all of them to 20221102 (develop).
- when the coordinator ran firmware 20220219, but the issue persists after upgrading it to 20221226 (master).
(my coordinator and 4x routers are all SONOFF Dongle-P).

I'm aware that the transmit_power was not yet implemented on the 20220125 router firmware. But when one router started dropping connections, I upgraded to 20221102 and realised that the one dropping connections is always the one that times out on changing transmit_power. In my setup, there is a clear correlation between the two issues.

radek-kafka, 2023-02-08:
As I wrote: "After update of firmware on coordinator to same 6.10.01.01 version is seems to be better .. Thx .." .. So I have updated version on Coordinator and Router to 6.10.01.01 brach and is working weel now.. The network is stable  ..

underqualifieditguy, 2023-02-08:
> > Coordinator and Router to 6.10.01.01 brach
> 
> they are not available anymore, there are new master and dev branches. I had the same problem with 6.10.01.01 too.

@Koenkk can you comment as to why the branch was removed?

Koenkk, 2023-02-08:
`6.10.01.01` has been integrated into the `master` branch.

underqualifieditguy, 2023-02-08:
> `6.10.01.01` has been integrated into the `master` branch.

not for the launchpad_router firmware though?

Koenkk, 2023-02-08:
@underqualifieditguy it was still in the `develop` branch, but I just added it to `master`.

underqualifieditguy, 2023-02-08:
> @underqualifieditguy it was still in the `develop` branch, but I just added it to `master`.

`20221102` contains the SimpleLink SDK version 6.10.01.01?

Koenkk, 2023-02-08:
[`6.30.00.84`](https://github.com/Koenkk/Z-Stack-firmware/blob/a7745634b539df3d0b1ad1ae84a137017af58fe2/router/Z-Stack_3.x.0/CHANGELOG.md)

underqualifieditguy, 2023-02-08:
> [`6.30.00.84`](https://github.com/Koenkk/Z-Stack-firmware/blob/a7745634b539df3d0b1ad1ae84a137017af58fe2/router/Z-Stack_3.x.0/CHANGELOG.md)

OK do you know if this is supposed to fix the TX power issue that this thread was started about?

Smandurlo, 2023-02-08:
May I be honest? I find a little bit confusing the way you name the versions. They looks like yyyymmdd and you don't get easily if they are old (2022) or new or a repack of an old build with the same name or whatever.

So, do we have a new router version? Does it solve this problem with the last added router?

Enzokot, 2023-02-08:
> Does it solve this problem with the last added router?

The last added firmware router `20221102` doesn't solve this problem.

Smandurlo, 2023-02-08:
ok... thank. So we will wait for it :-)

underqualifieditguy, 2023-02-08:
> > Does it solve this problem with the last added router?
> 
> The last added router firmware `20221102` doesn't solve this problem.

@Koenkk is there a release planned to address?

Enzokot, 2023-02-08:
And ping router timeout with availability setting

Koenkk, 2023-02-09:
> Does it solve this problem with the last added router?

Can you give me a summary of this problem? The OP is about something different it seems.

Smandurlo, 2023-02-09:
@Koenkk it started differently, but now I norrowed down the problem: I still get the timeout error while changind the TX power to the sonoff dongle plus router, but it seems it is related to the last added router. Meanwhile, I switched some coordinator/router firmwares and things changed, but the timeout problem still happens.

I wrote it 2 days ago:
"My second router didn't want to update the transmit power: timeout error. Yesterday I added a 3rd router: now the second router updates the transmit power and the 3rd doesn't with the transmit power timeout error. I have no other router available, but I am pretty sure that if I add a 4th router, the 3rd will start to work properly and the 4th won't.

Lastest master firmware for the coordinator and dev for the routers."

Enzokot, 2023-02-10:
@Smandurlo could you try to restart `z2m` after add the last router? It's seems this work for me. I can change TX power and the route is `online`
My config:
z2m v.`1.30.1`
Firmware from `master` branch:
Coordinator v.`20221226` [link](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20221226.zip)
Router v.`20221102` [link](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20221102.zip)

Smandurlo, 2023-02-10:
@Enzokot thanks for your input, I am using the same versions. I did a restart and now the last added router works, but the 2nd don't. Moreover, 2 window/door xiaomi sensors near the router that doesn't change the TX power anymore are offline 😭

s0ftice, 2023-02-10:
exactly what I reported, this issue goes hand in hand with dropping off of devices from that particular router.
I had this issue for the last two months. A simple restart does not help @Enzokot.

Enzokot, 2023-02-10:
Maybe my example is bad, because I have only one TI router

Smandurlo, 2023-02-10:
I tried to unplug a random router (so 2 left plugged in) and then reboot z2m: I always have 1 router that doesn't accept the TX change. That is odd.

Koenkk, 2023-02-10:
@Smandurlo can you try moving them in direct range of the coordinator? Looks that the are unreachable.

s0ftice, 2023-02-10:
I tried that, LQI around 200, same issue

Smandurlo, 2023-02-10:
They are in the same room for the tests, but they are normally near enough to see each other.

gerard33, 2023-02-11:
I experience the same issue with not being able to set transmit power to the latest router that is added.

1. Router 1 is a SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P, is on correct firmware and setting transmit power to 20 works (after having added router 2 and 3)
2. Router 2 is a SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P as well, is on correct firmware and setting transmit power to 20 works (after having added router 3)
3. Router 3 is a SMLIGHT CC2652P Zigbee USB Adapter SLZB-02 and is added as the last of the 3 routers, is on correct firmware and setting transmit power to 20 does not work but gives this error:

`
Publish 'set' 'transmit_power' to 'Smartlight SLZB-02 router' failed: 'Error: Write 0x00124b002579310d/242 genBasic({"4919":{"value":20,"type":40}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 21789 - 242 - 155 - 0 - 4 after 10000ms)'
`

Changing the transmit power for the first 2 routers immediately shows an increase of the linkquality.

Smandurlo, 2023-02-11:
@gerard33 try to change the TX power to all your router. Pay attention to the other routers cause it probably won't give you any error, but the TX power will not actually changed. Try this: change the TX power, move to another page and then return back to the "Exposes" page. I bet you will not see the new value, but it is locked to the previous one. The linkquality doesn't change anymore. I am pretty sure it will happen to one of your routers only.

billyjoebob999, 2023-02-13:
I have 3 Sonoff dongles, one is the coordinator, and two are acting as routers. All have been updated with the latest firmware. I have no issues changing the TX on the routers. I can update the TX in Z2M and its reflected in HA. I can then change the TX in HA and confirmed Z2M shows the new value.

One thing I noticed after updating both routers, Z2M still showed them running the previous firmware. I had to removed both of then and add then back for things to work correctly.

Smandurlo, 2023-02-13:
@billyjoebob999 could you please tell me how your coordinator/routers are arranged? I mean, is your coordinator in the middle of the two routers? Do your routers see each other?

billyjoebob999, 2023-02-13:
@Smandurlo with the layout of my home I'm able to centrally locate the coordinator. Each router is then located at each end of the house. I don't know how to tell if the two routers are talking to each other. I assume they are.

Smandurlo, 2023-02-13:
Just look at your map if your routers are connected :-) my coordinator is not in the middle of the routers, but on a side, so a router is connected to the coordinator  through another router. Maybe this is the problem.

billyjoebob999, 2023-02-14:
Per the map they connect to the coordinator. This also true for any mains powered devices that act as routers. They connect to the coordinator regardless of where they are placed (several placed through out the house).

gmaslowski, 2023-03-14:
@gerard33 @Smandurlo @Koenkk 

I'm experiencing similar issues, but I managed to configure the transmit power in the following manner.

**Setup**
- 1x Sonoff Zigbee 3.0 USB Dongle Plus as Coordinator on `20221226`
- 2x Sonoff Zigbee 3.0 USB Dongle Plus as Routers on `20221102` 

**Procedure**
1. Add 1st Router to the network
2. Restart zigbee2mqtt
3. Set TX Power on Router
4. Add 2nd Router to the network
5. Restart zigbee2mqtt
6. Set TX Power on Router

Tried other options, with no luck. Looks like somehow, after the Routers pairing I needed to restart zigbee2mqtt. Not sure if that will really hold for long though :P. Looks like the value is set for now.

Smandurlo, 2023-03-16:
nothing changed for me. I was able to change the TX power in my 2nd router when I added a 3rd one, but now the 2nd is "stuck". It shows +20dB, but it doesn't change anymore and I am not actually sure it is really +20dB. The first one works as expected.

gmaslowski, 2023-03-16:
@Smandurlo I was not trying to change those values again though. 

The json State states 20 as power for both the routers. The mesh network and LQI on all the devices in my network "looks better".

klu16, 2023-03-29:
Workaround from @gmaslowski worked for me.
I restarted the Z2M addon and after that the transmit power setting to 20 was saved. Thanks!

Smandurlo, 2023-04-29:
Sadly the problem is still there

Smandurlo, 2023-05-31:
I wish it was solved 😢

closed 2023-07-09 by github-actions[bot]

baerni, 2023-08-08:
Which router firmware version should I upload to have +20 transmit power?