Koenkk/zigbee2mqtt #15916: map "special" commands in z2m to effects

Bascht74, 2023-01-03:
### Is your feature request related to a problem? Please describe

m

### Describe the solution you'd like

m

### Describe alternatives you've considered

n

### Additional context

m

Bascht74, 2023-01-03:
Hi, 

I am bringing this up again, because of the new "Switch manager" (Custom component). 
I would really like to be able to use those zigbee functions within Home Assistant. 

There was no reaction the last time. I am hoping to get a positive one this time.
I am not a programmer, but I hope that just "passing through" the special commands and process it within z2m (effect --> command) could be hopefully not too difficult. So I keep my fingers crossed this time!

If there are other easy possibilities to acces those z2m command within a "normal" HA entity / device, pleas let me know. Maybe I oversaw something.

Gerben321, 2023-01-26:
Would be awesome!

closed 2023-03-05 by github-actions[bot]