Koenkk/zigbee2mqtt #15924: IH012-RT01 not reporting battery nor voltage

FragMenthor, 2023-01-03:
### What happened?

I have 2 of this motion sensors, that correctly reported everything until now, but now they don't report **battery %** nor **voltage**.
I don't rebember if the device model was this one before the last upgrade, I just know that these sensors stopped reporting, and I am not able to change the **sensivity** or **keep time** anymore.

### What did you expect to happen?

That the device continued working like before version 1.29.0.

### How to reproduce it (minimal and precise)

Upgrade to zigbee2mqtt 1.29.0.

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220726

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

FragMenthor, 2023-01-04:
EDIT: I reverted Zigbee2MQTT to 1.28.4 and the device model is the same and it is correctly working, but it dows not expose battery % or voltage. It seems this settings are incorrectly exposed with value "none" in 1.29.0. The device only normally exposes "battery_low" sensor:

```
{
    "battery_low": false,
    "keep_time": 30,
    "last_seen": "2023-01-04T07:34:29+00:00",
    "linkquality": 116,
    "occupancy": false,
    "sensitivity": "high",
    "tamper": false
}
```

callacomp, 2023-01-23:
Same here, not done any digging yet, but agree with @FragMenthor.

closed 2023-03-03 by github-actions[bot]

vxav, 2024-04-23:
I found the battery level and voltage on this device didn't get updated in z2m after replacement.