Koenkk/zigbee2mqtt #15926: Xiaomi ZNCLDJ11LM state is always open

ASNNetworks, 2023-01-03:
### What happened?

After updating Zigbee2mqtt to 1.29.1 this [curtain (motor)](https://www.zigbee2mqtt.io/devices/ZNCLDJ11LM.html) always shows the state as open. When closing the curtain, the position attribute works fine and is set at 0%, however the state of the curtain remains as open. 

I'm not sure if it's the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/15873. It's a different model number, so I think it's best to create a ticket for this. Also with this curtain opening/closing (running states) never worked.

### What did you expect to happen?

I expected it to show the correct state when closed

### How to reproduce it (minimal and precise)

Updating to 1.29 causes this. Reverting to 1.28.4 fixes it again.

### Zigbee2MQTT version

1.29.0-1

### Adapter firmware version

20220928

### Adapter

Zigstar V4

ASNNetworks, 2023-01-03:
EDIT: I saw in that other ticket what the dev build fixed it for that curtain. I tried the dev build myself and it seems it also fixed my model version. I will keep this ticket open for awareness. Perhaps other users have my model.

ASNNetworks, 2024-10-05:
This exact issue is happening again, now running latest Z2M (stable) version: 1.40.2-1

I tried removing (forced) and repairing the device and also reset its limits and setup again. State keeps on open, even when at 0%, which breaks automations and button-cards that rely on closed state.

Drafteed, 2024-10-05:
Please close this in favor of #24221

closed 2024-10-05 by ASNNetworks