Koenkk/zigbee2mqtt #15928: [New device support]: Philips Infuse (new model) 929003531702

mwerline, 2023-01-03:
### Link

https://www.philips-hue.com/en-us/p/hue-white-and-color-ambiance-infuse-medium-ceiling-lamp/046677579845

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0x001788010cbdcee2","nwkAddr":40711,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"929003531702","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"zclVersion":2}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20210305","swBuildId":"1.82.10","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1672767840990,"defaultSendRequestWhen":"immediate"}

### Comments

There appears to be another model for the Infuse medium ceiling lamp. I copied/modified the code for one ofthe existing models (915005997301) and it appears to work OK. I just copied the existing Philips.js and then paired down the device module.export to just this model.. Sorry if this request is a bit off.. Haven't put one of these requests in before, so hopefully it's not too far off from how these are supposed to look.. :-)

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const philips = require('zigbee-herdsman-converters/lib/philips');
const utils = require('zigbee-herdsman-converters/lib/utils');
const libColor = require('zigbee-herdsman-converters/lib/color');
const herdsman = require('zigbee-herdsman');
const e = exposes.presets;
const ea = exposes.access;

// Make sure extend.light_* is not used (hueExtend should be used instead)
const extendDontUse = require('zigbee-herdsman-converters/lib/extend');
const extend = {switch: extendDontUse.switch};

const manufacturerOptions = {manufacturerCode: herdsman.Zcl.ManufacturerCode.PHILIPS};

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extendDontUse.light_onoff_brightness(options),
        ota: ota.zigbeeOTA,
        meta: {turnsOffAtBrightness1: true},
        toZigbee: extendDontUse.light_onoff_brightness(options).toZigbee.concat([
            tzLocal.hue_power_on_behavior, tzLocal.hue_power_on_error,
        ]),
    }),
    light_onoff_brightness_colortemp: (options={}) => {
        options = {disableHueEffects: true, ...options};
        if (!options.disableHueEffects) options.disableEffect = true;
        const result = extendDontUse.light_onoff_brightness_colortemp(options);
        result['ota'] = ota.zigbeeOTA;
        result['meta'] = {turnsOffAtBrightness1: true};
        result['toZigbee'] = result['toZigbee'].concat([tzLocal.hue_power_on_behavior, tzLocal.hue_power_on_error]);
        if (!options.disableHueEffects) {
            result['toZigbee'] = result['toZigbee'].concat([tzLocal.effect]);
            result['exposes'] = result['exposes'].concat([exposes.enum('effect', ea.SET,
                ['blink', 'breathe', 'okay', 'channel_change', 'candle', 'finish_effect', 'stop_effect', 'stop_hue_effect'])]);
        }
        return result;
    },
    light_onoff_brightness_color: (options={}) => {
        options = {disableHueEffects: true, ...options};
        if (!options.disableHueEffects) options.disableEffect = true;
        const result = extendDontUse.light_onoff_brightness_color({supportsHS: true, ...options});
        result['ota'] = ota.zigbeeOTA;
        result['meta'] = {turnsOffAtBrightness1: true};
        result['toZigbee'] = result['toZigbee'].concat([tzLocal.hue_power_on_behavior, tzLocal.hue_power_on_error]);
        if (!options.disableHueEffects) {
            result['toZigbee'] = result['toZigbee'].concat([tzLocal.effect]);
            result['exposes'] = result['exposes'].concat([exposes.enum('effect', ea.SET, [
                'blink', 'breathe', 'okay', 'channel_change',
                'candle', 'fireplace', 'colorloop',
                'finish_effect', 'stop_effect', 'stop_hue_effect',
[... 349 lines cut ...]
            return {state: {led_indication: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('genBasic', [0x0033], manufacturerOptions);
        },
    },
};

module.exports = 
    {
        zigbeeModel: ['929003531702'],
        model: '929003531702',
        vendor: 'Philips',
        description: 'Hue Bluetooth white & color ambiance ceiling lamp Infuse medium',
        extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    };
```

### Supported color modes

color temp, rbg

### Color temperature range

colorTempRange: [153, 500]

zegl, 2023-01-04:
Looks good! I've sent a PR!

Koenkk, 2023-01-04:
Thanks @zegl !

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-04 by Koenkk