Koenkk/zigbee2mqtt #15929: Hue white ambiance incorrect colour temperature range

macmanroy, 2023-01-03:
### What happened?

In home assistant my hue white ambiance bulbs report a colour temp range of 2000 - 6666 kelvin (150-500 mireds) this value is over the rated range of these bulbs and should be 2200k - 6500k

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

0x26780700

### Adapter

ConBee ii

macmanroy, 2023-01-04:
> Hey @macmanroy! What's the model number of your device?

It's the hue gu10 white ambiance model LTW013

zegl, 2023-01-05:
Thanks, I've sent a PR to correctly specify the temperature range for this device. :-)

closed 2023-01-06 by Koenkk