Koenkk/zigbee2mqtt #15932: Template warning in log

zipzagster, 2023-01-04:
### What happened?

```
2023-01-03 15:44:33.758 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'
```

### How to reproduce it (minimal and precise)

Continually reported in log

### Zigbee2MQTT version

1.29.0-1

### Adapter firmware version

i see no frontend info except port

### Adapter

tubeszb CC2652P2

### Debug log

```
2023-01-03 18:17:11.463 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on zigbee2mqtt/FP1B: b'{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}'
2023-01-03 18:17:11.463 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'binary_sensor.0x54ef4410005166c9_presence', 'name': 'FP1B presence', 'this': <template TemplateStateFromEntityId(binary_sensor.0x54ef4410005166c9_presence)>} and Template("{{ value_json.presence }}")
2023-01-03 18:17:11.463 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'binary_sensor.0x54ef4410005166c9_update_available', 'name': 'FP1B update available', 'this': <template TemplateStateFromEntityId(binary_sensor.0x54ef4410005166c9_update_available)>} and Template("{{ value_json['update']['state'] == "available" }}")
2023-01-03 18:17:11.463 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'sensor.0x54ef4410005166c9_presence_event', 'name': 'FP1B presence event', 'this': <template TemplateStateFromEntityId(sensor.0x54ef4410005166c9_presence_event)>} with default value 'default' and Template("{{ value_json.presence_event }}")
2023-01-03 18:17:11.464 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'sensor.0x54ef4410005166c9_device_temperature', 'name': 'FP1B device temperature', 'this': <template TemplateStateFromEntityId(sensor.0x54ef4410005166c9_device_temperature)>} with default value 'default' and Template("{{ value_json.device_temperature }}")
2023-01-03 18:17:11.464 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'sensor.0x54ef4410005166c9_power_outage_count', 'name': 'FP1B power outage count', 'this': <template TemplateStateFromEntityId(sensor.0x54ef4410005166c9_power_outage_count)>} with default value 'default' and Template("{{ value_json.power_outage_count }}")
2023-01-03 18:17:11.464 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'select.0x54ef4410005166c9_monitoring_mode', 'name': 'FP1B monitoring mode', 'this': <template TemplateStateFromEntityId(select.0x54ef4410005166c9_monitoring_mode)>} and Template("{{ value_json.monitoring_mode }}")
2023-01-03 18:17:11.464 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'select.0x54ef4410005166c9_approach_distance', 'name': 'FP1B approach distance', 'this': <template TemplateStateFromEntityId(select.0x54ef4410005166c9_approach_distance)>} and Template("{{ value_json.approach_distance }}")
2023-01-03 18:17:11.465 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'select.0x54ef4410005166c9_motion_sensitivity', 'name': 'FP1B motion sensitivity', 'this': <template TemplateStateFromEntityId(select.0x54ef4410005166c9_motion_sensitivity)>} and Template("{{ value_json.motion_sensitivity }}")
2023-01-03 18:17:11.465 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'update.fp1b', 'name': 'FP1B', 'this': <template TemplateStateFromEntityId(update.fp1b)>} and Template("{{ value_json['update']['installed_version'] }}")
2023-01-03 18:17:11.466 DEBUG (MainThread) [homeassistant.components.mqtt.update] No valid (JSON) payload detected after processing payload 'unknown' on topic zigbee2mqtt/FP1B
2023-01-03 18:17:11.466 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"approach_distance":"far","device_temperature":25,"last_seen":"2023-01-03T18:17:11-06:00","linkquality":120,"monitoring_mode":"undirected","motion_sensitivity":"high","power_outage_count":895,"presence":true,"presence_event":"away","update":{"installed_version":"unknown","state":"idle"},"update_available":null}' with variables {'entity_id': 'update.fp1b', 'name': 'FP1B', 'this': <template TemplateStateFromEntityId(update.fp1b)>} and Template("{{ value_json['update']['latest_version'] }}")
2023-01-03 18:17:11.466 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'
```

GerdRuetten, 2023-01-04:
It's the same error as #15747 

Should be merged

Koenkk, 2023-01-04:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/15747

closed 2023-01-04 by Koenkk