Koenkk/zigbee2mqtt #15933: Overridden entity types aren't disabled

deviantintegral, 2023-01-04:
### What happened?

I disabled a device as added in https://github.com/Koenkk/zigbee2mqtt/issues/10003. While the various diagnostic entities were disabled, the main switch was not.

### What did you expect to happen?

I expected all entities to be disabled.

As well, I'm not finding the docs (searching for "disable") on how exactly this is supposed to work. Does it just disable the entities in HA and the controls in the z2m dashboard? Does it affect networking or the mesh at all? In my case, I'm using a plug for my Christmas tree lights, and since I'd usually unplug it I'm getting the sense that this would really be best for lights that are on Zigbee directly.

### How to reproduce it (minimal and precise)

Here's my configuration snippet:

```
  '0xb0ce1814036bfb11':
    friendly_name: Christmas Tree
    homeassistant:
      switch:
        type: light
        object_id: light
      light:
        name: Christmas Tree
        state_value_template: '{{ value_json.state }}'
    disabled: true
    legacy: false
    optimistic: true
```

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20221102

### Adapter

zzh!

Koenkk, 2023-01-04:
The zigbee2mqtt disabled feature does not disable devices in HA. AFAIK this is not possible since z2m cannot change from enabled -> disabled after initial discovery, this is managed within HA.

deviantintegral, 2023-01-06:
Reasonable enough. I ended up deleting the device in zigbee2mqtt. Thank you!

closed 2023-01-06 by deviantintegral