Koenkk/zigbee2mqtt #15939: Device type detected incorrectly

JanPannier, 2023-01-04:
### What happened?

I want to add a TuYa 2 gang dimmer module (https://www.zigbee2mqtt.io/devices/TS110E_2gang.html). However, it detects a s a new, unsupported device.

Here's the entry in database.db:
`{"id":54,"type":"Router","ieeeAddr":"0xa4c13859cd06cc9b","nwkAddr":64398,"manufId":4417,"manufName":"_TZ3000_zjtxnoft","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAA"}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"switchType":0}},"genLevelCtrl":{"attributes":{"currentLevel":254,"minLevel":2,"maxLevel":254,"onOffTransitionTime":50}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345,8],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"attributes":{"currentLevel":254,"minLevel":2,"maxLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1672833303629,"defaultSendRequestWhen":"immediate"}`

### What did you expect to happen?

That it would detect the correct device

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20210120

### Adapter

zzh!

sisalik, 2023-01-18:
I just ran into the same issue with [these](https://www.aliexpress.com/item/1005004766854041.html?spm=a2g0o.order_list.order_list_main.19.16d0a396MTKYK6) 2ch Zigbee dimmers from AliExpress which I assumed were TS1101/TS110E but they came up as TS0052 as well. I kind of worked around it by creating a new converter for them ([gist link](https://gist.github.com/sisalik/ba4568a1fa930eb87d4d7b09639373df)), by basically copying the code from the `TS1101_dimmer_module_2ch` device in [tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js).

Have a go and see if this helps. It mostly works for me - I can toggle each of the two channels and control their brightness values. However, the behaviour is kind of buggy, e.g. the lights sometimes get stuck on or off and the fade transition time setting doesn't quite work. Not sure if it's because my cobbled-together converter is missing something or if the device firmware is weird.

closed 2023-02-26 by github-actions[bot]

dartviper, 2023-03-06:
I've got the same problem, but only with a 1 ch. Zigbee Dimmer. The solution of sisalik didn't help to resolve my problem. Does anyone have a solution for this problem?