Koenkk/zigbee2mqtt #15940: Fails to start after restart

fuzzy01, 2023-01-04:
### What happened?

Running docker version of zigbee2mqtt. Every second restart fails. It stalls after opening the serial port. See debug log. It shows nothing after the port was opened. If I stop it and start it again it starts successfully.

### How to reproduce it (minimal and precise)

After a successful start stop it and start is again immediately.

### Zigbee2MQTT version

1.29.0-dev commit: 15ccf94

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle-E

### Debug log

2023-01-04T12:14:18.091Z zigbee-herdsman:adapter:ezsp:ezsp Close ezsp
2023-01-04T12:14:18.091Z zigbee-herdsman:adapter:ezsp:driv Close driver
Zigbee2MQTT:info  2023-01-04 13:14:18: Stopped zigbee-herdsman
Zigbee2MQTT:info  2023-01-04 13:14:18: Stopped Zigbee2MQTT
Using '/app/data' as data directory
Zigbee2MQTT:debug 2023-01-04 13:14:25: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2023-01-04 13:14:25: Logging to console and directory: '/app/data/log/2023-01-04.13-14-25' filename: log.txt
Zigbee2MQTT:debug 2023-01-04 13:14:25: Removing old log directory '/app/data/log/2023-01-02.15-55-20'
Zigbee2MQTT:info  2023-01-04 13:14:25: Starting Zigbee2MQTT version 1.29.0-dev (commit #15ccf94)
Zigbee2MQTT:info  2023-01-04 13:14:25: Starting zigbee-herdsman (0.14.83)
Zigbee2MQTT:debug 2023-01-04 13:14:25: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
2023-01-04T12:14:26.530Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[236,233,34,25,123,122,26,170,106,184,123,172,226,28,220,36],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"/dev/ttyACM0","adapter":"ezsp"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2023-01-04T12:14:26.559Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
2023-01-04T12:14:26.575Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2023-01-04T12:14:26.575Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2023-01-04T12:14:26.576Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2023-01-04T12:14:26.578Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2023-01-04T12:14:26.591Z zigbee-herdsman:adapter:ezsp:uart <-- [0a0bb1a96b2a164fa674eb04aaa76d499db3d8a4e7ce67bc5d7e0a0bb1a96b2a164fa674eb04aaa76d499db3d8a4e7ce67bc5d7e0a0bb1a96b2a164fa674eb04aaa76d499db3d8a4e7ce67bc5d7ec20251a8bd7e]
2023-01-04T12:14:26.605Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,2,1): 0a0bb1a96b2a164fa674eb04aaa76d499db3d8a4e7ce67bc5d7e
2023-01-04T12:14:26.606Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2023-01-04T12:14:26.606Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2023-01-04T12:14:26.607Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 2 | 0
2023-01-04T12:14:26.608Z zigbee-herdsman:adapter:ezsp:uart Skipping the packet as repeated (1)
2023-01-04T12:14:26.609Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,2,1): 0a0bb1a96b2a164fa674eb04aaa76d499db3d8a4e7ce67bc5d7e
2023-01-04T12:14:26.609Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2023-01-04T12:14:26.609Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2023-01-04T12:14:26.610Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 2 | 0
2023-01-04T12:14:26.610Z zigbee-herdsman:adapter:ezsp:uart Skipping the packet as repeated (1)
2023-01-04T12:14:26.610Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,2,1): 0a0bb1a96b2a164fa674eb04aaa76d499db3d8a4e7ce67bc5d7e
2023-01-04T12:14:26.610Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2023-01-04T12:14:26.610Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2023-01-04T12:14:26.611Z zigbee-herdsman:adapter:ezsp:uart Unexpected packet sequence 2 | 0
2023-01-04T12:14:26.611Z zigbee-herdsman:adapter:ezsp:uart Skipping the packet as repeated (1)
2023-01-04T12:14:26.611Z zigbee-herdsman:adapter:ezsp:uart <-- Error: c20251a8bd7e
2023-01-04T12:14:26.611Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2023-01-04T12:14:26.611Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2023-01-04T12:14:26.611Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
2023-01-04T12:14:26.704Z zigbee-herdsman:adapter:ezsp:uart <-- [11]
2023-01-04T12:14:27.705Z zigbee-herdsman:adapter:ezsp:uart <-- [c1020b0a527e]
2023-01-04T12:14:27.706Z zigbee-herdsman:adapter:ezsp:uart <-- RSTACK: c1020b0a527e
2023-01-04T12:14:27.707Z zigbee-herdsman:adapter:ezsp:uart RSTACK Version: 2 Reason: RESET_SOFTWARE frame: c1020b0a527e

CrazyCoder, 2023-01-23:
Looks like https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/431.

chrislentz, 2023-02-23:
I am still experiencing the issue with a Sonoff Dongle-E on firmware version 7.2.0.0.

mamrai1, 2023-02-23:
try 7.1.4.0 build 389 .

chrislentz, 2023-02-24:
> try 7.1.4.0 build 389 .

Confirmed, the issue still occurs with [ncp-uart-sw_7.1.4.0_115200.gbl](https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.1.4.0_115200.gbl).

fuzzy01, 2023-03-13:
Looks like it is fixed by the latest dev build.
Maybe it was fixed by https://github.com/Koenkk/zigbee-herdsman/pull/669

closed 2023-03-13 by fuzzy01