Koenkk/zigbee2mqtt #15941: Namron 540139X mode and temperature settings stopped working

hmorkemo, 2023-01-04:
### What happened?

After the last update it's impossible to turn the panel heater on or off, and to set the thermostat temperature.

### What did you expect to happen?

Setting mode=0 (off) should turn the heater off, and setting the temperature should set the desired temperature.

### How to reproduce it (minimal and precise)

Start zigbee2mqtt from HA and select the panel heater.
Go to the exposes tab and try to change the mode or temperature.
You'll get an error message immediately, like these:
```
Error 2023-01-04 14:04:37Publish 'set' 'system_mode' to 'Kjøkkenovn' failed: 'Error: Write 0x3fff000400000518/1 hvacThermostat({"systemMode":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'
Error 2023-01-04 14:16:41Publish 'set' 'occupied_heating_setpoint' to 'Kjøkkenovn' failed: 'Error: Write 0x3fff000400000518/1 hvacThermostat({"occupiedHeatingSetpoint":650}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'FAILURE')'
```

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 flashed. zStack3x0

hmorkemo, 2023-01-05:
Help, please. Is there a way that I can test / work around this problem? The panel heaters (2) are in my cabin at a remote location. It's quite inaccessible, and the water pipes will freeze if I can't turn on the heat... Any detailed info on using e.g. the dev console?

hmorkemo, 2023-01-07:
Tested in 1.29.1-1. Same problem.

Koenkk, 2023-01-08:
What was the last known working version, 1.28.4?

hmorkemo, 2023-01-08:
I'm not sure about the version number, but it was installed the first week of december. Then it worked perfectly.

Koenkk, 2023-01-09:
Can you try to revert and confirm it is still?

hmorkemo, 2023-01-09:
Funny thing happened. I came to my cabin just now and adjusted the heaters manually. After that I can adjust and turn them both on/off from Home Assistant/z2m! I also unplugged and re-plugged them to see if the problem had been caused by a power failure. I can still control them, so that wasn't the problem.
So I have no idea what has caused it, and not if/when it will return. Might it be some sleep/wake up problem?

hmorkemo, 2023-02-05:
The problem is still there. It has reappeared after the last time I was at the cabin.
But: I restored version 1.28.2-1, and the problem is the same with that version too!
So I guess it's a Namron problem.

closed 2023-03-18 by github-actions[bot]