Koenkk/zigbee2mqtt #15942: impossible to force pairing on a particular device

apages2, 2023-01-04:
### What happened?

Hello,

I am trying to pair Xiaomi end device on the coordinator. in zigbee2mqtt, I select the coordinator only in the pairing and I activate it. After having paired the xiaomi end device, in the schema tab, I see that this equipment is paired with a router other than the coordinator (a legrand router). I think there is a problem with this feature

### What did you expect to happen?

the end device must be paired with the selected device (coordinator, specific router, etc...)

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

 debug 2023-01-04 14:34:50Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":"Coordinator","time":254,"transaction":"ac8sw-18","value":true}'
info 2023-01-04 14:34:50Zigbee: allowing new devices to join via Coordinator.
info 2023-01-04 14:34:50MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"Coordinator","time":254,"value":true},"status":"ok","transaction":"ac8sw-18"}'
debug 2023-01-04 14:35:04Received Zigbee message from 'Physique Temp Bureau', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3005,"10":18959,"100":2116,"101":6145,"102":100969,"4":424,"5":85,"6":[0,1]}}' from endpoint 1 with groupID null
debug 2023-01-04 14:35:04lumi.weather: unknown key 6 with value 0,1
debug 2023-01-04 14:35:04lumi.weather: Processed data into payload {"voltage":3005,"battery":100,"power_outage_count":84,"temperature":21.16,"humidity":61.45,"pressure":1009.7}
debug 2023-01-04 14:35:04lumi.weather: Processed data into payload {"voltage":3005,"battery":100,"power_outage_count":84,"temperature":21.16,"humidity":61.45,"pressure":1009.7}
info 2023-01-04 14:35:04MQTT publish: topic 'zigbee2mqtt/Physique Temp Bureau', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20191205","friendlyName":"Physique Temp Bureau","hardwareVersion":30,"ieeeAddr":"0x00158d00048000a6","manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","networkAddress":30316,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"humidity":61.45,"linkquality":248,"power_outage_count":84,"pressure":1009.7,"temperature":21.16,"voltage":3005}'
debug 2023-01-04 14:35:08Received Zigbee message from 'Physique Temp Bureau', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2121}' from endpoint 1 with groupID null
info 2023-01-04 14:35:08MQTT publish: topic 'zigbee2mqtt/Physique Temp Bureau', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20191205","friendlyName":"Physique Temp Bureau","hardwareVersion":30,"ieeeAddr":"0x00158d00048000a6","manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","networkAddress":30316,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"humidity":61.45,"linkquality":244,"power_outage_count":84,"pressure":1009.7,"temperature":21.21,"voltage":3005}'
debug 2023-01-04 14:35:08Received Zigbee message from 'Physique Temp Bureau', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6195}' from endpoint 1 with groupID null
info 2023-01-04 14:35:08MQTT publish: topic 'zigbee2mqtt/Physique Temp Bureau', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20191205","friendlyName":"Physique Temp Bureau","hardwareVersion":30,"ieeeAddr":"0x00158d00048000a6","manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","networkAddress":30316,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"humidity":61.95,"linkquality":255,"power_outage_count":84,"pressure":1009.7,"temperature":21.21,"voltage":3005}'
debug 2023-01-04 14:35:08Received Zigbee message from 'Physique Temp Bureau', type 'attributeReport', cluster 'msPressureMeasurement', data '{"measuredValue":1009,"scale":-1,"scaledValue":10096}' from endpoint 1 with groupID null
info 2023-01-04 14:35:08MQTT publish: topic 'zigbee2mqtt/Physique Temp Bureau', payload '{"battery":100,"device":{"applicationVersion":5,"dateCode":"20191205","friendlyName":"Physique Temp Bureau","hardwareVersion":30,"ieeeAddr":"0x00158d00048000a6","manufacturerID":4151,"manufacturerName":"LUMI","model":"WSDCGQ11LM","networkAddress":30316,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"humidity":61.95,"linkquality":255,"power_outage_count":84,"pressure":1009.6,"temperature":21.21,"voltage":3005}'
debug 2023-01-04 14:35:11Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":"Coordinator","time":254,"transaction":"ac8sw-19","value":false}'
info 2023-01-04 14:35:11Zigbee: disabling joining new devices.

danieledwardgeorgehitchcock, 2023-01-04:
Zigbee is a self-healing network so, end devices will always hop around different routers depending on the strongest route back to the coordinator. You can't stop that, it's baked into the protocol itself

apages2, 2023-01-04:
the problem is that some end device are incompatible with some router. this is the case for example with Xiaomi end devices with Legrand routers. How to get rid of these problems?

closed 2023-02-11 by github-actions[bot]