Koenkk/zigbee2mqtt #15946: For newly added TS0601_soil TuYa Soil sensor: expose soil_moisture instead of humidity

f0gster, 2023-01-04:
### What happened?

1st of, thanks for adding TS0601_soil TuYa Soil sensor.
I have the device, so I played with an external converter before 1.29. It was awfully similar to your added code, apart of one thing: I'm suggesting changing humidity, that you are currently exposing, to soil_moisture.

All existing soil sensors are exposing soil_moisture (some also humidity on top of that).
TuYa Soil sensor is clearly not sending air humidity - that is easy to check. When I have a sensor in the air (not in soil), for example just near me, it will send 0. Obviously, air humidity is not 0 (all other sensors exposing humidity will show something different).

Hope it makes sense to you.

### What did you expect to happen?

Expose soil_moisture 

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20221102

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus CC2652P

Koenkk, 2023-01-05:
@kris969 @cloudbr34k84  what do you think?

cloudbr34k84, 2023-01-05:
> @kris969 @cloudbr34k84  what do you think?
I'm not fussed, makes sense tbh, but doesn't bother me either way

kris969, 2023-01-05:
I have no idea yet on how it's reported on other soil sensors...
Anyway it could make sense.

The exact is description is:  `Tuya Zigbee Wireless Soil Moisture Meter Temperature Humidity`, so base on this everything is possible!

A suggestion should be to expose both soil_moisture and humidity with, of course, the same report value. This way, everyone will be ok, and existing usages will not be impacted.
But, @Koenkk I don't know if it will be tricky for you to do this?

kris969, 2023-01-05:
@f0gster : I agree!

Koenkk, 2023-01-06:
Done!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-06 by Koenkk

kris969, 2023-01-10:
Hi,

A rapid check with the dev branch reveale an issue.
no soil_moisture was reported!

I had to pair my sensors again to have it working fine.
This is just a comment for everyone. If you started with previous z2m
version, you will need to delete your sensor from z2m then repair it
again.

kris969, 2024-02-10:
I use this channel to address this comment to soil sensor'users.
This comment is not relative to an z2m issue, but only talk about a sensor issue.
It concern battery drain with such sensor that should last more than one year, in reality after around 3 months batteries are low. I'm not sure the sensor use the hibernate mode and I notice it send report values every minutes that is really to much. I tried to find a new firmware for this device but I have found none.
Did someone has found a fix or a workaround to this sensor issues?

here are information reportes by z2m:
 "manufacturerID": 4417
"hardwareVersion": 1
"applicationVersion": 72
 "dateCode": ""
 "stackVersion": 0
 "zclVersion": 3

Hellingajes, 2024-06-18:
Hello, I got the new version of this device, they said the battery issue is fixed with this version. After switching z2m to dev branch I was able to pair it succesfully, however, the states are not reporting, only lqi, the rest says: null
What can I do to fix it?

engusz, 2024-07-06:
> Hello, I got the new version of this device, they said the battery issue is fixed with this version. After switching z2m to dev branch I was able to pair it succesfully, however, the states are not reporting, only lqi, the rest says: null What can I do to fix it?

Same here :(

Hellingajes, 2024-07-07:
> Hello, I got the new version of this device, they said the battery issue is fixed with this version. After switching z2m to dev branch I was able to pair it succesfully, however, the states are not reporting, only lqi, the rest says: null What can I do to fix it?

After a few days, it fixed it self!
Perhaps a reminder to move this device to main branch?