Koenkk/zigbee2mqtt #15950: Sinopé SP2600ZB not reporting  consumed energy - Null value

maxperron, 2023-01-04:
### What happened?

Hello, I just bought 2 [Sinopé SP2600ZB](https://www.zigbee2mqtt.io/devices/SP2600ZB.html) switches. I added them to my network and the exposed energy attributes is always null kWh (I waited 48 hours).

[image]

Switch details:
[image]

### What did you expect to happen?

As per the documentation, I should see a value for the sum of consumed energy.

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0

### Debug log

MQTT publish: topic 'zigbee2mqtt/BUR Server Switch Switch Sinope', payload '{"ac_frequency":60,"current":0.19,"energy":null,"linkquality":116,"power":13,"state":"ON","voltage":116.62}'

maxperron, 2023-01-06:
I'm now on 1.29.1 version and still have the issue.

maxperron, 2023-01-14:
In the meantime I created a custom sensor that calculate the energy based on the current watt.

Koenkk, 2023-01-14:
@mattL0 could you check this?

maxperron, 2023-02-10:
Thanks, it's working!

closed 2023-02-10 by maxperron