Koenkk/zigbee2mqtt #15951: ZG-227ZL (TS0601 Temperature) : Temperature doesn't change after calibration modification

bamnoru, 2023-01-04:
### What happened?

When modifying the temperature calibration, the temperature sent by the sensor doesn't change, even on new value sent by the sensor.

### What did you expect to happen?

When calibration is changed, the next value sent by the sensor should be adjust with the value.

### How to reproduce it (minimal and precise)

Change the calibration value, wait for next refresh, check if the new temperature.

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF ZB Dongle-E

bamnoru, 2023-01-04:
Calibration from "Expose" tab is working only when a value is set in the Calibration value from the "Setting Specifics" tab.
By default, the calibration value is not set.

KenGuru123, 2023-01-13:
Can you explain in detail what you mean? I've the same problem, and can't solve it.
Do you mean, that both values have to be the same? Playing around with the values, but everything seems to be messed up right now.
Z2M are different values, than displayed on the device...

bamnoru, 2023-01-15:
Hi,
I've thought that the calibration was taken into account on next temperature refresh, but, after some days of test, I'm not sure.
It seems that this device send an update of the temperature when the difference is > 0.3°C.
I don't know if the poor accuracy is an issue of the device or something else.

But, there's still a problem, the temperature calibration change does'nt affect the temperature:

info  2023-01-15 09:23:15: MQTT publish: topic 'homeassistant/sensor/0xa4c1383e2f776a05/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1383e2f776a05"],"manufacturer":"TuYa","model":"Temperature & humidity LCD sensor (ZG-227ZL)","name":"Sonde Amory"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"Sonde Amory last seen","state_topic":"zigbee2mqtt/Sonde Amor ","unique_id":"0xa4c1383e2f776a05_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
info  2023-01-15 09:27:04: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":60,"humidity_calibration":0,"last_seen":"2023-01-15T09:27:04+01:00","linkquality":184,"temperature":20,"temperature_calibration":0,"temperature_unit":"celsius"}'
info  2023-01-15 09:27:05: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:27:05+01:00","linkquality":180,"temperature":20,"temperature_calibration":0,"temperature_unit":"celsius"}'
info  2023-01-15 09:34:07: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:27:05+01:00","linkquality":180,"temperature":20,"temperature_calibration":0.5,"temperature_unit":"celsius"}'
debug 2023-01-15 09:36:09: Received MQTT message on 'zigbee2mqtt/Sonde Amory/set/temperature_calibration' with data '0.3'
debug 2023-01-15 09:36:09: Publishing 'set' 'temperature_calibration' to 'Sonde Amory'
info  2023-01-15 09:36:09: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:27:05+01:00","linkquality":180,"temperature":20,"temperature_calibration":0.3,"temperature_unit":"celsius"}'
debug 2023-01-15 09:37:49: Received Zigbee message from 'Sonde Amory', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":4}],"seq":14336}' from endpoint 1 with groupID 0
info  2023-01-15 09:37:49: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:37:49+01:00","linkquality":172,"temperature":20,"temperature_calibration":0.3,"temperature_unit":"celsius"}'
debug 2023-01-15 09:38:43: Received MQTT message on 'zigbee2mqtt/Sonde Amory/set/temperature_calibration' with data '0.5'
debug 2023-01-15 09:38:43: Publishing 'set' 'temperature_calibration' to 'Sonde Amory'
info  2023-01-15 09:38:43: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:37:49+01:00","linkquality":172,"temperature":20,"temperature_calibration":0.5,"temperature_unit":"celsius"}'
debug 2023-01-15 09:42:38: Received Zigbee message from 'Sonde Amory', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,200],"type":"Buffer"},"datatype":2,"dp":1}],"seq":14592}' from endpoint 1 with groupID 0
info  2023-01-15 09:42:38: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:42:38+01:00","linkquality":184,"temperature":20,"temperature_calibration":0.5,"temperature_unit":"celsius"}'
debug 2023-01-15 09:42:38: Received Zigbee message from 'Sonde Amory', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,59],"type":"Buffer"},"datatype":2,"dp":2}],"seq":14848}' from endpoint 1 with groupID 0
info  2023-01-15 09:42:38: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:42:38+01:00","linkquality":188,"temperature":20,"temperature_calibration":0.5,"temperature_unit":"celsius"}'
debug 2023-01-15 09:44:22: Received Zigbee message from 'Sonde Amory', type 'attributeReport', cluster 'genBasic', data '{"65506":34,"65508":0,"appVersion":67}' from endpoint 1 with groupID 0
info  2023-01-15 09:44:22: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:44:22+01:00","linkquality":188,"temperature":20,"temperature_calibration":0.5,"temperature_unit":"celsius"}'
debug 2023-01-15 09:45:07: Received Zigbee message from 'Sonde Amory', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
info  2023-01-15 09:45:07: MQTT publish: topic 'zigbee2mqtt/Sonde Amory', payload '{"battery":100,"humidity":59,"humidity_calibration":0,"last_seen":"2023-01-15T09:45:07+01:00","linkquality":192,"temperature":20,"temperature_calibration":0.5,"temperature_unit":"celsius"}'

closed 2023-02-22 by github-actions[bot]