Koenkk/zigbee2mqtt #15952: Add sensor device status offline\online

DivanX10, 2023-01-04:
### Is your feature request related to a problem? Please describe

It happens that devices fall off the network and we don't know about it until something works, so it's useful to monitor devices for unavailability. In zigbee2mqtt there is an item about the availability of the device and the time is indicated that if the device is unavailable for more than 1500 minutes, the status will be offline. This status is visible in every device via MQTT Explorer, but only last seen, linkquality, voltage, battery sensors are available in zigbee device objects. Could you add a device accessibility sensor? This status sensor will display the current status of the device. I also ask you to enable by default the availability of the last seen sensor for all devices, otherwise this sensor is deactivated by default

[image]

Now, in order to monitor whether the device is available, you need to activate the last seen sensor in each device, and if the device was removed and added in a new way, then you need to activate this last seen sensor again, which is very inconvenient, and then create such a sensor

We monitor the availability of devices
```
Number of unavailable devices
{% set lapsed_hours = 36 %}
{% set ns = namespace(count=0) %}
{%- for state in states.sensor | selectattr('entity_id', 'search', '.*_last_seen$') -%}
  {%- if states(state.entity_id) == 'unavailable' or ((as_timestamp(now()) - as_timestamp(states(state.entity_id),0)) > ((lapsed_hours | int) * 60 * 60)) -%}
    {%- set ns.count = ns.count + 1 -%}
  {%- endif -%}
{%- endfor -%}
Кол-во:{{ ns.count }}     

Which devices are not available
{% set lapsed_hours = 36 %}
{% set result = namespace(sensors=[]) %}
{%- for state in states.sensor | selectattr('entity_id', 'search', '.*_last_seen$') -%}
  {%- if states(state.entity_id) == 'unavailable' or ((as_timestamp(now()) - as_timestamp(states(state.entity_id),0)) > ((lapsed_hours | int) * 60 * 60)) -%}
    {%- set result.sensors = result.sensors + [state.name | regex_replace(find='_last_seen', replace='') ~ 
     ' (' ~ relative_time(strptime(states(state.entity_id), '%Y-%m-%dT%H:%M:%S%z', 'unavailable')) ~ ')'] -%}
  {%- endif -%}
 {%- endfor -%}
{{ result.sensors }}
```

### Describe the solution you'd like

So that the device displays an accessibility sensor

### Describe alternatives you've considered

You can, for example, create an mqtt sensor and extract data from the topic, but prescribing for each device, for me, is not the best option.

[image]

Kitchen: TIV - TIV this is abbreviated temperature and humidity.  Sensor Xiaomi WSDCGQ11LM

```
mqtt:
  sensor:
    - unique_id: monitoring device kitchen tiv
      object_id: monitoring device kitchen tiv
      name: "Monitor the sensor Kitchen: TIV"
      state_topic: "zigbee2mqtt_sonoff_usb_dongle/Kitchen: TIV/availability"
      value_template: "{{ value_json.state }}"
      icon: mdi:zigbee
```

### Additional context

without information

DivanX10, 2023-02-04:
we are waiting for a response from the author of the zigbee2mqtt project

DivanX10, 2023-03-07:
The issue is still unresolved. Please pay attention to my request and implement it if possible

DivanX10, 2023-04-09:
Please do not close the topic, as I am waiting for an answer from the author

closed 2023-06-18 by github-actions[bot]

dan2cil, 2024-07-26:
Why none of the team of zigbee2mqtt is responding, I have the some necessity too