Koenkk/zigbee2mqtt #15957: Device Pairs as Incorrect Model

NexGen-3D-Printing, 2023-01-05:
### What happened?

Device paired as incorrect model: 

- TS0001

Exposes only:

- power_on_behavior
- state

### What did you expect to happen?

Device should have paired as: 

- TS0001_switch_module

Device should expose:

- power_on_behavior
- state
- Switch_type

### How to reproduce it (minimal and precise)

Unpair and Repair

### Zigbee2MQTT version

1.29.0-dev commit: cf801344

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

Koenkk, 2023-01-05:
Can you provide the data/database.db entry of this device?

NexGen-3D-Printing, 2023-01-06:
Hi Bud, not sure exactly where the start is for the device, so I just grabbed the last chunk of the data base, if you need anything else, let me know.

"TRADFRI control outlet","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,${"id":47,"type":"Router","ieeeAddr":"0x00158d0003553cd0","nwkAddr":38303,"manufId":4151,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.relay.c2acn01","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,1,2,10,6,16,${"id":48,"type":"Router","ieeeAddr":"0xa4c138dd08ec3209","nwkAddr":36635,"manufId":4417,"manufName":"_TZ3000_gjrubzje","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,$

Koenkk, 2023-01-06:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-06 by Koenkk

NexGen-3D-Printing, 2023-01-06:
Awesome, I'll test it out in the morning and provide feedback.

NexGen-3D-Printing, 2023-01-06:
You sir, are a legend, updated and the device detected correctly and the switch type is fully functional.