Koenkk/zigbee2mqtt #15958: Philips 9290012607: Entity sensor.xxx_illuminance (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('illuminance') it is using

GerdRuetten, 2023-01-05:
### What happened?

After update HA to Verion 2023.1.0, i get in the HA logs for the device https://www.zigbee2mqtt.io/devices/9290012607.html following error:

```
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:980
Integration: Sensor (documentation, issues)
First occurred: 11:29:30 (1 occurrences)
Last logged: 11:29:30

Entity sensor.bwm_og_dg_illuminance (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('illuminance') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

### What did you expect to happen?

No error :-)

### Zigbee2MQTT version

1.29.0-dev commit: cf80134

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N)

GerdRuetten, 2023-01-05:
The device exposes this:

```
{
  "battery": 44,
  "illuminance": 0,
  "illuminance_lux": 0,
  "last_seen": "2023-01-04T21:34:28.493Z",
  "led_indication": true,
  "linkquality": 18,
  "motion_sensitivity": "high",
  "occupancy": false,
  "occupancy_timeout": 0,
  "temperature": 19.8,
  "update": {
    "installed_version": 1107323831,
    "latest_version": 1107323831,
    "state": "idle"
  },
  "update_available": false
}
```

With the new HA-Version the device_class and the unit_of_measurement must be correct, see:

https://www.home-assistant.io/integrations/sensor/#device-class

The "illuminance"-value is the raw measured for illuminance. It has no unit.
So i think the device_class must set to "None".

tsspmq, 2023-01-05:
This is a change in HA recently, nothing to do with Z2M.  I have Z-Wave devices that do this with illuminance and opened an issue in the HA branch.

GerdRuetten, 2023-01-05:
> This is a change in HA recently, nothing to do with Z2M. I have Z-Wave devices that do this with illuminance and opened an issue in the HA branch. Those devices I have report illuminance in %, not lux. I think there were a few other issues opened for other sensor types and fixed already.

I don't think so.

The unit of measurement are not in the zigbee stream and not in mqtt.

In the preparation of the HA sensors, this is attached by z2m.
So it must be change here.

serlinGi, 2023-01-05:
I think it doesn't depend on Z2M;  I have the same warning for several sensors, both with Z2M and with tasmota devices created with my yaml files;
  no errors detected prior to version 2023.1.0
[image]

dedors, 2023-01-06:
I have the same with two Xiaomi GZCGQ01LM light sensors:

```
Entity sensor.sensor_lumen_1_illuminance (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('illuminance') it is using; 
Entity sensor.sensor_lumen_2_illuminance (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '' which is not a valid unit for the device class ('illuminance') it is using; 
```

There is also a issue raised at home assistant: [here](https://github.com/home-assistant/core/issues/85262)

Koenkk, 2023-01-08:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

GerdRuetten, 2023-01-09:
The problem is still there

After updating to the latest z2m edge version and restart from HA the error message has changed slightly.

`Entity sensor.bwm_og_dg_illuminance (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement **'None'** which is not a valid unit for the device class ('illuminance') it is using; Please update your configuration if your entity is manually configured,`

The HA entity looks like:

[image]

Koenkk, 2023-01-09:
Can you provide the debug log when starting z2m?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

GerdRuetten, 2023-01-09:
> Can you provide the debug log when starting z2m?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

I upload it

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10376383/log.txt)

Koenkk, 2023-01-10:
@frenck I think:

```
Entity sensor.bwm_og_dg_illuminance (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement **'None'** which is not a valid unit for the device class ('illuminance') it is using; Please update your configuration if your entity is manually configured,
```
is caused by the lack of `unit_of_measurement` (which is optional according to the [docs](https://www.home-assistant.io/integrations/sensor.mqtt/#unit_of_measurement)). But is it also optional when `"state_class":"measurement"`? 

Full discovery payload for future reference:

```
{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/BWM OG/DG/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00178801086331ec"],"manufacturer":"Philips","model":"Hue motion sensor (9290012607)","name":"BWM OG/DG","sw_version":"6.1.1.27575"},"device_class":"illuminance","enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/BWM OG/DG","name":"BWM OG/DG illuminance","state_class":"measurement","state_topic":"zigbee2mqtt/BWM OG/DG","unique_id":"0x00178801086331ec_illuminance_zigbee2mqtt","value_template":"{{ value_json.illuminance }}"}
```

frenck, 2023-01-10:
> is caused by the lack of unit_of_measurement 

Correct.

The unit of measurement is optional, but once a device class is set, we only accept a subset of units. This has always been the case, but only recently have we started adding runtime checks for this (which currently result in a warning log).

Ref: <https://developers.home-assistant.io/docs/core/entity/sensor#available-device-classes>

Koenkk, 2023-01-11:
@frenck thansk!

Issue should be fixed now, please confirm.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

frenck, 2023-01-11:
That linked commit seems wrong. State class and UoM are not related?

GerdRuetten, 2023-01-12:
> @frenck thansk!
> 
> Issue should be fixed now, please confirm.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

I test it with Zigbee2MQTT Version
1.29.1-dev commit: [328b9fb](https://github.com/Koenkk/zigbee2mqtt/commit/328b9fb) and the error still available

Log file attached

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10398825/log.txt)

Koenkk, 2023-01-12:
@frenck whoops, I was confused. Thanks for your sharp eyes, fixed it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

GerdRuetten, 2023-01-12:
> test it with Zigbee2MQTT Version
> 1.29.1-dev commit: [328b9fb](https://github.com/Koenkk/zigbee2mqtt/commit/328b9fb) and the error still available

test it with Zigbee2MQTT Version
1.29.1-dev commit: [1d8b19d](https://github.com/Koenkk/zigbee2mqtt/commit/1d8b19d) 

and the error still available

GerdRuetten, 2023-01-13:
```
// When state_class == measurement, unit_of_measurement must be set, otherwise warnings are generated.
            // https://github.com/Koenkk/zigbee2mqtt/issues/15958#issuecomment-1377483202
            if (discoveryEntry.discovery_payload.state_class === 'measurement' &&
                !discoveryEntry.discovery_payload.unit_of_measurement) {
                delete discoveryEntry.discovery_payload.state_class;
            }

``` 

Wouldn't it be better to delete the "device_class" if there is no "unit_of_measurement" instead of the "state_class"?

Koenkk, 2023-01-14:
@GerdRuetten is this issue fixed now?

GerdRuetten, 2023-01-14:
> @GerdRuetten is this issue fixed now?

Zigbee2MQTT Version
[1.29.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.29.2)

error still available

Zigbee2MQTT Version
1.29.2-dev commit: [2c509c0](https://github.com/Koenkk/zigbee2mqtt/commit/2c509c0)

error is fixed

GerdRuetten, 2023-01-14:
But there is a second error

```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:385
Integration: Sensor (documentation, issues)
First occurred: 08:55:10 (3 occurrences)
Last logged: 08:55:10

The unit of sensor.rauchmelder_schlafzimmer_smoke_density (None) can not be converted to the unit of previously compiled statistics (). Generation of long term statistics will be suppressed unless the unit changes back to or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.rauchmelder_marvin_smoke_density (None) can not be converted to the unit of previously compiled statistics (). Generation of long term statistics will be suppressed unless the unit changes back to or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
The unit of sensor.bwm_og_dg_illuminance (None) can not be converted to the unit of previously compiled statistics (). Generation of long term statistics will be suppressed unless the unit changes back to or a compatible unit. Go to https://my.home-assistant.io/redirect/developer_statistics to fix this
```

Should i open a new issue?

Best regards

Gerd

Koenkk, 2023-01-14:
@GerdRuetten this error is because the unit is changed from an empty string to none now. I suggest that you clear the old data which should get rid of this error.

GerdRuetten, 2023-01-14:
as suggested, i deleted the old data via https://my.home-assistant.io/redirect/developer_statistics.

there is no more error.

good job

i close this issue.

closed 2023-01-14 by GerdRuetten

JimGermany, 2023-02-05:
_Zigbee2MQTT Version
1.29.2-dev commit: [2c509c0](https://github.com/Koenkk/zigbee2mqtt/commit/2c509c0)
error is fixed_

Sorry I have to ask again: I just installed the update from 1.29.1-1 to 1.30.0-1 but the error still exists. 

This applies to the Zigbee plugs with Tasmota.

`Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:980
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 13:32:49 (27 occurrences)
Last logged: 13:36:19

Entity sensor.tas5klein_nuc_moni_energy_reactivepower (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'VAr' which is not a valid unit for the device class ('power') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.tas5klein_nuc_moni_energy_apparentpower (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'VA' which is not a valid unit for the device class ('power') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.tas_sd_2_hifitv_energy_factor (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'Cos φ' which is not a valid unit for the device class ('power_factor') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.tas_sd_2_hifitv_energy_reactivepower (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'VAr' which is not a valid unit for the device class ('power') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.tas_sd_2_hifitv_energy_apparentpower (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'VA' which is not a valid unit for the device class ('power') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

Where and how do I have to delete old data under developer_statistic? I don't see any way to delete old data there?