Koenkk/zigbee2mqtt #15959: >  1.28  brakes  completely sonoff  Dongle PRO-E

trackhacs, 2023-01-05:
### What happened?

after update from 1.24 to 1.28 or 1.29 no communication with devices anymore.  pairing basically impossible.  (was working fine with 1.24)

### What did you expect to happen?

normal operation / communication

### How to reproduce it (minimal and precise)

fresh docker compose setup. tested with ConBeeII  ->  joining & communication possible 

fresh docker compose setup. tested with Dongle Pro - E ->  no joining possible: error log:smb://serv/lucian/work/docker/dockerzigbee_V2/logherds.txt

```
zigbee2mqtt2  | Zigbee2MQTT:info  2023-01-05 11:46:46: Zigbee: allowing new devices to join.
zigbee2mqtt2  | Zigbee2MQTT:error 2023-01-05 11:46:46: Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Connection not initialized'
zigbee2mqtt2  | Zigbee2MQTT:debug 2023-01-05 11:46:46: Error: Connection not initialized
zigbee2mqtt2  |     at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:505:19)
zigbee2mqtt2  |     at Driver.permitJoining (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:621:26)
zigbee2mqtt2  |     at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:249:35)
zigbee2mqtt2  |     at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42)
zigbee2mqtt2  |     at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18
zigbee2mqtt2  |     at new Promise (<anonymous>)
zigbee2mqtt2  |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16)
zigbee2mqtt2  |     at EZSPAdapter.permitJoin (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:234:34)
zigbee2mqtt2  |     at Controller.permitJoinInternal (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:246:32)
zigbee2mqtt2  |     at Controller.permitJoin (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:234:20)
zigbee2mqtt2  | Zigbee2MQTT:info  2023-01-05 11:46:46: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{},"error":"Connection not initialized","status":"error","transaction":"9wu3w-1"}'
zigbee2mqtt2 exited with code 1
```

### Zigbee2MQTT version

1.29.0 commit: dddce57

### Adapter firmware version

6.10.3.0 build 297

### Adapter

6.10.3.0 build 297

### Debug log

[logherds.txt](https://github.com/Koenkk/zigbee2mqtt/files/10351462/logherds.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10351466/log.txt)

lavinir, 2023-01-05:
Also running the  Sonoff Zigbee Dongle Plus-E.
Starting experiencing issues after the 1.29 upgrade though (1.28 seems to work fine). Basically after a few hours looks like something breaks and devices go offline. Clicking on "Pairing" reveals that the connection is "Not Initialized"
Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Connection not initialized'

```
Debug 2023-01-05 13:43:40Error: Connection not initialized at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:505:19) at Driver.permitJoining (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:621:26) at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:249:35) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:42) at /app/node_modules/zigbee-herdsman/src/utils/queue.ts:21:18 at new Promise (<anonymous>) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:19:16) at EZSPAdapter.permitJoin (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:234:34) at Controller.permitJoinInternal (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:246:32) at Controller.permitJoin (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:234:20)
```

Restarting the add-on temporarily resolves this. 
Rolled back to 1.28 for now.

cloudbr34k84, 2023-01-05:
im using sonoff dongle 
i just woke up to over 200 entities unavailable as well.  im have had to do a few restarts but there has been alot of device drops off as well as pairing issues since i updated to 1.29.0

Ill persevere for now

trackhacs, 2023-01-05:
which one you have?  E or P?

cloudbr34k84, 2023-01-05:
Sorry I haven't the sonoff dongle P

cloudbr34k84, 2023-01-06:
its causing alot of issues, slow response to devices being toggled.

LookInIs, 2023-01-11:
and I have the same problem
`zigbee-herdsman:adapter:ezsp:erro Request error Error: Connection not initialized: Error: Connection not initialized`

lavinir, 2023-02-08:
Any news on this ? Was it solved in 1.30 ?

aioue, 2023-02-12:
Seeing similar trace on `1.30.1`

```
Zigbee2MQTT:info  2023-02-12 07:33:31: Starting Zigbee2MQTT version 1.30.1 (commit #unknown)
Zigbee2MQTT:info  2023-02-12 07:33:31: Starting zigbee-herdsman (0.14.89)
Zigbee2MQTT:error 2023-02-12 07:33:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-02-12 07:33:51: Failed to start zigbee
Zigbee2MQTT:error 2023-02-12 07:33:51: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-02-12 07:33:51: Exiting...
Zigbee2MQTT:error 2023-02-12 07:33:51: Error: Connection not initialized
    at Ezsp.execCommand (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:518:19)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:325:35)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:141:25)
    at EZSPAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:165:16)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)
```

IgnacioHR, 2023-02-25:
same here. how can I roll back to previous versions?

shooteverything, 2023-03-13:
I'm experiencing the exact same issue using the "E" adaptor. 
Has anyone successfully corrected this?

kirovilya, 2023-03-13:
try with atest z2m release 1.30.2

shooteverything, 2023-03-13:
Currently running 1.30.2

kirovilya, 2023-03-13:
@shooteverything please enable and record herdsman-logs https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging

shooteverything, 2023-03-13:
Thanks for taking the time to comment and attempt to assist! 
I actually decided to totally uninstall and remove z2m and then re-install. 
Re installing the same configuration file, and all seems to be working again. 
I unfortunately have no idea where the issue was, but grateful it is no longer an issue!

IgnacioHR, 2023-03-14:
I run some tests.

I'm using SonOff-e dongle.

I updated to version 1.30.2
launched with DEBUG=zigbee-herdsman* npm start

And I got an error. Here is the detailed log.

[z2mqtt-test.txt](https://github.com/Koenkk/zigbee2mqtt/files/10974286/z2mqtt-test.txt)

Hope this information helps. I've actually 2 zigbee radios and this dongle I'm using just for testing. it is so unstable that I can't consider migrating my network at home

Thank you for your help!

kirovilya, 2023-03-15:
@IgnacioHR you must set `adapter: ezsp` in `serial:` section configuration.yaml

IgnacioHR, 2023-03-15:
@kirovilya thank you!

This is the new error!

[z2mqtt-test.txt](https://github.com/Koenkk/zigbee2mqtt/files/10978244/z2mqtt-test.txt)

Cheers
Ignacio

kirovilya, 2023-03-15:
@IgnacioHR hmmm. try to update to dev/edge version of z2m

IgnacioHR, 2023-03-15:
Now in dev version (I'm running on linux)

New log file:

[z2mqtt-test.txt](https://github.com/Koenkk/zigbee2mqtt/files/10979244/z2mqtt-test.txt)

I'm going to flash again latest SonOFF firmware

kirovilya, 2023-03-15:
@IgnacioHR hmm... try to just pull out and screw the stick into the usb.
I hope your stick is not plugged into USB3 ... you can also try to take the stick out on a USB extension cable.
looking by the logs - a problem with communication with the stick.

IgnacioHR, 2023-03-15:
Unplugged from a USB3 and plugged in to a USB2. Still using the extension cable.

[z2mqtt-test.txt](https://github.com/Koenkk/zigbee2mqtt/files/10979432/z2mqtt-test.txt)

The device was working on the same hardware with firmware from factory. I flashed firmware from here (https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router) which is the official distribution and it is 9 months old! and since then, the device is broken. I'm going to try flashing it again. I was actually waiting for a new firmware version to see if that fixes the issues. Also, there is no previous firmware versions available and when starting in boot mode there is no way to obtain the current firmware version installed.

IgnacioHR, 2023-03-15:
Firmware updated, making sure I'm using the coordinator firmware:

[image]

Device plugged and software restarted

[z2mqtt-test.txt](https://github.com/Koenkk/zigbee2mqtt/files/10979899/z2mqtt-test.txt)

The process ended but this is looking much better now!

kirovilya, 2023-03-15:
@IgnacioHR Now it starts normally, but due to problems with the MQTT it stops. Solve the problem with MQTT and everything will be ok.
```
Zigbee2MQTT:error 2023-03-15 13:33:46: MQTT error: connect ECONNREFUSED 127.0.0.1:1883
Zigbee2MQTT:error 2023-03-15 13:33:46: MQTT failed to connect, exiting...
Zigbee2MQTT:info  2023-03-15 13:33:46: Stopping zigbee-herdsman...
```

IgnacioHR, 2023-03-15:
Once MQTT access is fixed, I can confirm version 1.30.2 works for my SonOFF Dongle-E device.

Side note. IMO the device itself is not ready for production yet. I can pair a router (TS011F_plug_1) that is in radio access of the dongle, but I can't pair another router that is in radio access of the TS011F_plug_1 but not in the radio access of the dongle so I need to wait for a new firmware to see if that is fixed.

jerrm, 2023-03-15:
Could be an issue with the default +20db signal. It may be a long wait hoping there is a firmware fix. 

Have you tried pairing directly with the router device?  "Permit Join" is a drop down and you can select a specific router device to join to. 

If it is signal strength related, you may need to lower the output of the coordinator.

IgnacioHR, 2023-03-15:
@jerrm if I drop down the "Permit join" and select the specific router I can successfully join the second router. I'll test lowering the strength of the dongle. 

Thank you all for your help!

jerrm, 2023-03-15:
You might leave it for a while and see.  It could be ezsp pairing oddities.  I see those under ZHA as well.

The problem with the +20db is it can make the end device think the coordinator is closer than it really is.    The end device may think the coordinator shouting at +20 is closer than the +0 router, but when it tries to connect to the loud coordinator, the coordinator can't hear the end device's at a relative +0 whisper.

mamrai1, 2023-03-15:
how do you lower coordinator transmit power? Is it possible?

aioue, 2023-03-18:
> how do you lower coordinator transmit power? Is it possible?

Brute force method: unscrew the antenna?

IgnacioHR, 2023-03-18:
@mamrai1 look at advanced options here https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html but I’m not sure it works for the Dongle-E

aioue, 2023-04-10:
Tested fresh build of latest z2m on ZBDongle-E (factory default firmware). It's working fine, joins OK, no errors or instability. @trackhacs test and close issue?

Zigbee2MQTT version [1.30.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.3) commit: [24c6b2e](https://github.com/Koenkk/zigbee2mqtt/commit/24c6b2e)
Coordinator type, EZSP v8
Coordinator revision, 6.10.3.0 build 297
Frontend version, 0.6.127

trackhacs, 2023-05-12:
currently switched to different dongle (which also prevented the fast discharge of Ikea remotes) and currently do not have the time to switch back for testing. 

but since it seems to be fixed now - i am closing the topic.

closed 2023-05-12 by trackhacs