Koenkk/zigbee2mqtt #15962: Nodon Sin-4-1-20 ending up read-only

MichaelMure, 2023-01-05:
### What happened?

Following https://github.com/Koenkk/zigbee2mqtt/discussions/15683 (more details there), I'm opening an issue. I'm not currently running the last release, but I've seen nothing related in the release notes.

I paired my Nodon Sin-4-1-20 and while most thing seemed OK I couldn't control the relay state, getting `failed (Status 'READ_ONLY')'` errors.

Following the discussion above I stopped zigbee2mqtt, edited the `database.db` file, started it again and everything worked.

I only changed `"epList":[242, 1]` to `"epList":[1,242,]`.

Config before:
> {"id":30,"type":"Router","ieeeAddr":"0x04cd15fffea57734","nwkAddr":38585,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-1-20","epList":[242, 1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,7,4096,64599],"outClusterList":[3,6,25],"clusters":{"genBasic":{"attributes":{"appVersion":2}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":103,"hwVersion":34,"dateCode":"2020","zclVersion":8,"interviewCompleted":true,"meta":{"configured":-1141624182},"lastSeen":1672926795987,"defaultSendRequestWhen":"immediate"}

Config after
> {"id":30,"type":"Router","ieeeAddr":"0x04cd15fffea57734","nwkAddr":38585,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-1-20","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,7,4096,64599],"outClusterList":[3,6,25],"clusters":{"genBasic":{"attributes":{"appVersion":2}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001caa8446","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":103,"hwVersion":34,"dateCode":"2020","zclVersion":8,"interviewCompleted":true,"meta":{"configured":-1141624182},"lastSeen":1672926795987,"defaultSendRequestWhen":"immediate"}

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

zzh!

Koenkk, 2023-01-06:
Found and fixed the issue!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-06 by Koenkk

MichaelMure, 2023-01-06:
Awesome, thank you :-)

Wireheadbe, 2023-02-10:
For info, I was having the same with the Roller Shutter module from Nodon on latest stable release. @Koenkk - seeing the commit was done on the 6 jan - is it possible these were not included in the latest stable release? Switching to latest-dev fixes the issue immediately.

Koenkk, 2023-02-10:
@Wireheadbe it is in z2m 1.30.1

Wireheadbe, 2023-02-11:
Odd - was running:
```
koenkk/zigbee2mqtt                      latest       sha256:a5ecbd5fccbc0379dc1da2a5a886d81270d009ab2b756c1e37fe7f96b592b2d2   4 days ago
```
Anyhow, fixed with the commit in zigbee-herdsman-converters 👍🏻 thanks!

Wireheadbe, 2023-02-15:
@Koenkk  - just for info: ordered more of these [SIN-4-RS-20](https://www.zigbee2mqtt.io/devices/SIN-4-RS-20.html#nodon-sin-4-rs-20).
The issue was in fact not related to latest or dev.. But everytime I add one of these, they remain read-only, until I reload my instance of zigbee2mqtt. Once I've reloaded zigbee2mqtt, they all work fine. Something with initial add to the config?

Koenkk, 2023-02-15:
That's very strange indeed, can you provide the herdsman debug log of both the working and non working situation?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

moshiko2312, 2023-04-22:
It doesn't work for me either, everything seems to be configured well but the product doesn't respond to commands

kachalex, 2023-04-22:
install docker koenkk/zigbee2mqtt:1.30.3
this device Nodon Sin-4-1-20  does not work.
It is paired and detected, but that's all

Wireheadbe, 2023-04-22:
The SIN-4-RS-20 works well, after initial pairing, restart zigbee2mqtt. It will work.
There's something in the initial pairing that is off, I haven't found out yet.

Note that you can install the NoDon connect app. You can update the firmware of the modules via bluetooth. Give that a try, maybe you're on an old version.

moshiko2312, 2023-04-22:
Ok i will check again thanks

kachalex, 2023-04-22:
> The SIN-4-RS-20 works well, after initial pairing, restart zigbee2mqtt. It will work. There's something in the initial pairing that is off, I haven't found out yet.
>
> Note that you can install the [...]

it's working
thank you

moshiko2312, 2023-04-22:
2023-04-22 16:39:57Publish 'set' 'state' to 'nodon_sh' failed: 'Error: Command 0x5c0272fffe62a2e5/1 closuresWindowCovering.stop({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 12639 - 1 - 7 - 258 - 11 after 10000ms)'

moshiko2312, 2023-04-23:
Thank you very much, everything works great, I had a very old module with a version that cannot be updated, I replaced the module with the latest version, everything works great

wucke13, 2023-09-14:
For me, restarting zigbee2mqtt did the trick for making it work as well.