Koenkk/zigbee2mqtt #15964: Template variable warning: 'dict object' has no attribute 'latest_version'

DieMetRik, 2023-01-05:
### What happened?

Many error at log. 18h is over 20k hits

### Zigbee2MQTT version

1.29.0-1

### Adapter firmware version

20220219

### Adapter

Lily poe

### Debug log

Logger: homeassistant.helpers.template
Source: helpers/template.py:589
First occurred: 09:07:33 (19225 occurrences)
Last logged: 17:56:56

Template variable warning: 'dict object' has no attribute 'power_on_behavior' when rendering '{{ value_json.power_on_behavior }}'
Template variable warning: 'dict object' has no attribute 'latest_version' when rendering '{{ value_json['update']['latest_version'] }}'

KitHubek, 2023-01-05:
https://github.com/home-assistant/core/issues/85138

Koenkk, 2023-01-06:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/15747, will be fixed in todays hotfix release.

closed 2023-01-06 by Koenkk