Koenkk/zigbee2mqtt #15965: Reset / hang of coordinator not handeled by z2m

Bascht74, 2023-01-05:
### What happened?

Hi, 

I a have >120 zigbee devices an a CC2652P (zigstar) as coordinator. Sometimes the coordinator seems to reset itself and sends `00 fe 06 41 80 00 02 01 02 07 01 c0` to the client (Z2M 1.29.0), but Z2M is not reacting anymore or handeling that.
 
It just doesn't recognize the situation, restarts, initialize die communication again.
Just nothing, pushing out some mqtt messages and saving the state...: 
```
Zigbee2MQTT:info  2023-01-05 16:45:14: MQTT publish: topic 'zigbee2mqtt/Motion_Aussen_Balkon', payload '{"battery":100,"illuminance":19576,"illuminance_lux":91,"last_seen":"2023-01-05T16:40:25+01:00","led_indication":true,"linkquality":58,"motion_sensitivity":"medium","no_occupancy_since":600,"occupancy":false,"occupancy_timeout":10,"temperature":12.7,"update":{"installed_version":"2.53.6","state":"idle"},"update_available":null}'
Zigbee2MQTT:debug 2023-01-05 16:45:23: Saving state to file /config/zigbee2mqtt/state.json
...
```
Is there a way to recognize this situation and re-establish a connection again...?

Right now I have to check with an automatisation within HA that there is no device message coming in anymore and restart the add-on. After that it works again for some time (18 - 48 hours)

### What did you expect to happen?

Re-establish the connection to the coordinator, e.g. restart Z2M by itself or try re-initialize the communication. 
Maybe have an entity to show if the communication with the coordinator is "running" / "not running" ...
That way I could be informed about it...

Maybe z2m could even reset the coordinator itself if it hangs by sending the topic to zigstar? 
(`ZigStarGW-topic/cmd` --> `rst_zig`)
This could help in other situations as well. Maybe because of this some devices did go offline after a while because there is no communication from the coordinator anymore...
(But that would be a feature request)

### How to reproduce it (minimal and precise)

From time to time the coordinator seems to restart itself. So just wait (and have lots of zigbee devices)

But you can force it e.g. with zigstar, if you press "reset zigbee". 
After that Z2M is still running, but no reaction. 

If you e.g. turn of a lamp, it will get some response:  
`Debug 2023-01-05 16:53:24Error: Command 0x001788010124504c/11 genLevelCtrl.moveToLevelWithOnOff({"level":2,"transtime":30}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequest - {"dstaddr":58098,"destendpoint":11,"srcendpoint":1,"clusterid":8,"transid":6,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,4,4,2,30,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)')) at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27) at processTicksAndRejections (node:internal/process/task_queues:96:5) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)`

But doing nothing about it... Maybe there is a better way handeling basic instabilities like this...

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20221220 (right now, tried others as well)

### Adapter

Zigstar GW 0.6.10 with CC2652P

cloudbr34k84, 2023-01-06:
i think i have the same issue. Whenever i enable an entity with the device page in HA (not Z2M frontend all my devices become unavailable for 30 seconds.

pch18, 2023-01-08:
i have same problem !

Bascht74, 2023-01-09:
@cloudbr34k84 That is not my problem here. Please do open an issue.

RubenKelevra, 2023-01-28:
@Bascht74 I think I got possible the same issue?

https://github.com/Koenkk/zigbee2mqtt/issues/16372

closed 2023-03-07 by github-actions[bot]