Koenkk/zigbee2mqtt #15968: "No Update Available" shown for Hue bulbs in Z2M Dashboard, but Home Assistant reports an available update and can successfully trigger it.

jakemauer, 2023-01-05:
### What happened?

Several of my Hue bulbs are showing available updates in Home Assistant, but report no available update in the Z2M dashboard OTA page. When I "check for new updates" in Z2M it reports there are no available updates. However I can trigger an update from Home Assistant and see the update complete in Z2M.

For example my Par 30 Hue downlight bulbs, model 464800 are on firmware 1.93.11. When I ask Z2M to check for updates it says "no update available" but Home Assistant was able to send the request to update a bulb to version 1.101.2.

Is this because Home Assistant is providing the firmware and Z2M currently doesn't see that latest firmware in its sources? 

### What did you expect to happen?

I expected to see the same available updates in Z2M that I see in Home Assistant

### How to reproduce it (minimal and precise)

1. Check for new update on a Hue bulb with outdated firmware like 1.93.11
2. If unable to get update check Home Assistant's settings and trigger the update there.

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

20220219

### Adapter

Sonoff P Dongle running zStack3x0

stonie81, 2023-01-06:
same here.
Adapter is the one from slaeh (CC2652RB)
Z2M 1.29.0
IKEA Switch shows it's updatable, but update won't start in Z2M. But if initiated from Home Assistant it starts.
Other IKEA stuff shows no update notification in Z2M. But as the switch is updatable from HA.

Koenkk, 2023-01-06:
Will be fixed in todays hotfix release.

closed 2023-01-06 by Koenkk

jakemauer, 2023-01-06:
Can confirm 1.29.1 fixes it. Thank you to all the devs!