Koenkk/zigbee2mqtt #15969: OTA Update available (and not) at the same time.

trackhacs, 2023-01-05:
### What happened?

I paired   IKEA E1744. 

Afterwards HASS showed me that an update is available.

[image]

Going afterwards to Z2MQtt   via UI and checking for update gives: "no update available". 

[image]

The state.json:

[image]

### What did you expect to happen?

a)  same feedback from HASS and Z2M about available firmware update.

b) same coding/naming of available updates in human readable notion.

### How to reproduce it (minimal and precise)

pair  E1744

wait for HASS showing update availabel 

test via Z2M  for an update 

### Zigbee2MQTT version

1.29.0

### Adapter firmware version

0x26720700

### Adapter

Conbee2

### Debug log

```
Debug 2023-01-05 19:35:27Received Zigbee message from '0x60a423fffef48f79', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":553797169,"imageType":4554,"manufacturerCode":4476}' from endpoint 1 with groupID null
Info 2023-01-05 19:35:27MQTT publish: topic 'zigbee2mqtt/0x60a423fffef48f79', payload '{"action":null,"battery":60,"last_seen":"2023-01-05T18:35:27.602Z","linkquality":255,"update":{"installed_version":553797169,"latest_version":587728433,"state":"idle"},"update_available":null}'
Debug 2023-01-05 19:35:27Device '0x60a423fffef48f79' requested OTA
Debug 2023-01-05 19:35:30Received Zigbee message from '0x60a423fffef48f79', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":553797169,"imageType":4554,"manufacturerCode":4476}' from endpoint 1 with groupID null
Info 2023-01-05 19:35:30MQTT publish: topic 'zigbee2mqtt/0x60a423fffef48f79', payload '{"action":null,"battery":60,"last_seen":"2023-01-05T18:35:30.711Z","linkquality":255,"update":{"installed_version":553797169,"latest_version":587728433,"state":"idle"},"update_available":null}'
Debug 2023-01-05 19:35:30Device '0x60a423fffef48f79' requested OTA
Debug 2023-01-05 19:35:34Received Zigbee message from '0x60a423fffef48f79', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":553797169,"imageType":4554,"manufacturerCode":4476}' from endpoint 1 with groupID null
Info 2023-01-05 19:35:34MQTT publish: topic 'zigbee2mqtt/0x60a423fffef48f79', payload '{"action":null,"battery":60,"last_seen":"2023-01-05T18:35:33.809Z","linkquality":255,"update":{"installed_version":553797169,"latest_version":587728433,"state":"idle"},"update_available":null}'
Debug 2023-01-05 19:35:34Device '0x60a423fffef48f79' requested OTA
```

jakemauer, 2023-01-05:
Wow we posted the same issue like a minute a part lol.

trackhacs, 2023-01-05:
;)

yes you right. now I can trigger it via HASS too. Tried before and didnt start.

stigvi, 2023-01-05:
https://github.com/Koenkk/zigbee2mqtt/issues/15877#issue-1516178147

Koenkk, 2023-01-06:
Will be fixed on todays hotfix release.

closed 2023-01-06 by Koenkk