Koenkk/zigbee2mqtt #15970: [Feature request]: Avatto 3G Switch with Neutral backlight mode

ajunio3, 2023-01-05:
### Is your feature request related to a problem? Please describe

Currently Z2M Detects this device as a Zemismart one, that seems not to have backlight and power_on_behaviour exposed. But whithing Tuya gateway I can control both entities

This is the actual device

https://pt.aliexpress.com/item/1005001947460842.html

This is how it looks in Z2M

[image]
[image]

### Describe the solution you'd like

Include both entities Power_on_behaviour and backlight

### Describe alternatives you've considered

Tried using some external converters but no success

### Additional context

I have another model, the version without neutral also from the same supplier that exposes both entities but they don't respond

[image]

ajunio3, 2023-01-17:
I'v tested the 6 gang version of the same switch and have the same issue, not exposing the backlight_mode nor the power_on_state entities

ajunio3, 2023-01-20:
by the way, for those who are interested, even though I was not able to add the backclight entity for the switches in HA I can use script to publish in their individual topics to turn on/off their bacligkts

This way ("tuyaBacklightSwitch":1 to rurn on and tuyaBacklightSwitch":0 to rurn 0ff, example below I"m turning on

topic: zigbee2mqtt/AVATTO 3GN/1/set
payload: {"write":{"cluster":"genOnOff","options":{},"payload":{"tuyaBacklightSwitch":1}}}

ajunio3, 2023-02-02:
This is the database entry:

{"id":2,"type":"Router","ieeeAddr":"0xa4c138dc20a0578a","nwkAddr":62540,"manufId":4417,"manufName":"_TZ3000_pv4puuxi","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000f@T+\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_pv4puuxi","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":5,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1,"switchType":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00256c31a3","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00256c31a3","endpointID":1}],"configuredReportings":[],"meta":{"scenes":{}}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00256c31a3","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00256c31a3","endpointID":1}],"configuredReportings":[],"meta":{"scenes":{"0_2":{"name":"Scene 0","state":{"color_mode":"xy"}}}}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00256c31a3","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00256c31a3","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-150731130},"lastSeen":1675368644813,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-02-04:
Could you check if the issue is fixed with the following external converter: 

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0003', ['_TZ3000_vsasbzkf', '_TZ3000_odzoiovu', '_TZ3000_4o16jdca', '_TZ3000_pv4puuxi']),
    model: 'TS0003_switch_module',
    vendor: 'TuYa',
    description: '3 gang switch module',
    whiteLabel: [{vendor: 'OXT', model: 'SWTZ23'}],
    extend: tuya.extend.switch({switchType: true, endpoints: ['l1', 'l2', 'l3']}),
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2, 'l3': 3};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

ajunio3, 2023-02-04:
> Could you check if the issue is fixed with the following external converter:
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

Thanks a lot for responding

Did as suggested, It exposes power on behaviour and the switch type (used for internal micro modules not for wall mounted switches)

But the important Backlight is not exposed

[image]

Koenkk, 2023-02-05:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/15970#issuecomment-1416747601 to add backlight.

ajunio3, 2023-02-05:
It get's exposed but no effect

[image]

[image]

ajunio3, 2023-02-05:
However, if I publish this I can get on/off on the backlight

Debug 2023-02-05 13:32:25Received MQTT message on 'zigbee2mqtt/Avatto 3GN/1/set' with data '{"write":{"cluster":"genOnOff","options":{},"payload":{"tuyaBacklightSwitch":0}}}'
Debug 2023-02-05 13:32:25Publishing 'set' 'write' to 'Avatto 3GN'

Koenkk, 2023-02-06:
What are the possible values and what to they do? e.g. `"tuyaBacklightSwitch":0`/`"tuyaBacklightSwitch":1`/`"tuyaBacklightSwitch":2`/`"tuyaBacklightSwitch":3`

ajunio3, 2023-02-06:
0 = off
1 = on
2 = never tested 🗡️ 
But using Tuya gateway, there's only one switch to turn on/off as the normal gangs

ajunio3, 2023-02-06:
Exactly like this

[image]

Koenkk, 2023-02-07:
Thanks, I've added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-02-07 by Koenkk