Koenkk/zigbee2mqtt #15971: Iris 3460-L Unsupported Attribute Errors

shackrat, 2023-01-05:
### What happened?

The Iris 3460-L does not support binding the on/off cluster and will throw UNSUPPORTED_ATTRIBUTE errors when configuring.  When this occurs none of the reporting will work.

### What did you expect to happen?

The device to configure correctly and report temperature.

### How to reproduce it (minimal and precise)

Reset the device and pair as normal.  The button will function after the interview and configuration are complete.  Errors will be shown but the button will work.

### Zigbee2MQTT version

1.29.0 commit: dddce57

### Adapter firmware version

20220219

### Adapter

Sonoff ZBDongle-P

### Debug log

Failed to configure 'Media Center Button', attempt 4 (Error: ConfigureReporting 0x000d6f000b11cb52/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22) at Object.onOff (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:75:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/iris.js:115:13) at Configure.configure (/app/lib/extension/configure.ts:117:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))
Failed to configure (ConfigureReporting 0x000d6f000b11cb52/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status

shackrat, 2023-01-05:
A follow up to this, a custom converter that removes the on/off binding and reporting eliminates the UNSUPPORTED_ATTRIBUTE error and allows reporting to work.

```
module.exports = [
  {
      zigbeeModel: ['3460-L'],
      model: '3460-L',
      vendor: 'Iris',
      description: 'Smart button',
      fromZigbee: [fz.command_on, fz.command_off, fz.battery, fz.temperature],
      toZigbee: [],
      exposes: [e.battery(), e.temperature(), e.action(['on', 'off'])],
      meta: {battery: {voltageToPercentage: '3V_2100'}},
      configure: async (device, coordinatorEndpoint, logger) => {
          const endpoint = device.getEndpoint(1);
          await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'msTemperatureMeasurement']);
          await reporting.batteryVoltage(endpoint);
          await reporting.temperature(endpoint);
      },
  },
];
```

Will try to set up a repo and submit a PR if needed.

Koenkk, 2023-01-06:
Thanks, pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-06 by Koenkk

KyleStilkey, 2023-08-16:
Sorry to respond to this but did this fix get reverted? I'm currently getting this problem trying to pair my buttons.

2023-08-16 15:06:49Failed to configure '0x000d6f000b11bba5', attempt 3 (Error: ConfigureReporting 0x000d6f000b11bba5/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:419:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:22) at Object.onOff (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:74:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/iris.ts:116:13) at Configure.configure (/app/lib/extension/configure.ts:117:13))

annbob, 2023-12-21:
Thank you to @shackrat  for a fix to the Iris 3460-L Button problem.

The only thing that needs to be added to the custom converter below are definitions for the constants.

The revised code is below the quoted code.

> A follow up to this, a custom converter that removes the on/off binding and reporting eliminates the UNSUPPORTED_ATTRIBUTE error and allows reporting to work.
> 
> ```
> module.exports = [
>   {
>       zigbeeModel: ['3460-L'],
>       model: '3460-L',
>       vendor: 'Iris',
>       description: 'Smart button',
>       fromZigbee: [fz.command_on, fz.command_off, fz.battery, fz.temperature],
>       toZigbee: [],
>       exposes: [e.battery(), e.temperature(), e.action(['on', 'off'])],
>       meta: {battery: {voltageToPercentage: '3V_2100'}},
>       configure: async (device, coordinatorEndpoint, logger) => {
>           const endpoint = device.getEndpoint(1);
>           await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genPowerCfg', 'msTemperatureMeasurement']);
>           await reporting.batteryVoltage(endpoint);
>           await reporting.temperature(endpoint);
>       },
>   },
> ];
> ```
> 
> Will try to set up a repo and submit a PR if needed.
> 
> EDIT: Not sure why this issue didn't open in the converters repo. My apologies for that.
---
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
  {
      zigbeeModel: ['3460-L'],
      model: '3460-L',
      vendor: 'Iris',
      description: 'Smart button',
      fromZigbee: [fz.command_on, fz.command_off, fz.battery, fz.temperature],
      toZigbee: [],
      exposes: [e.battery(), e.temperature(), e.action(['on', 'off'])],
      meta: {battery: {voltageToPercentage: '3V_2100'}},
      configure: async (device, coordinatorEndpoint, logger) => {
          const endpoint = device.getEndpoint(1);
          await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genPowerCfg', 'msTemperatureMeasurement']);
          await reporting.batteryVoltage(endpoint);
          await reporting.temperature(endpoint);
      },
  },
];
```
---

Koenkk, 2023-12-21:
@annbob could you submit a pr for this?

annbob, 2023-12-21:
@Koenkk I created pull request. Not my specialty, so I hope I went down the right path.